AllskyTeam/allsky #382: bug in timelapse.sh when copying files locally

joshuakoplik, 2021-06-18:
line 51:

cp $FILENAME-resize.$EXTENSION /var/www/html/$MP4DIR &

should be something more like this I think?

cp images/$1/allsky-$1.mp4 $MP4DIR &

I also changed references to KEOGRAM_DIR and STARTRAILS_DIR in endOfNight.sh and ftp-settings.sh to be absolute paths - mad more sense to me that way, but not sure if that breaks anything else since I'm just copying for local apache.

EricClaeys, 2021-08-01:
Joshuakoplik, you are correct.  I, and a couple other people, are working on many changes to allsky that Thomas will likely buy back, and I corrected the cp statement you identified.  Thanks for bringing it up.
Eric

EricClaeys, 2021-08-20:
Joshuakoplik,
This has been fixed in the 0.8 version uploaded a couple weeks ago.  Thanks for pointing it out.
Would you please close the issue?
Thanks - Eric

closed 2021-08-21 by joshuakoplik