AllskyTeam/allsky #383: Wrong month on thumb tag

greenembrace, 2021-06-20:
Hi I.m getting the wrong month on the thumb tag seems to be exactly one month in future.  System date is correct and current:

Current default time zone: 'Australia/Melbourne'
Local time is now:      Sun Jun 20 09:55:38 AEST 2021.
Universal Time is now:  Sat Jun 19 23:55:38 UTC 2021.

[image]

AndreasLMeg, 2021-06-25:
maybe the wrong usage of string-substring is the reason
https://www.javatpoint.com/java-string-substring

https://github.com/thomasjacquin/allsky-portal
https://github.com/thomasjacquin/allsky-portal/blob/master/includes/images.php

```
function getTimeStamp(url)
{var filename = url.substring(url.lastIndexOf('/')+1); 

                                                    1111
                                          01234567890123
------------------------------------------------------------
var timeStamp = filename.substring(6); -> 20210624215115.jpg
var year = timeStamp.substring(0, 4);  -> 20210 ?
var month = timeStamp.substring(4, 6); ->     062 ?
var day = timeStamp.substring(6, 8);   ->       242 ?
var hour = timeStamp.substring(8, 10); ->         215 ? 
var minute = timeStamp.substring(10, 12); ->        511 ?
var seconds = timeStamp.substring(12, 14) ->          15. ?

```

thomasjacquin, 2021-06-25:
The javascript month index is 0 based which means 0 is January. 
https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Date/Date

Try using this 
`var date = new Date(year, month - 1, day, hour, minute, seconds, 0);`

on line 51 of this file. https://github.com/thomasjacquin/allsky-portal/blob/master/includes/images.php#L51

I'll make the change in the repo if it works.

greenembrace, 2021-06-25:
Hi,
Yep that would work, taking a one away, but why?
-rw-r--r-- 1 pi pi  4444752 Jun 20 11:12 image-20210620111223.jpg
date looks correct.

I tried your suggestion and it does the job. I just a bit confused.

Thanks for responding
daniel

thomasjacquin, 2021-06-25:
That's because the month index in the new Date() constructor is 0 based. (00 -> Jan, 01 -> Feb, 02 -> Mar, etc).
So if you give it 06, it will translate it to July.

AndreasLMeg, 2021-06-25:
[image]

greenembrace, 2021-06-25:
cheers

EricClaeys, 2021-08-21:
Thomas,
Has this been updated in the master?  If so, we can close this issue.
Eric

closed 2021-08-21 by greenembrace