AllskyTeam/allsky #384: Problem timelapse & live

flipper07, 2021-06-21:
Hello I have just restarted my Allsky, after several weeks of hiatus.
I have some problems:
- After some time after starting the raspberry on the live view remains on the same image and does not move, I must restart the rpi.
- Then the timelapses are not done all the time, however the images are in the image folder.
I would like to know how to do a timelapse with the files that have not been created.
- I still have a problem on my site to access the images and timlapse too.
If need config file or other, address of my site I will give them to you.
- is there an update from allsky which version is the latest and how do we know the version?
sorry for the english translation
Thank you so much for your help

jcauthen78, 2021-06-21:
Only way i know to check the version is to look at this file in the top line - as far as I know v0.7 is the most current. 
`/home/pi/allsky/README.md`

To try a manual timelapse, in the command line make sure you're in the /home/pi/allsky directory and use: 
`./scripts/timelapse.sh 20210620`
That will run the timelapse script on last night's images. 
Keep an eye on the terminal window and see if it gives any errors without finishing. 

As for the website, I don't have any suggestions

flipper07, 2021-06-22:
Thank you for the answer
For my version it's 0.6.
how to update?
I made the timelapse command in manual
It works but it doesn't end, the terminal window closes by itself after a few minutes.
Why are timelapses not always done automatically?
On the other hand I have a message at the top right of the rpi office:
Low voltage warning Please check your power supply.
I have my power supply which is 2.1ampere may not be sufficient?
maybe my problems come from that?
I have a pi3B.
I still have a lot of space on the sd card. it is the images that take up a lot of space ...
Thank you again for your help

[image]

jcauthen78, 2021-06-22:
Your image is showing your memory card almost completely full. Depending on the size of your saved images, they can take up a lot of space. Mine are sometimes over 1gb per night. Check your recycle bin, as well as delete or backup some of the older nights of images to make some space. 

As for updating, they still don’t have a process to automatically update. What I do is backup the main config file: 
`cp /home/pi/allsky/config.sh /home/pi/Documents/ `
 If you’ve modified any of the other files in the Scripts folder, back those modified files up as well.

After that, re-run the install process from the main gut hub page and it’ll install the most current version. 

After that, I reconfigure it, using the backed up config.sh file, and other modified scripts. The newer versions sometimes contain new information, so you don’t want to just copy the old files over the new ones - that can lead to headaches.  

* As for the under-voltage, it can be a headache to see. If you have more things than just the camera plugged in (fans, hard drive, etc), it’s possible all of it can eat up the available voltage and amperage. If your Pi3 hangs, crashes, or reboots randomly, it may be the case. If your pi is stable, and doesn’t have a lot of extra things plugged into it, you should be ok. Mine ran for years with the under voltage warning- but that’s just my experience.

flipper07, 2021-06-22:
297 / 5000
Résultats de traduction
I just changed the diet I took a 5v 3a before it was a 5v 2a.
It seems to work better, I'll see in time
I have another problem with condensation on the dome, I put resitances to remove the condensation but it does not work.
If you have a tip.
Thank you 
[image]

flipper07, 2021-06-25:
Good evening I just updated to 0.7
it works better
I still have a little problem on my site but the link of all Keograms and startrail videos does not work.
Here is the site link:
https://mescreations-r-b-07.pagesperso-orange.fr/allsky/
The videos are on the server here is the link:
https://mescreations-r-b-07.pagesperso-orange.fr/allsky/videos/allsky-20210614.mp4
I do not understand.
thank you for helping me

EricClaeys, 2021-09-05:
Is this still an issue?  I tried going to both URLs and they no longer exist.
When troubleshooting issues like this, first make sure the files are on the web server.  If they are, then AllSky is working fine and you need to modify the web page on your server that displays the images to point to the correct location.
If the images are NOT on your webserver, make sure all the directories specified in the ftp_settings.sh file have been created ON THE SERVER. If "/keograms" is listed in ftp_setting.sh but doesn't exist on the web server, AllSky will not be able to copy the keogram file there.