AllskyTeam/allsky #385: RPiHQ Camera issue when changing to Night

NeilBuc007, 2021-06-23:
Has anyone seen this before with the HQ camera? The log shows it changing to NIGHT but for some reason it decides to take a 0s exposure.

Jun 22 23:04:38 allskystenton allsky.sh[4209]: Determine if it is day or night using variables: desired sun declination angle: -6 degrees, latitude: 55.96N, longitude: 2.606W
Jun 22 23:04:38 allskystenton allsky.sh[4209]: Check for day or night: NIGHT
Jun 22 23:04:38 allskystenton allsky.sh[4209]: Saving 0 seconds exposure images with 5000 ms delays in between...
Jun 22 23:04:38 allskystenton allsky.sh[4209]: Press Ctrl+Z to stop capturing images...
Jun 22 23:04:38 allskystenton allsky.sh[4209]: Capturing & saving image...
Jun 22 23:04:38 allskystenton allsky.sh[4209]: Capture command: nice raspistill --nopreview --thumb none --output image.jpg --burst -st --mode 3 --exposure auto --analoggain 4 --awb off --awbgains 3.24,1.47 --vflip --saturation 50 --quality 95 -a 1036 -ae 32,0xff,0x808000

Basically the output was a black picture. It was working ok. But I decided to update it and since then it hasn't worked. Daytime is fine as can be seen here: https://stenton.allsky-scotastro.co.uk

Cheers

Neil

NeilBuc007, 2021-06-23:
Automatic gain transition #358 
This looks promising. I have implemented it and will see what happens this evening.

EricClaeys, 2021-09-05:
Is this still an issue?  If so, are you using auto-exposure for night?  If not, what's your exposure set to?  Note that exposure time is in milliseconds (1000 ms = 1 second).
Looking at this line in the log may give a clue - note it says "0 seconds".
`Saving 0 seconds exposure images with 5000 ms delays in between...`

EricClaeys, 2021-09-05:
If this is no longer a problem, please close the issue.  Thanks.

closed 2021-10-16 by EricClaeys