AllskyTeam/allsky #386: Video Streaming Option

KH6JAG, 2021-06-25:
I currently use RPi Cam Web Interface with an RPiHQ camera as a digital eyepiece for the telescopes at the South Florida Science Center observatory.  The real-time observation is then displayable on any browser app.  I have a couple of camera units I am evaluating to use as an allsky camera.  These would be setup at a dark site and the real-time video linked back to the science center.  Is it possible to incorporate the RCBI steaming capability into the allsky capability to integrate an option for live streaming?
I also have another option where I setup the individual cameras as a Zoom meeting participant and show real-time observations.

Tartouffe, 2021-06-29:
I think it is not possible to run the camera either in still mode AND film mode (streaming). It also would collide, if switchable, with other website visitors (who's first?!). 
At second, to my understanding, streaming requires live encoding of the video stream. I think a Raspi would be unable to cope one or several streams/users the same time in reasonable image quality. A better way would be buying a cheap / second system or surveillance / network cam (under $50) for that, and seperate link to.

But well, it is possible to stress your Raspi with e.g. a VLC stream. But that solves not the problem with the camera-in-use by allsky, which had to stop the stills (and later manual restart) for the request of a video stream.

KH6JAG, 2021-06-29:
Thanks for the reply.  
I’m currently and experimenting with multiple cameras.  In one configuration I use the RPiHQ cam as an all skycam and do a stream of what is being observed by the telescope simultaneously.   I use RPi Cam Web Interface as the viewing SW.  At the telescope I have the camera set up with a 7” touchscreen for control.  On a larger monitor I use a split screen view. 
 I just started experimenting with a ZWO ASI385MC but I don’t have the flexibility of SW packages I have with the RPi cameras.  
The other configuration I am experimenting with involves using an RPi GOES receiver for a look at the earth and an AllSky cam for the skyward view.

EricClaeys, 2021-09-21:
@KH6JAG, adding video streaming would be a very large undertaking.  The whole design of AllSky is for taking pictures, not movies, so the basic design would need to change in order to support movies.  It would basically be writing a whole new program and wouldn't be ready in the timeframe you need something.
I suggest looking at other alternatives.  If you settle on something, let us know so we can put it in a FAQ.

If this answers you question, would you please close the issue?  Thanks - Eric

KH6JAG, 2021-09-21:
Thanks for the response.  I currently using the RPi HQ camera for live streaming of observations at the South Florida Science Center observatory.  This was to prevent potential viral transfers off the eyepiece.  I currently use RPI Cam Web Interface as the SW.  It has a similar interface to the AllSky GUI where all control is done the browser.  

The observation can be seen thru any browser, (Win, Mac, Linux, smartphone).  The unit is battery powered, works with either a 1.25” or 2” eyepiece tube.  It is very simple to operate, turn it on, insert it into the tube using either prime focus or with an angle.  Log onto the internal or external IP address if port forwarded.  If desired user privileges can also be assigned.  This configuration only works with the RPi CSI interface. 

Thomas has done great work with his AllSky SW that allows the use of both the RPiHQ (CSI) and ZWO (USB) cameras.  What I thought might be possible would be to have a streaming capability using a similar browser interface that would allow for a USB camera that would support real-time observations.  I currently use both the ZWO120MC-S and ZWO385MC cameras.

My current work around for the USB camera streaming is to use the RPi with Zoom.  The RPi camera becomes a Zoom meeting participant.  This configuration also allows the observation to be viewed remotely and have another participant provide narration concerning the observation.

Because the observatory is in a light polluted area, I also have been experimenting with an AllSky / remote telescope camera setup that would be installed at a darksky location.  One of the camera configurations is the build described by Thomas.  However, because this is at a remote site it needs to be battery operated and solar powered.  Real-time viewing is more challenging as there are no cellular capabilities at most of these areas.  Currently experimenting using DVB-S and DVB-T with a Portsdown transmitter and amateur radio frequencies.

I also have a configuration where you get view from space-to-Earth from the GOES-16 satellite and the view from Earth-to-space from the AllSky camera.  This camera configuration is also used for the weather station.

All the above are based on RPi configurations.

closed 2021-09-21 by KH6JAG