AllskyTeam/allsky #387: keogramm: labeling, startrails up or down and horizontal dark areas (pollution)

AndreasLMeg, 2021-06-27:
Yesterday I had a closer look at the keogram pictures.
To be honest, I wasn't exactly familiar with the function.

After analyzing the software, here is my interpretation:
* Exactly one column (middle of the picture) is taken from each picture of the night and this is lined up chronologically. (comparable to the finishing photo in sports)
* This gives you a quick overview of the night (exposure, clouds, ...)

A review of various AllSky websites showed me the following:
1. The displayed text information was mostly not readable or not interpretable (character size, meaning)
2. Mostly there were strange effects (lines going up od down) which I could not interpret exactly
3. some images also had dark horizontal tracks or areas

Such abnormalities interest me and I always want to find an explanation (nerd?)
to 1.:
  The cause of the unreadable text information is the text size. Depending on the set resolution, this should be adapted!
  Keogramm also has various start parameters, which are, however, unused (`endOfNight.sh`)
  I have now adapted this script to my requirements:
`  ../keogram $ ALLSKY_HOME / images / $ LAST_NIGHT / $ EXTENSION $ ALLSKY_HOME / images / $ LAST_NIGHT / keogram / keogram- $ LAST_NIGHT. $ EXTENSION -fontsize 1.0 -fontline 1 -fontcolor 255 255 255`
  It is now clearly visible that the hour information marks every hour change (vertical dashed line)

  I have also adapted the program in order to receive additional information when the date is changed
  (https://github.com/AndreasLMeg/allsky/commit/0476c0920f515da7c3093b537cce3f1aa2b95a2b)

to 2 .:
  Traces are created when points of light do not move horizontally across several images. Compare with "Startrails"
  Draw a vertical line in the middle of the picture.
  All stars that move along this line leave traces. (the decisive factor is the horizontal speed)
  That is not a problem in itself, but if you compare this with a photo of the finish (horse race, 100m sprint) I would like only one dot for a star transition.
  However, this is only possible if the vertical line is laid exactly through the axis of rotation (~ polaris). The horizontal speed is then at its maximum there.

  For this I have now adjusted the program again and added a new parameter -finishline. This allows you to place the recording at a specific point in the image
  (https://github.com/AndreasLMeg/allsky/commit/0476c0920f515da7c3093b537cce3f1aa2b95a2b)
  Open the startrail and determine the X coordinate of the axis of rotation (center of the circles)
 ` ../keogram $ ALLSKY_HOME / images / $ LAST_NIGHT / $ EXTENSION $ ALLSKY_HOME / images / $ LAST_NIGHT / keogram / keogram- $ LAST_NIGHT. $ EXTENSION -fontsize 1.0 -fontline 1 -fontcolor 255 255 255 -finishline 1363
`
to 3.:
  Some AllSky users have the problem that there is a "pollution" in the center of the image (anemometer, tree, street lamp) when setting up
  This pollution can be seen in every keogram as a horizontal line or area.
  With the adjustment from 2.) this can now be avoided - just look a little left or right (Always Look On The Bright Side Of Life)

I hope to have helped some interested people.
(sorry for my bad "google translate" english)

[image]

[image]

jcauthen78, 2021-07-24:
Thanks for these observations & analysis - I like the explanations you've provided. I really like the ability to move the end line. 

Question for you, if you have the time: With image sensor sizes increasing, the created keogram files are a little skinny/tall when compared to a 1800px tall image (what i've resized & cropped for my own system), but only, say, about 500-600 images per night, resulting in 500-600px wide. Is there a spot in the CPP code to tell it to grab a 2px column to make the resulting image a little wider? I'm thinking 2pix wouldn't add in too much distortion, while adding a little more visually comfortable width.

AndreasLMeg, 2021-07-24:
I already thought to increase the width, but rejected it because I think that this would create unsightly optical effects.
I will try out two more variants for narrow keograms:
1.) use a column twice
2.) use 2 columns of the original image
Simply enlarging is not an option, otherwise the font will be distorted.

AndreasLMeg, 2021-07-24:
here is the test.Made so much copies the get the source image size (~) 
original:
[image]

copy (same row): (looks better)
[image]

copy (neighbor): 
looks confused, but just shows the temporal behavior of an area
copy = star off, star on, star off
Neighbor = shows the passing of the star
[image]

jcauthen78, 2021-07-24:
Agree that the copy-same looks better, and gives a nice width to the resulting image

AndreasLMeg, 2021-07-24:
And here is change: https://github.com/AndreasLMeg/allsky/commit/0968e96014468c4c5936b527221e46d853b492ce
 new Param: -addRow

-addRow                      - Default = 0    - standard, keogram maybe small
1    - copy the same row to get nearby source images size
2    - copy the neighbor row to get nearby source images size

closed 2021-08-06 by AndreasLMeg