AllskyTeam/allsky #388: Several issues (timelapse, log, gain, starmap overlay) and future enhancements

Tartouffe, 2021-06-28:
ERRORS / ENHANCEMENTS ALLSKY
- timelapse.sh: hard coded path “/var/www/html”. Shouldn’t it use use the path variables from ftp-settings?!

- timelapse.sh: wrong (copy&pasted) code at option “local”: “($FILENAME-resize.$EXTENSION)”. --> Has to be: “images/$1/allsky-$1.mp4”  

- from the allsky.log, error at _“/scripts/darkCapture.sh”: line 10: printf: “25.2” invalid number_
   It works, but there is an ‘.’ vs. ‘,’ mismatch, comma vs. point for decimal separator / different country codes

- Gain/Max Gain seems to be limited (internally somewhere) to 100?, when using option Autogain=true. 
  When turned off, the value Maxgain is used, as it should. I presume its somewhere deep at the capture.cpp.

- Web preview image should be restricted NOT to zoom or resize on smaller screens, or the stars overlay would not fit. --> set fixed width @ allsky.css, section img.current

- “virtualsky.json” at web root has no function on parameter changes?! (or only used at installation?)

- difficult for begginers to tune the web preview picture (and stamap overlay), especially when the aspect ratio of the camera did not fit with the default settings

Would be nice:
- Dimensions, Datarate and FPS of the timelapse video editable over the GUI  

- width / height of the live picture and Web picture editable over the GUI

- Little comment at ‘config.js’ to edit/tune HERE the starmap overlay (and much much more parameters at /virtualsky/virtualsky.js)

- manually name and set the image directory, so i can link it easy e.g. to a server or NAS, not on the internal SD card

- Drop a note somewhere, that when the power connection is weak, or has too low Ampere (e.g. also for fireing up a USB 3 camera), it is possible that e.g. a ZWO camera stops working, while allsky and the raspberry is still in a somehow alive state. To prevent that (any) ghost state, allsky capture could check if the capture file has -really- written on disk. If not, it can restart itself / restart the service.

EricClaeys, 2021-08-03:
Tartouffe, several of us are working on many enhancements/fixes to allsky.  Your first two bugs will be fixed as you mention.
For the third, can you let me know what's in the ~/allsky/temperature.txt file?  Is your decimal separator a period or comma?
For the 4th (Gain/Max Gain), do you have a ZWO (which model) or Pi camera?   If ZWO, you can try adding this to capture.cpp after the line "ASIGetNumOfControls(CamNum, &iNumOfCtrl);", then run "make capture" and restart it:
        printf("Control Caps:\n");
        for (i = 0; i < iNumOfCtrl; i++)
        {
            ASIGetControlCaps(CamNum, i, &ControlCaps);
            printf("- %s:\n", ControlCaps.Name);
            printf("   - MinValue = %ld\n", ControlCaps.MinValue);
            printf("   - MaxValue = %ld\n", ControlCaps.MaxValue);
            printf("   - DefaultValue = %ld\n", ControlCaps.DefaultValue);
            printf("   - IsAutoSupported = %d\n", ControlCaps.IsAutoSupported);
            printf("   - IsWritable = %d\n", ControlCaps.IsWritable);
            printf("   - ControlType = %d\n", ControlCaps.ControlType);
        }
See what's in the log file for auto gain.  With my ASI178MC, "AutoExpMaxGain" is from 0-510 with a default of 255.  The name implies it's only used with auto-exposure, but I don't know for sure.  "Gain" goes from 0-510 with a default of 210.

For “virtualsky.json”, you may need to force your browser to refresh the page after changing the file.  One way to do this is press F12 when on the page, then on the "Network" tab check the box to disable the cache.  Then reload the page.

What do you mean by "tune the web preview picture"?  I agree tuning the star overlay is a pain.  After hours of playing, mine is pretty close (Polaris and the Big Dipper are almost perfect), but stars in the south are off, especially near the horizon.

For "Dimensions, Datarate and FPS..." you only want them on the timelapse, not on each image?  To put on each image you can use the "text" option (which will work in the enhanced allsky).  Ditto for width and height.

Can you propose something for "Little comment at ‘config.js’"?  Or do a draft?

"manually name and set the image directory" will take some work, but I also have thought it would be nice to rename it.  You could try doing a symbolic link of ~/allsky/images to somewhere else.  If you do, let me know how it goes.

In the new version, if you have the "uhubctl" command installed, allsky.sh will reset the USB bus if it can't find the ZWO camera, or if it finds it but it's not configured correctly.

The new version allows setting daytime and nighttime gain and exposure.  Night max gain is supported but not day, although it will default to the night max.  You can't set the max exposure for daytime but it will be 60 seconds.
Auto-exposure in the camera is used for daytime, but due to a bug in ZWO's code, the new version will have a workaround based on the mean brightness of the image when the user specifies daytime autoexposure.
Wider range and faster response is camera-dependent so we can't change them.

Eric

Tartouffe, 2021-08-04:
Eric,

> Tartouffe, several of us are working on many enhancements/fixes to allsky.
I'm so glad for that. Fine and interesting piece of software. Its good that it will get some updates or enhancements.

Your questions:
> For the third, can you let me know what's in the ~/allsky/temperature.txt file?
There is a dot "." in my file. But my country code (Germany) decimal seperator is comma "," - so i presume the system expects a comma. And this will produce the warning at the log.
In e.g. whole Europe "." stands for thousands separator and the decimal separator is a comma "," - the opposite like its in the states.
Played around with some format conversations before the file was written, but had no luck (not the programmer experience). 

>For the 4th (Gain/Max Gain), do you have a ZWO (which model) or Pi camera?
I have a ZWO ASI183MC with 20MP (5496 x 3672 Pixel) with an Fujinon 185° fisheye attached. So i can test allsky with much more resolution and filesize than usual.
Also i'm having an (old, unused) UHC-S / L-Booster-Filter inside the camera for longer night exposures (max. 75-80s) and filtering out the light pollution. Also it protects the camera CCD from being burned in by glaring summer sun rays.

> If ZWO, you can try adding this to capture.cpp after the line (...)
Will try, but I'm not well-known at the compiling things... I will try in the next days.
The best pictures/settings in my case i'm having with:
- Auto Exposure: no
- max exposure: 75000   *camera dependant
- Gain: 25
- Max Gain: 225   *camera dependant
- Auto Gain: Yes
- Angle: -8   *if you use a filter at the camera like me: later switch to night mode

> For “virtualsky.json”, you may need to force your browser to refresh the page (...)
Ah. Will try that.

> What do you mean by "tune the web preview picture"? I agree tuning the star overlay is a pain.
The Web preview picture (live picture) was reduced in size automatically, when using a camera with higher resolution. I have to change the CSS command NOT to fit the image dynamically inside the window. Also, if not changed, the starfiled overlay would never really fit (only on big screens), because the starfield is never dynamically zoomed (always fixed size).
The user only sees always not fitting images. But someone has to decide: Proper starfield in place and on smaller screens (Tablet etc) not full fitted camera preview, or fitted preview and displaced starfield on smaller screens.
(If i remember it right it was the "img.current" CSS parameter at allsky.css file, where 'width' and 'max-width' all has to be set at the same value, plus "width" at the 'body' section also with this same value to prevent resizing by the browser)

> For "Dimensions, Datarate and FPS..." you only want them on the timelapse, not on each image? To put on each image you can use the "text" option (which will work in the enhanced allsky). Ditto for width and height.
No, only for the timelapse. Had to search until i found the place, where i can set the parameters e.g. for the size and datarate. As you can expect, my 20MPx camera produces much larger images, also the video can have a lot more quality.
It will be nice to set the (new) timelapse parameters (dimension and datarate) at the config editor.

> Can you propose something for "Little comment at ‘config.js’"? Or do a draft?
Like that (but in correct english ;) ):
  To set the starfield properly please refer all the option parameter at the file "virtualsky/virtualsky.js"
  To set the proper dimensions of the starfiled you normally only need the parameters "width", "height" and "az" (to rotate the starfield like your north-south alignment of your camera).
  You can set them at the file "virtualsky.json" at the root dir of the allsky (public) webserver. Please be aware, that the skychart always has to be some pixels bigger than your preview image, because the virtual sky covers full 180 degrees, what normally the fisheye picture doesn't.
  Also, if displayed on smaller screens, the camera preview *can* be resized dynamical by the browser (depends also on the camera resolution), but the star-overlay would never resize. So there can be some size and overlay mismatches. To prevent this, you have to study & modify the allsky.css, sections 'img.current' and 'body' to prevent resizing by browser.  
  
> "manually name and set the image directory" will take some work, but I also have thought it would be nice to rename it. You could try doing a symbolic link of ~/allsky/images to somewhere else. If you do, let me know how it goes.
Already done that. My "Images" dir is redirected onto my NAS, so no capture file was written onto the SD-card.
Only the internal picture viewer over the admin panel does not work with that, because its somewhere hard coded at the /var/www location...
Didn't explore that to be more precise. But this were my own modification, not to blame allsky for not working properly while redirecting "Image" to another place.

> In the new version, if you have the "uhubctl" command installed, allsky.sh will reset the USB bus if it can't find the ZWO camera, or if it finds it but it's not configured correctly.
Currently doing all the power cables new. Then it will feed the Raspi 4 over the GPIO Header, Pins 1 to 3 with 12V, instead over USB-C (can use thicker cables with more Amp). It is needed because i have a USB-3 camera attached, which drops the connection immediately on any small voltage change (allsky runs at my 12V solar grid). 
If this won't work, i will try it with uhubctl. At the moment i'm having 1-4 losses of my camera USB3 connection a day (depends on usage of the 12V grid by other components at my DC net). 

> The new version allows setting daytime and nighttime gain and exposure. 
Perfect. Waiting for that :) 

One left i found which needs some update:
Update the "font-awesome" at the head of "index.html" to a newer version. The used one v4 (?!) is rather old...  

Cheers

Jonk2, 2021-08-05:
Just to add some my thoughts on improvements (where possible).

1. Autoexposure should always (in my opinion) go to maximum before the gain is turned up. Think of it like iso on a dslr, this should be the lowest setting possible at all times, to achieve the required brightness on the histogram and keep the noise to a minimum.
2. Speaking of the histogram, can the autoexposure target be a region of interest? Another popular allsky program in use has this function, where you draw a circle on the part of the sky you want to use as the autoexposure area, with everything else outside of it not used. This will keep the sky fairly evenly exposed across many frames over the night and reduce or avoid flickering in the final timelapse, for the sky portion of it at least.
3. Choose across how many frames the autoexposure takes effect. What I mean is if a large cloud passes the ROI over the course of 1 frame (say 30second interval), then the camera will then spend the next x frames (a few minutes possibly) recovering from the sudden increase in brightness due to the cloud. The other software I mention gives you the option to set the number of frames, meaning 1 frame in this examaple will be too bright, possibly blown out, but the next once the cloud passes, will be ok, and won't take ages to recover.
4. Add the possibility to show a weather feed from the internet - I use openweathermap data in my sequencing software, as many others do. This would be useful to show the weather for that given frame, and may even look good on a timelapse, where the temperature, humidity, windspeed will chage throughout the timelapse.
5. Add the possibility of data from a local weather / sky quality device as text on the overlay, much in the same way as point 4.
6. Full customisation of the displayed text, plus logo capability.

I appreciate some of this will take a lot of work, and may not even be possible, but I believe that this could be the best allsky software available to the masses, and being able to run on an RPi4 on the end of a POE cable is brilliant, and the reason I currently want to use it in a fit and forget scenario.

jcauthen78, 2021-08-05:
In regards to 4, 5, 6:
4 - I've thought about this one for a while too and I'm partially there - I've managed to get python to pull some of the data I want from their .JSON files, but it's a stand-alone script that I'm still working on. Was trying to go with Wunderground, but that's not too feasible for default end-user, as they'd need to generate their own API keys.

5 - My system has a simple Adafruit BME280 (temp, hum, pressure) I2C sensor. It's read every minute & dumped to a CSV for graphing later. Python again come in and reads the CSV, then creates the overlay text I want on my images.

6 - Similar to 5, it's the same script, but overlays all the data I wanted, as well as a logo (still re-doing the logo, so not currently used). It worked well with the logo.  

Here's an idea of what the current output is of my main system (note that I've been testing the new changes, so some of the info overlay in the top left may be a little different). https://jimsnerdstuff.com/allsky/ 

I'm not sure how all that would be incorporated into a default end-user setup, and my code isn't all that elegant, but I can provide what I've got of the main Overlay script if interested.

Jonk2, 2021-08-05:
That's looking more like the kind of thing I mean yes.

I use Rob Brown's MYSQM Pro (arduino based) and that spits out a csv. A simple reading of the header to choose what to display and show the last line only (i.e. the latest or as 'live' as you can get).

I'll have to wait until someone updates the software on here before I try it, as I'm not confident in too much modification, but if it's an easy 'put this file here and enable it' then I can do that!

EricClaeys, 2021-08-06:
Tartouffe,

 

The new version lets you set the locale so "capture" should output a comma for the decimal point, which is what bash is looking for.  You'll use either "de_DE" or "de_DE.UTF-8".  Run the "locale" command on your Pi and let me know what "LC_NUMBER" is.

 

You can set the timelapse size now via the config.sh file.  See the TIMELAPSEWIDTH and TIMELAPSEHEIGHT variables.

It's possible to overlay a bitmap on the timelapse video; I'll need to see if it's possible to overlay text.

I believe the datarate is 2000k.  It should be possible to change this via a new variable in config.sh.

 

My CSS knowledge is fairly basic so I'm not sure how helpful I'll be with modifying allsky.css.

 

Thanks for the draft wording about config.js.  I wonder if there's  a way to get the star overlay to resize?  That would solve a lot of problems.

 

For linking "images" directory to your NAS, try the new version Thomas just merged into his master.  In config.sh, set the "IMG_DIR" to "current".  That will tell the web GUI to look in /home/pi/allsky for the images rather than /var/www/html.

 

Additional comments by you:

1. We use ZWO's auto-exposure at night time.  It's recommended NOT to use auto-exposure and auto-gain at the same time since it can produce black frames.

2. The histogram code looks at a rectangle that by default is 500x500 pixels in the center of the image.  You can change that.

3. The histogram code takes multiple pictures until one has a mean brightness within a pre-defined range.  Since this is only done during the day, and daytime pictures are usually 1/2 second or less, this goes pretty quickly.  Hence, I don't think we need what you describe.  And once ZWO fixes their bug, we'll disable the histogram code.

4. Once you have the data in a file, you can pass that file to the "capture" program via the "extratext" option and it will display it.  Per Jim's comment, he's working on getting information into a file.

5. Same as # 4.

6. What kind of customization of the overlay text are you wanting?

EricClaeys, 2021-08-06:
The new version lets you set the locale so "capture" should output a comma for the decimal point, which is what bash is looking for.  You'll use either "de_DE" or "de_DE.UTF-8".  Run the "locale" command on your Pi and let me know what "LC_NUMBER" is.

You can set the timelapse size now via the config.sh file.  See the TIMELAPSEWIDTH and TIMELAPSEHEIGHT variables.
It's possible to overlay a bitmap on the timelapse video; I'll need to see if it's possible to overlay text.
I believe the datarate is 2000k.  It should be possible to change this via a new variable in config.sh.

My CSS knowledge is fairly basic so I'm not sure how helpful I'll be with modifying allsky.css.

Thanks for the draft wording about config.js.  I wonder if there's  a way to get the star overlay to resize?  That would solve a lot of problems.

For linking "images" directory to your NAS, try the new version Thomas just merged into his master.  In config.sh, set the "IMG_DIR" to "current".  That will tell the web GUI to look in /home/pi/allsky for the images rather than /var/www/html.

Additional comments by you:
1. We use ZWO's auto-exposure at night time.  It's recommended NOT to use auto-exposure and auto-gain at the same time since it can produce black frames.
2. The histogram code looks at a rectangle that by default is 500x500 pixels in the center of the image.  You can change that.
3. The histogram code takes multiple pictures until one has a mean brightness within a pre-defined range.  Since this is only done during the day, and daytime pictures are usually 1/2 second or less, this goes pretty quickly.  Hence, I don't think we need what you describe.  And once ZWO fixes their bug, we'll disable the histogram code.
4. Once you have the data in a file, you can pass that file to the "capture" program via the "extratext" option and it will display it.  Per Jim's comment, he's working on getting information into a file.
5. Same as # 4.
6. What kind of customization of the overlay text are you wanting?

Tartouffe, 2021-08-06:
Hi Eric,

> The new version lets you set the locale so "capture" should output a comma for the decimal point, which is what bash is looking for.
> Run the "locale" command on your Pi and let me know what "LC_NUMBER" is.
Good news. The result of 'locale' is:
LANG=de_DE.UTF-8
LANGUAGE=de_DE.UTF-8
LC_CTYPE="de_DE.UTF-8"
LC_NUMERIC="de_DE.UTF-8"
LC_TIME="de_DE.UTF-8"
LC_COLLATE="de_DE.UTF-8"
LC_MONETARY="de_DE.UTF-8"
LC_MESSAGES="de_DE.UTF-8"
LC_PAPER="de_DE.UTF-8"
LC_NAME="de_DE.UTF-8"
LC_ADDRESS="de_DE.UTF-8"
LC_TELEPHONE="de_DE.UTF-8"
LC_MEASUREMENT="de_DE.UTF-8"
LC_IDENTIFICATION="de_DE.UTF-8"
LC_ALL=de_DE.UTF-8

> You can set the timelapse size now via the config.sh file.  See the TIMELAPSEWIDTH and TIMELAPSEHEIGHT variables.
I know. I found (and modified) the parameters at timelapse.sh. I only mentioned for a third, the datarate parameter as additional variable. Some user have a lot of space free for the timelapse videos, or want better quality, if the camera has higher resolution.
Just one picture how it would looks with a high res cam and a lens who is almost capable to handle 20MPx: [URL=https://www.directupload.net/file/d/6267/wvnb9ugo_jpg.htm][IMG]https://s20.directupload.net/images/210806/temp/wvnb9ugo.jpg[/IMG][/URL]
Its one picture right out of the camera.

> Thanks for the draft wording about config.js.  I wonder if there's  a way to get the star overlay to resize?  That would solve a lot of problems.
Yes sure. You have to edit the /allsky/config.js at the webserver HTML pages.
To resize the star map, change (or add) any poosible StarMap variaböe here (all existing variables: see at the webserver /virtualsky/virtualsky.js, the list at the top).
var config = {
    title: "SkyCam",
    imageName: "image-resize.jpg",
    location: "Home",
    latitude: xx.xx,
    longitude: xx.xx,
    az: 175,
    camera: "ASI183MC",
    computer: "Raspberry Pi 4B",
    owner: "",
    auroraMap: "north",
#   overlaySize: 950,
    overlayOffsetLeft: 170,
    overlayOffsetTop: 0,
    auroraForecast: true,
}

You can change at the #-position (do not include the '#') the width/height of the star map. Or add/delete any other existing starmap variable.
In my case i have a light rotation of 5 degress to the left (variable az with 175 degrees) and a starmap width/height of 950px.
If this did NOT work properly, you have to detach the star container from the limiting container below.
Set at the '/allsky.css' at the root directory of the webpages the CSS-command 'position:relative':
starmap_container {
   position: relative;
(...)
}
to "postition:fixed". Then it is not bounded/linked to the item (-dimensions) which lays below the starmap (the preview picture). 
The items (picture and starmap) are working like a sandwich - the item above (starmap) could not be bigger without cropping, than the base below (preview image), until you remove 'the link between them', detach if.

> For linking "images" directory to your NAS, try the new version Thomas just merged into his master.  In config.sh, set the "IMG_DIR" to "current".  That will tell the web GUI to look in /home/pi/allsky for the images rather than /var/www/html.
Thats interesting. Will have to do that with the next update .. anyway you have to overwrite all (i hope i wrote down all of -my- modifications...)

> 2. The histogram code looks at a rectangle that by default is 500x500 pixels in the center of the image.  You can change that.
> 3. The histogram code takes multiple pictures until one has a mean [...]
Thats the ways it works. Ever wondered because i did not found anything at the (web) code. 
Lets wait how it works when the ZWO bug is fixed and histogram code is removed. 

So far for now :)

Tartouffe, 2021-08-06:
@Jonk2 

> gives you the option to set the number of frames, meaning 1 frame in this examaple will be too bright, possibly blown out, but the next once the cloud passes, will be ok, and won't take ages to recover.

i also noticed that, but not very often. 
On day time when very dark clouds covered the cam, the gain does *not* go up enough in that manner, that something can be seen on the picture. I'm having several weather conditions at the afternoon (!) with 1-2hours full of black images, until the storm was over. Allsky did not control the brightness or gain enough to compensate (rare/extreme) weather conditions.

EricClaeys, 2021-08-07:
Tartouffe,  I assume you had automatic _gain_ set but not automatic _exposure_?  With the new version, you'll probably want to change that to manual gain of 0 but auto exposure.

Tartouffe, 2021-08-08:
> I assume you had automatic gain set but not automatic exposure? With the new version, you'll probably want to change that to manual gain of 0 but auto exposure.

Eric, yes right, so that's my current set. Will try it that way.

EricClaeys, 2021-10-16:
@Tartouffe,
any update on this?