AllskyTeam/allsky #389: Version of AllSky

maserowik, 2021-06-28:
Is there a way to see what version of the AllSky program I have running?

AndreasLMeg, 2021-06-28:
I dont think so,

README.md
and the truth you will get with `cd allsky; git log`

EricClaeys, 2021-07-08:
Look in the logs (probably /var/log/allsky.log) near the top.  The version I have is "Allsky Camera Software v0.7 | 2020".  It's displayed whenever the "capture" program starts.

Jonk2, 2021-08-10:
Can this information be added to the admin page? Should be straighforwards.

EricClaeys, 2021-08-11:
The admin GUI is actually a separate software package called "allsky-portal", versus the "allsky" package that contains the programs to actually take and save the pictures.  "allsky" is now version 0.8.  "allsky-portal" is currently version 0.6, but should really be 0.7 since we just updated it.

closed 2021-08-16 by maserowik