AllskyTeam/allsky #390: Ftp broken "image-resize.jp" images at night

koal01, 2021-06-30:
Hi all,
I'm a new one and i built my first allsky with a T7 and a a Pi4 2 Go Ram. The webiste is hosted at Amazon using an EC2 instance and almost everything is working nice
I just have a little issue, at night the "image-resize.jpg" images transfered through FTP every 20 seconds are broken, I checked last night on the Pi4 the images are good but when transfered on the FTP there are corrupted, it seems that the process in Allsky of saving images and transfering is not handled correctly but i don't know what is causing this. During the day, the problem is not visible.
Do you have any idea, advice to solve this problem
Thank you
http://koal01.com
Koal01

EricClaeys, 2021-07-08:
Are there any errors in the log file (probably /var/log/allsky.log)?
In saveImageNight.sh, copy the lftp line near the end and prepend "echo", then delete the "&" at the end.  This will display the lftp command that's being executed.  Try running it manually (you'll need to add double quotes where you see them in the file).

What do you mean by "corrupted"?  Zero-length?  Different length than the one on the Pi?  If so, do an "ls -l" of the file being copied and see what size it is.

koal01, 2021-07-08:
Thanks for answering.
I partially solved the problem replacing the lftp lines in SaveImageNigh and Day.sh with an scp transfer.
When i use this method the liveview images on the website are normal but when i switch back to the ftp transfer the problem comes again with liveview images displaid partially as shown below.

_Are there any errors in the log file (probably /var/log/allsky.log)?_
Yes here is the error i can see in the log saying that the image size is decreased during transfer, this message does not appear when i'm sending through scp
Jul  9 01:26:21 boubous allsky.sh[805]: Resizing
Jul  9 01:26:21 boubous allsky.sh[805]: Uploading
Jul  9 01:26:34 boubous allsky.sh[805]: put: image-resize.jpg: file size decreased during transfer

_In saveImageNight.sh, copy the lftp line near the end and prepend "echo", then delete the "&" at the end. This will display the lftp command that's being executed. Try running it manually (you'll need to add double quotes where you see them in the file)._
I launched it manually and here is the message i get 
/home/pi/allsky/scripts/darkCapture.sh: ligne 10 : printf: 18.5: nombre non valable
Resizing 

Uploading 

_What do you mean by "corrupted"? Zero-length? Different length than the one on the Pi? If so, do an "ls -l" of the file being copied and see what size it is._
No the file corrupted is not a 0 size file but instead of having a size of 240 KBytes like the image on the Pi the same image on the website is just  70 KByte or 80, so the image is cut on the screen, the screenshot below shows this clearly.

Thank you

[image]

thomasjacquin, 2021-07-09:
Try increasing the delay between images and see if it helps. I believe the problem here is that while the `image-resize.jpg` file is 
uploaded, it gets replaced by the new one. This can come from a slow internet upload speed or a large image file.
Basically, you want to make sure the image is fully uploaded before it get overwritten by the new one.

koal01, 2021-07-09:
ok i'll increase the delay between uploads and check the difference, may i point just two observations, first my internet is 400 MB/s speed fibre link and secondly the bug is not visible during the day or exeptional. The bug occurs only during the night with the sameImageNight.sh process. Everything is working nice. Very Great job your Allsky Thomas !!!!

EricClaeys, 2021-07-09:
Are you doing dark subtraction?  That’s only done at night and takes some time (not sure how much though).  Are you saving daytime pictures?  Not sure what else is different between day and night.

 

Eric

koal01, 2021-07-09:
Hi Eric
_Are you doing dark subtraction?_ 
No
_Are you saving daytime pictures?_
No
Thanks for your help !

EricClaeys, 2021-07-09:
Hi,

 

If you’re not saving daytime pictures you obviously won’t see the problem.  For debugging, try saving daytime pictures and see if you see the problem during the day as well.

 

In a prior email I suggested running lftp manually.  It sounds like you ran something else manually.  Can you try running the lftp manually to see if you get the error?

 

Eric

EricClaeys, 2021-08-09:
koal01,
Did you try my latest suggestions?

Also, FYI, there's a newer version of AllSky available that might solve the problem of corrupted file, or at least give you a better idea what's happening.  You can also set your locale so you shouldn't see the message about "/home/pi/allsky/scripts/darkCapture.sh: ligne 10 : printf: 18.5: nombre non valable".

A couple things to note with the new version: 1) you'll probably need to increase your "USB Bandwidth" from 40 to, for example, 90.  2) "image-resize.jpg" is now just called "image.jpg".

Eric

koal01, 2021-08-11:
Hi Eric,
Sorry, i'm preparing a new hardware that will be finished in September,
I'll  test the new version and let you know if the FTP broken images issue
is still present at that time.
Thank you

Koal01

EricClaeys, 2021-10-16:
@koal01, any update?  The newest version 0.8.1 has a separate script (upload.sh) that does all the uploading of files.  It can be called directly to test uploads.  See the Wiki for more info.

koal01, 2022-04-07:
> Hi Eric, Thomas,
> I'm still working on my hardware trying to cool a datyson T7 with a
> peltier system.
> I had to double the peltier to reach an interesting temperature around
> -10° indoor.
> I still need a month to finish it and then i will be happy to set up your
> nice Allsky solution and test the update.
> Thank you so much for getting news
>
> Nice skies guies !
> Koal01

Hi Thomas, Eric
I'm back after a several month break.
Last year on my first Allsky version i had ftp problems in spite of a high
speed link with a corrupted images-resized file  and you helped me to solve
the problem
At the same time i decided to improve the hardware and my Allsky has to be
suspended so it was impossible for me to test you recommendations, my
apologies for that !
I'd like to test the 0.8 allsky version that Eric is talking in his latest
mail but i have 2 questions, where can i find this version information and
second question do i have to update all the files of the website ?
Thank you
Koal01

EricClaeys, 2022-04-08:
@koal0  Welcome back!   There have been a LOT of changes in the last several months so you'll want to do a complete install.  Please make sure you follow the instructions [here](https://github.com/thomasjacquin/allsky/wiki/Upgrade-from-0.8.2-or-prior-versions) **exactly** since you are upgrading from a much older version.

Upgrading the WebUI and Allsky Website (if you have it) are now non-destructive so installation is much easier.  They move your images and configuration files from the prior release to the new one.

Uploading files is MUCH more reliable now.

koal01, 2022-04-09:
Hi Eric, I tested the 0.8.3 but since I’m using two datyson T7 which are not supported any more in the latest version of Allsky I prefer using an older version.  I regret that since this camera is working nicely in my 0.7 version.
For ftp problems I found a solution transferring images via scp commands. 
Thank you

Koal01

EricClaeys, 2022-04-10:
@koal01 We don't **officially** support the T7, but the "Requirements" section of the README.md file has a link to unofficial instructions on getting the T7 to work.  You may not even need to do anything.

You can always try.  Just follow the instructions in the link I gave in my last post.  One of the steps is to save your current version of Allsky, in case you need to go back.

EricClaeys, 2022-05-31:
Closing Issue.  User is able to transfer files with scp.
Note: newer versions of Allsky eliminate the problem of decreasing file size by never uploading the same file more than once, so it can never be truncated.  And only one upload is done at a time, further decreasing the chances of a conflict.

closed 2022-05-31 by EricClaeys

koal01, 2022-10-11:
Eric, thanks for answering ! I imagine the hard job behind Allsky and the
huge efforts needed.
The T7 is not handled in the new version since i tested it carefully taking
requirements into account updating the firmware of the camera.
By chance i have two separate Rasps with the second running the 0.7 version
of Allsky, running nicely with the T7 camera
In the 0.83 version Allsky with the same camera remains in the status
"Allsky is starting up", sometimes the image comes after minutes and other
times it crashes with the red message "error too many time-out...".
As i said i don't want to bother you with my personal choice since i want
to use my T7 cameras as i spent some time to cool it  and i really like
Allsky project.
The 0.7 version suits me for the moment, have a look on my Allsky with a
double peltier coolling system, i can cool the camera to - 21 ° LOL

[image: IMG_9326.jpg]
[image: IMG_9327.jpg]

EricClaeys, 2022-10-11:
@koal01. when replying via email, GitHub deletes attachments, so I can't see your images.  You'll need to log into GitHub and attach them.
I spend a lot of time on Allsky, but am recently retired and enjoy working on it.  It makes me feel useful.
If you change the **Debug Level** to 4 and look in `/var/log/allsky.log` when you see the "Allsky is starting up" message, the log file will show what's happening.  Usually Allsky is deleting very dark or very light images.  But I suppose if you are getting ASI_ERROR_TIMEOUTs, it won't produce a valid picture.
Take a look at the Wiki page on ASI_ERROR_TIMEOUTs for some ways to (try and) fix it.  Those suggestions helped a lot of people, but unfortunately not everyone.