AllskyTeam/allsky #392: Question more than issue

khobar95, 2021-07-01:
Hello!

I have a Pi Zero and an HQ cam. I've been working my way through but am wondering what kind of performance I should expect? 

Currently, I have an ASI 120MM connected to a NUC running SharpCap and AllSkEye, and I generally capture 15s exposures as .png's with zero delay between. If I were to upgrade from the Zero to a Pi 4, could I expect something similar with allsky and the HQ cam?

Cheers and clear skies,

Tartouffe, 2021-08-06:
Running allsky with a 20MPx cam with a Pi 4B. No speed issues. CPU Load normally is at 3-4%, CPU power only is needed in the morning, when the timelapses etc. were created. A slower machine only needs more time; depending on the resolution (single file sizes) and quantity of image which has to be processed. In my option a minimum of 30s a picture by day is very enough for a very smooth time lapse. (and add some pause of 1-2s after easy image to give the system some gasp/break for clearing its cache and writing down the file with no stress.)

bbillp, 2021-08-06:
I’m interested in higher resolution, which camera is a 20mpx ?

Tartouffe, 2021-08-08:
E.g. any ZWO with higher resolution. In my case a ZWO ASI183MC (5496 x 3672 Pixel, 20,18 MP).
But thats not the point to crack, you have to find a wide field lens / fisheye which is able to handle the big(ger) sensor size and you have to find a lens, which is able to handle the high resolution above 5MPx. 
99% of the available fisheye lenses for C/CS-mount only can deal with at maximum half, or a quarter of the CCD size and maximum 5 MPx of image resulution.

EricClaeys, 2021-08-20:
khobar95,
Other people are using the Pi Zero with no problems.  It obviously won't be as fast as a Pi 4, but like Tartouffe said the only time lots of CPU is needed is when it goes from night to day and the timelapse, keogram, and startrails are created.  Creating 15 second exposures with no delay shouldn't be a problem for the Zero, and definitely won't be an issue for the Pi 4.

If this answers your question, please close the issue.
Thanks - Eric

linuxkidd, 2021-10-10:
@khobar95 hopefully the discussion has answered your questions.  I'll close this issue for now -- if you need more assistance or have other questions, don't hesitate to open another issue. :)

closed 2021-10-10 by linuxkidd