AllskyTeam/allsky #393: Timelapse display problem, keograms, startrail on website

flipper07, 2021-07-01:
Hello,
I come back again, because the links of Timelapse, keograms, startrail, of my site are not displayed.
They its fine to upload to the ftp server, but the links don't work.
I am not very good at http.
Maybe it comes from the orange supplier?
or the index.html file or the one in the videos folder (index.php)?
Thanks for helping me I'm starting to despair
fichier index.php /videos
<html>
	<head>
		<script src="https://ajax.googleapis.com/ajax/libs/jquery/2.2.0/jquery.min.js"></script>
        <script src="---%20Documents%20Persos%20Divers/--%20Astronomie/===%20Mon%20Projet%20ALLSKY/Sur%20le%20Serveur%20INTERNET/allsky/tmp/analyticstracking.js"></script>
        <link rel="stylesheet" href="https://maxcdn.bootstrapcdn.com/font-awesome/4.5.0/css/font-awesome.min.css">
		<link href="---%20Documents%20Persos%20Divers/--%20Astronomie/===%20Mon%20Projet%20ALLSKY/Sur%20le%20Serveur%20INTERNET/allsky/tmp/allsky.css" rel="stylesheet">
	</head>
	<body>
		<?php

		$files = array();
		if ($handle = opendir('.')) {

			while (false !== ($entry = readdir($handle))) {

				if (strpos($entry, 'mp4') !== false) {

					$files[] = $entry;
				}
			}

			closedir($handle);
		}

		asort($files);

		echo "<a class='back-button' href='..'><i class='fa fa-chevron-left'></i>Back to Live View</a>";
		
		
		echo "<div class=archived-videos>";

		foreach ($files as $file) {
			$year = substr($file, 7, 4);
			$month = substr($file, 11, 2);
			$day = substr($file, 13, 2);
			$date = $year.$month.$day;
			if (file_exists("thumbnails/".$date.".gif")){
				//echo "[image]";
				echo "<a href='./$file'><div class='day-container'><div class='image-container'>[image]</div><div>$year-$month-$day</div></div></a>";
			} else {
				echo "<a href='./$file'><div class='day-container'><div class='image-container'>[image]</div><div>$year-$month-$day</div></div></a>";
			}
			
		}
		echo "</div>";
		?>
		<script>
			function onImgEnter(img) {
				$(img).attr("src", "thumbnails/" + $(img).attr("id") + ".gif");              
			}
			function onImgLeave(img) {
				$(img).attr("src", "../aurora-snap.jpg");              
			}
		</script>
	</body>
</html>

[image]
[image]

AndreasLMeg, 2021-07-02:
Maybe  "Google Chrome Developer Tools Crash Course" could help
https://www.youtube.com/watch?v=x4q86IjJFag

flipper07, 2021-07-02:
thank you for the answer
But I don't see why it would come from google.
Maybe a server problem or something else not?
Here is the address of my site:
https://mescreations-r-b-07.pagesperso-orange.fr/allsky/index.html
I can access the videos by entering the address of the video:
https://mescreations-r-b-07.pagesperso-orange.fr/allsky/videos/allsky-20210614.mp4

AndreasLMeg, 2021-07-02:
google network tool is not the Problem, but can guide you.

[image]

I think the problem is in /videos/index.php (php supported ?)
But thats all I can say/write

SBRCoach, 2021-07-05:
Re-Salut

(SBRCoach ... tu m'avais contacté par mail)

Comme je te disais j'ai utilisé ton fichier "index.php" sur mon site et ça fonctionne à peu pres ... Donc je pense que ça vient d'ORANGE ton fournisseur d'accès qui doit bloquer certaines fonctions je pense

flipper07, 2021-07-06:
Oui je pense aussi
Je suis en train d’essayer de créer un site avec free. (j’attend. Création de votre espace en cours...)
Tu a quel fournisseur ? tu a crée ton site ? avec quel fournisseur ?
Je vais essayer aussi avec amazon aws. On peut aussi crée un site gratuitement je crois.
Merci encore

Provenance : Courrier pour Windows 10

De : SBRCoach
Envoyé le :mardi 6 juillet 2021 00:19
À : thomasjacquin/allsky
Cc : flipper07; Author
Objet :Re: [thomasjacquin/allsky] Timelapse display problem, keograms,startrail on website (#393)

Re-Salut
(SBRCoach ... tu m'avais contacté par mail)
Comme je te disais j'ai utilisé ton fichier "index.php" sur mon site et ça fonctionne à peu pres ... Donc je pense que ça vient d'ORANGE ton fournisseur d'accès qui doit bloquer certaines fonctions je pense

flipper07, 2021-07-06:
Je viens de voir ton site c’est bien celui la :
http://sbrcoach-allsky.lescigales.org/
Tu est bien sur free, d’après les videos.
Il y a pas mal d’infos par rapport a celui que j’ai.
Si tu peut me dire comment du a fait
Merci

Provenance : Courrier pour Windows 10

De : SBRCoach
Envoyé le :mardi 6 juillet 2021 00:19
À : thomasjacquin/allsky
Cc : flipper07; Author
Objet :Re: [thomasjacquin/allsky] Timelapse display problem, keograms,startrail on website (#393)

Re-Salut
(SBRCoach ... tu m'avais contacté par mail)
Comme je te disais j'ai utilisé ton fichier "index.php" sur mon site et ça fonctionne à peu pres ... Donc je pense que ça vient d'ORANGE ton fournisseur d'accès qui doit bloquer certaines fonctions je pense

SBRCoach, 2021-07-06:
Slt

 

En fait j’ai utilisé 2 fournisseurs …

 

Sur FREE j’ai toutes mes videos, images etc (j’ai dispo 10GO je crois) … mais le soucis chez eux VIRTUALSKY ne fonctionne pas (ce qui permet d’afficher la carte des etoiles sur to LIVE VIEW)

 

Du coup j’ai bidouillé – J’ai installé les données qui permettent d’afficher la page principale INDEX.HTML sur un autre serveur (lescygales.org … chez eux, VIRTUALSKY fonctionne. Mais J’ai pas pu tout faire chez eux car on a qu’un TRES FAIBLE espace disque)  et ensuite j’ai modifié la REDIRECTION des boutons chez FREE ou est hébergé le reste

En attendant de trouver une autre solution … payer … ou installer ma page sur mon NAS

 

A voir !!!

flipper07, 2021-07-10:
Bonsoir, ou bonjour..
Je viens de voir chez orange,  donc apparemment il ne gère pas le PHP
Par contre j'ai vu que dans la nouvelle version allsky on parle de Amazon S3.
Ne pourrai-je pas passer par eux?
On peut faire un site avec amazon?
chez Free suis toujours en attente de création de page perso ils sont pas rapide...

SBRCoach, 2021-07-10:
Ils sont vraiment NULS chez orange ...

Je ne connais pas AMAZON S3

Pour **FREE** il me semble que j'avais reçu les infos de connection par la poste

Sinon en attendant tu peux essayer de **l'héberger sur ton RaspBerry** - Thomas a mis en place cette possibilité tres recement pour ceux qui n'ont pas d'herbergeur de pages web ... pas encore essayé non plus !!!

flipper07, 2021-07-10:
Oui comme tu dit nul chez orange.
Je vais essaye sur le raspberry je de dit ça
Merci

flipper07, 2021-07-10:
Voila c'est fait.
J'ai bien installer le serveur web sur rpI.
Je sais pas si je ne comprend pas ou je alors je suis bette peut être lol..
Mais comment accéder au site a l'extérieur de cher moi. Ou donnes l'adresse du lien a des amis....
Peut être avec mon adresse ip de mon fournisseur (orange.... lol)  j'ai un domaine noip.com
J'ai regarder sur le site mais j'ai riens vu...
Merci encore

SBRCoach, 2021-07-11:
https://raspberry-pi.fr/mettre-en-ligne-serveur-web-raspbian-dydns-port-forwarding/
https://knowledge.parcours-performance.com/acceder-de-lexterieur-flux-http-dun-raspberry-pi/

flipper07, 2021-07-11:
Merci pour les sites
Mais c'est ce que j'ai déjàs fait.
Mais le problème c'est que que j'ai plusieurs rpi connecter sur ma box (livebox 5 sagem Fible).
J'ai un pour mon imprimante 3D (Octoprint) qui est sur les memes ports.et un autre pour (domotics) et quand je dit a un nouveaux port il me dit que la règle est en conflit.
Peut être un truc que je ne fait pas comme il faut?
Normalement con on veut accéder sur le rpi en extérieur ce que je fait avec mon imprimante 3d je tape l'adresse (ddns.org ou alors directement l'adresse ip de ma box) et puis le no de port, mais comment fait car normalement pour pouvoir acceder au rpi on doit mètre un port diffèrent non?
Merci encore
[image]

flipper07, 2021-07-11:
Bon j'avance un peut
J'arrive a me connecter depuis le reseau exterieur avec un adresse noip.
Mais par contre avec mon nom d'utilisateur et mot de passe Mais si je donne mon adresse avec mes identifiants  les personnes peuve ce connecter et changer le mot de passe.....
C'est bizare je e comprend pas
Le problème aussi c'est avec mes autres raspberry....

SBRCoach, 2021-07-12:
Bizarre ... tu as un lien pour voir ?

Tu partages la page Internet ou la page de configuration de l'application en interne ?

flipper07, 2021-07-12:
Je partage la page de l'application en interne.
C'est la ou je comprend pas, comment partager le lien de la page internet?? qui est dans le raspebery?
ya encore un truc qui ne va pas, les images et videos ne se mette pas dans les dossiers respectivement
Voici le fichier modifier pour être sur le serveur du raspberry
ftp-settings.sh
#!/bin/bash

# FTP/SFTP/S3 settings
PROTOCOL='local'
IMGDIR='/var/www/html/allsky/'
MP4DIR='/var/www/html/allsky/videos`
KEOGRAM_DIR=`/var/www/html/allsky/keograms`
STARTRAILS_DIR=`/var/www/html/allsky/startrails`

[image]
ici les images videos... qui son bien enregistre sur le dossier de allsky
[image]
J'espere que je m'explique bien (en tout ca, la je ne comprend pas tout le fonctionnement, je suis un peut perdu)

flipper07, 2021-07-19:
good evening I still have a problem with my site Tell you about amazon S3, what should I do? Thank you I despair

flipper07, 2021-07-29:
Bonsoir Je suis arriver enfin a faire fonctionne mon site avec free, voici le lien.
http://monsitepersobruno.free.fr/
j'aimerai bien rajouter un lien pour ajouter les photos de mon montage

SBRCoach, 2021-07-29:
Cool ça fonctionne 

 

Bravo et félicitations

 

… Moi j’ai eu une mauvaise nouvelle avec FREE … J’ai plus acces à mon site chez eux … j’ai perdu plisuiers journées du coup (sauvegarde trop ancienne) fait chier

 

En attendant je l’ai mise ailleurs, mais j’essai de tout mettre sur mon NAS en interne pour etre tranquille … Par contre j’ai quelques images qui s’affichent pas sans comprendre pourquoi

 

Par contre chez FREE, la superposition de la carte du ciel dans le LIVE VIEW ne fonctionne pas

 

Fait voir ton montage !!!

 

http://monsitepersobruno.free.fr/

flipper07, 2021-07-30:
Salut voici mon montage:
J'ai mis ce petit panneau solaire pour alimenter le ventilateur, je vais essayer de branche le raspberry mais je sais pas si ca va tenir la nuit sur la batterie, actuellement ça fonctionne toute la nuit avec le ventilateur.
Voice quelque photo la photo avec la condensation avec les résistance c'était avant de mètre le ventilateur, maintenant avec le ventilateur ca va mieux, cette nuit il a fait un orage et ce matin, a part a l'extérieur, pas de condensation.
Je vais changer mon boiter et mettre un boitier plexo car le scotch que j'ai mis c'est pas le top.
La dernière c'est le derrière du panneau solaire.
Par contre ton site est bien fait tu la bien améliore (comment a tu fait?) et pour les date comment tu a fait pour le metre en francais c'est a dire comme aujourd'hui le jour,mois,année, j'ai essayer de regarder dans le fichier capture.ccp, d'après thomas jacquin, mais ca ne fonctionne pas.
Désoler pour toi j'espère que tu va trouver une solution.
[image]
[image]
[image]
[image]
[image]
[image]
[image]

flipper07, 2021-07-30:
J'ai oublier le socle noir c'est moi qui l'ai imprimer en 3D pour tenir la camera, mais avec la chaleur il se déforme..

flipper07, 2021-07-30:
Bon, c'est trop beau, encore un souci.
Depuis hier j'ai la même image qui appareil en live sur panneau d'administration et sur le site.
Voir mon site:
http://monsitepersobruno.free.fr/
Donc plus de direct? bizarre non?
Je l'ai redémarre, arrête, débrancher, pareil.
Comment voir si la camera fonctionne bien?
Je l'ai brancher sur mon pc elle fonctionne bien.

SBRCoach, 2021-07-30:
Slt
Oui j'ai fait quelques ameliorations ... Pour cela il faut coder un peu ...

**### [image]** 

1/ Fait une sauvegarde de ton fichier **CAPTURE.CPP** en cas de mauvaise manip avant toute modif (il est ds le dossier : /home/pi/allsky)

2/ Modifier dans le code suivant ( Procedure **char *getTime()**) les parametres : 

"%Y%m%d %H:%M:%S" et les remplacer par **"%d.%m.%Y %H:%M:%S"** 

    strftime(TimeString, 80, "%Y%m%d %H:%M:%S", localtime(&curTime.tv_sec));
 
3/ Puis recompiler à partir de la console WINDOWS TERMINAL, le code avec la commande : **make capture**

4/ arrêter le service « **STOP ALLSKY** » puis le relancer « **START ALLSKY** »

flipper07, 2021-07-30:
Ok super ton install.
Bon ça y est je suis aller chercher mon boitier vais démonter et mètre tous ça dedans.
C'est quoi le boitier recouvert en aluminium?

SBRCoach, 2021-07-30:
> 
> 
> Ok super ton install.
> Bon ça y est je suis aller chercher mon boitier vais démonter et mètre tous ça dedans.
> C'est quoi le boitier recouvert en aluminium?

de quel Quel boitier parles tu ?

flipper07, 2021-07-30:
celui sur les photo les 2 derniere

SBRCoach, 2021-07-30:
> 
> 
> celui sur les photo les 2 derniere

C'est la coiffe pour proteger le boitier quand il fait vraiment trop chaud  (c'est un simple pare soleil de voiture)

flipper07, 2021-07-30:
Tes 2 1ere photos c'est la même camera? car je comprend pas tu a fixer ton asi 224 dans le boitier et celle qui est fixer au couvercle c'est pas la même??

flipper07, 2021-07-30:
A ok pour le pare soleil

SBRCoach, 2021-07-30:
Si si c'est la meme ... c'est juste la photo avec le boitier fermé

flipper07, 2021-07-30:
super merci.
Je sais pas trop je vais peu être fixe ma camera sur le couvercle.
Le rpi tu la mis ou?

SBRCoach, 2021-07-30:
Le RPI est dans mon sous sol ... d'ou le passage de cables

J'étais pas chaud pour le laisser à demeure dans le boitier ( froid, chaud, ecarts de temperatures .... humidité .... pas top pour le RPI et la clé USB je trouve) et pas accessible en cas de soucis

flipper07, 2021-07-30:
ok pour RPI je viens de voir sur une autre photo, tu la mis a part.
Moi, je pense mètre tous dans le même boitier, quand pense tu?
Sinon le RPI dans le tube PVC..
Je lit ta réponse en même temps.
Pour le dôme j'ai vu que tu a mis un (joint noir ou je sais pas trop).
Sinon je pense en faire un en 3D je peut aussi faire en 3D en poliflex (c'est du filament flexible).
A essayer.
c'est quoi qui y a autour de ta camera (des résistances chauffante?).
Voici une photo de mon matos, j'ai tout démonter.
Bon je sais pas encore comment je vais fixe le boitier sur le PVC...
[image]

flipper07, 2021-07-30:
Pour la ventilation j'ai prévu, des petite grille ronde

flipper07, 2021-07-30:
[image]
J'ai aussi sorti mon artillerie pour la ventilation (module peltier) je sais pas si tu connais (d'un cote fait du froid et de l'autre du chaud.(c'est ce qu'il y a dans les glacières électrique). c'est pour la condensation
Je vais cogiter un peut tout ça....

SBRCoach, 2021-07-30:
Sur ton support noir, je ferais des trous d'aeration pour permettre a l'air de circuler un peu ... Je voulais le faire en blanc pour renvoyer la chaleur mais peur de trop de reflets sur le dome.

Un ou 2 sachets de silican gel dans le boitier pour absorber l'humidité

Le joint pour le dome est indispensable et eventuellement rajouter du silicone

Oui autour c'est une resistance chauffante achetée 

J'avais pensé faire une grille d'aération pour le ventilo , mais ici avec l'été pourri trop d'humidité qui serait rentrée ... ou l'idée de faire un boitier été (avec grille d'aération) et un boitier hiver : SANS !!!

J'avais essayé un VENTIRAD à l'interieur mais j'ai juste testé 1 apres midi tres chaud, j'avais carrément de la flotte au fond du boitier ... du coup enlevé rapidement

flipper07, 2021-07-30:
[image]
Ca c'est la camera que j'avais pour mon télescope que j'ai revendu.
Super les photos merci ça va m'aider.
Oui ce que j'allais dire pour le support en 3D pas mal.
Pour les saches de gel silician j'en ai, j'essayerai
Je vais voir si je peut faire quelque chose pour le fixe sur mon PVC.
Je vais refaire mon support noir du dôme, je vais modifier celui que j'ai imprimer et peut être en faire un autre en poliflex pour faire le joint.
Mais avec la chaleur je sais pas ce que ça va donner.
LE problème c'est que j'aimerais pourvoir le démonter facilement (sans mettre de joint en silicone) pour pourvoir y accède facilement en ca de condensation....
Pas facile de prendre en compte tous cet petits inconvénient.....

SBRCoach, 2021-07-30:
J'avais fixé dans la 1ere version la camera sur le dome mais pour y acceder il fallait que je devisse tout le dôme . là en là fixant DANS LE BOITIER et non le couvercle, c'est rapide d'acces et pas besoin d'enlever le dôme et le joint à chaque fois...

Effectivement ça à l'air d'un fisheye, apres y'en a differentes sortes et de # degrés ... Moi c'est un 180° pour avoir un "plus grand ciel" d'acquisition

EricClaeys, 2021-08-09:
If this issues is no longer a problem, please close it.
Thanks!