AllskyTeam/allsky #395: Preview issue on arm64 system

laheller, 2021-07-10:
@tjacquin @vidma @dbainbridge @agnunez @ckuethe 

Hi!

I am using **Ubuntu Mate 20.04.2 LTS (Focal Fossa) 64-bit** installed on **Raspberry PI 4B**, 8GB model.

By extending Makefile with the below section I was able to compile and install allsky:
```
ifeq ($(platform), aarch64)
	OPENCV = $(shell pkg-config --cflags opencv4) $(shell pkg-config --libs opencv4)
	CC = g++
	AR= ar
	CFLAGS += -march=native -mtune=generic
	DEFS += -DOPENCV_C_HEADERS
	ZWOSDK = -Llib/armv8 -I./include
endif

```
However when I tried to run it via commandline in "preview" mode, I got following error:
```
laheller@laheller-desktop:~/Downloads/allsky$ ./allsky.sh preview

Making sure allsky.sh is not already running...
can't get debug descriptor: Resource temporarily unavailable
Trying to automatically choose between ZWO and RPI camera
ZWO camera found. Choosing ZWO
Settings check done
CAMERA: ZWO
CAMERA_SETTINGS: /home/laheller/Downloads/allsky/settings_ZWO.json
Starting allsky camera...

 ******************************************
 *** Allsky Camera Software v0.7 | 2020 ***
 ******************************************

Capture images of the sky with a Raspberry Pi and an ASI Camera

Add -h or -help for available options 

Author: Thomas Jacquin - <jacquin.thomas@gmail.com>

Contributors:
-Knut Olav Klo
-Daniel Johnsen
-Yang and Sam from ZWO
-Robert Wagner
-Michael J. Kidd - <linuxkidd@gmail.com>
-Chris Kuethe

Listing Attached Cameras:
- 0 ZWO ASI120MC-S

ZWO ASI120MC-S Information:
- Resolution:1280x960
- Pixel Size: 3.8μm
- Supported Bin: 1 2 
- Color Camera: bayer pattern:GR
- SDK version 1, 19, 0, 0
- Initialise Camera OK
- Sensor temperature:29.000000

Capture Settings: 
 Image Type: ASI_IMG_RGB24
 Resolution: 1280x960 
 Quality: 95 
 Exposure: 200ms
 Max Exposure: 20000ms
 Auto Exposure: 1
 Gain: 50
 Max Gain: 200
 Cooler Enabled: 0
 Target Temperature: 0C
 Auto Gain: 0
 Brightness: 50
 Gamma: 50
 WB Red: 53
 WB Blue: 90
 Binning: 1
 Delay: 10ms
 Daytime Delay: 5000ms
 USB Speed: 40
 Text Overlay: Hello
 Text Extra Filename: 
 Text Extra Filename Age: 0
 Text Line Height 30px
 Text Position: 15px left, 30px top
 Font Name:  0
 Font Color: 255 , 255, 255
 Small Font Color: 0 , 0, 255
 Font Line Type: 16
 Font Size: 7.0
 Font Line: 1
 Outline Font : 0
 Flip Image: 0
 Filename: images/image.jpg
 Latitude: 48.06N
 Longitude: 17.36E
 Sun Elevation: -6
 Preview: 1
 Time: 0
 Darkframe: 0
 Show Details: 1
 Debug Level: 0

Starting daytime capture
Saving auto exposed images every 5000 ms

Press Ctrl+C to stop

terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.2.0) ../modules/core/src/array.cpp:2492: error: (-206:Bad flag (parameter or structure field)) Unrecognized or unsupported array type in function 'cvGetMat'

./allsky.sh: line 86: 10495 Aborted                 (core dumped) $ALLSKY_HOME/capture $ARGUMENTS

```
Not sure, whether this is a serious issue but just to be sure I am reporting it.

BR,

Ladislav

ckuethe, 2021-09-12:
Well, you compiled it right, but run my allsky camera headless and just use the web UI.

laheller, 2021-10-17:
@EricClaeys 

Update: Tried it using my RPI connected **directly** to display via HDMI cable.
The result is the same, command `./allsky.sh preview` gives an error.
From the error message I don't think that the problem is with the remote desktop because it clearly says that it's an OpenCV4 issue.
[image]

linuxkidd, 2021-10-23:
This is likely due to PR #507 ... 

> NOTE: I had quite a bit of issue getting the Display() function to compile, but finally got something workable -- although, I've not tested the Display capability. I would love for someone to test the Display code prior to merge. Otherwise, the rest of the code compiles fine on both Raspberry Pi OS based on Debian Buster with OpenCV 3.2.0, as well as Debian Bullseye with OpenCV 4.5.1.

Sounds like there are some changes needed to the `Display()` function..

EricClaeys, 2021-10-30:
@laheller, you should set Locale in the WebUI, near the bottom of the page.  It's used by more than just darkCapture.sh.

commit d9b0be46 (2021-11-01): capture.cpp: fix preview

Fixes #395
closed 2021-11-01 by linuxkidd