AllskyTeam/allsky #398: ‘DARK IMAGE’ Generation

bbillp, 2021-07-17:
ALLSKY updated 7/4/2021 Raspberry Pi 3b+

Problem: Even after making a new darks folder and rebooting why is the dark image sequence number starting at 38.jpg then at 39.jpg it stops incrementing and just replaces 39.jpg file every 5-10 minutes?

I repeated the above 3 times with the same results, the DARK image sequence number starts at 38.jpg
Bill Prats

EricClaeys, 2021-07-23:
Bill,
The number in the file name is the temperature of the sensor, not an image sequence number.  Allsky will create a new file for every degree C change in temp.

EricClaeys, 2021-08-01:
Bill, did this answer your question?

bbillp, 2021-08-01:
Thank you, I read thru the scripts and found the answer in the createdarks shell then tested the routine by placing the warm camera outside a few hours to watch the temperature drop and the dark image name change respectfully.   After the camera temperature stabilized I reset the json file ‘darks’ to zero and observed a beautiful mp4 the following morning and each day since then.  Should I capture DARKS for a full night to get a full range of temperature files.      

ZWO ASI178MC here, I tried a new ASI120MC-S and the images do not compare, the 178 is so much better.    Thanks for responding .    
Bill P. in Southern California

bbillp, 2021-08-01:
I forgot to ask but where would I find an explanation for using DARKS, I just experimented on my own to figure it out after updating my years old copy of allsky and seeing DARKS in the configure.sh and ZWO.json scripts.   
Bill

EricClaeys, 2021-08-02:
Bill,

The ~/allsky/README.md file mentions darks and how to take them.  It initially took me a while, but once I figured it out, it's actually pretty easy.  Don't forget to cover the camera - I forgot once and the dark frames looked like a normal dark shot.  Ooops.

My camera is on a 17 foot pole and it's scary to climb that high, so when I recently took it off to fix the fan, I left it in the observatory overnight covered, and took darks all night and got 26 - 33 degrees.  Is it really important to get lots of shots?  Who knows - you'd really have to test.  It's probably more important to have several warm-night frames than winter frames, since dark current is higher when it's hot out.

If this issue is resolved, would you please click on the "Close issue" button at the bottom (or whatever it's called)?  Thanks.

Say, I also have the ASI178MC.  My night exposures are 60 seconds with 250 gain, and they are still darker than I'd like.  I'm in Bortle 2.  What's your exposure?

Eric

closed 2021-08-02 by bbillp

bbillp, 2021-08-02:
The ASI178 auto exposure is 20-23 seconds at midnight with fixed gain = 185 in light polluted Huntington Beach California.  I’m happy with the sharp JPG images and the MP4 video.  I was using my own script to create 800x600 AVI and 1280x800 AVI videos but I can’t see that they are better than the MP4 video even though AVI output is very large.   

Look at a push up fiberglass mast for ham radio , MAX-GAIN Systems,   mgs4u.com .  Its possible to purchase the large round sections to satisfy your requirements.  When camping I put the ALLSKY on the roof of the motor home powered by a 15 amp hour Bioenne 12V battery driving a ebay dual 3 amp USB power converter.  The battery sans charging will run the system for 2 days.      https://www.bioennopower.com/products/copy-of-12v-15ah-lfp-battery-pvc-blf-1215w       I hope this helps ….. Bill k6acj