AllskyTeam/allsky #400: maybe real <facepalm> question: Cannot access allsky gui on chromium: proxy issues?

GandaGit, 2021-07-28:
after the all sky install and gui, i canot access the gui.
http://allskipires.local/
http://<local ip address>

the public interface also does not work from another computer:
https://192.168.0.212/public.php

this seems to be very basic. can anyone help?

maserowik, 2021-07-28:
Did the ip address change for your pi?

 

http://allskipires.local/
http://

https://192.168.0.212/public.php

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

thomasjacquin, 2021-07-28:
Try http instead of https. There's no certificate installed.

GandaGit, 2021-07-28:
Jul 28 17:01:11 allskipires allsky.sh[2452]: Trying to automatically choose between ZWO a
Jul 28 17:01:11 allskipires allsky.sh[2452]: None of RPI or ZWO Cameras were found. Exiti
Jul 28 17:01:12 allskipires sudo[2476]:     root : unable to resolve host allskipires
Jul 28 17:01:12 allskipires sudo[2476]:     root : problem with defaults entries ; TTY=un
Jul 28 17:01:12 allskipires allsky.sh[2452]: sudo: unable to resolve host allskipires
Jul 28 17:01:12 allskipires sudo[2476]:     root : TTY=unknown ; PWD=/ ; USER=root ; COMM
Jul 28 17:01:12 allskipires sudo[2476]: pam_unix(sudo:session): session opened for user r
Jul 28 17:01:12 allskipires systemd[1]: Stopping All Sky Camera...
Jul 28 17:01:12 allskipires sudo[2476]: pam_unix(sudo:session): session closed for user r
Jul 28 17:01:12 allskipires systemd[1]: Stopped All Sky Camera.
~

> 
> 
> Try http instead of https. There's no certificate installed.

thanks. done that. did not work.

GandaGit, 2021-07-28:
> http://allskipires.local/

thanks. I've checked that and its the same one. 
also the http://allskipires.local/ does not work-

GandaGit, 2021-07-28:
> 
> 
> Try http instead of https. There's no certificate installed.

Thanks. did that but did not work.

GandaGit, 2021-07-28:
Just did a status check and I have several issues that I need to address.
Maybe sort out why there are no cameras detected
also: why is the below?
_root : unable to resolve host allskipires_

Jul 28 17:01:11 allskipires allsky.sh[2452]: Trying to automatically choose between ZWO and RPI camera
Jul 28 17:01:11 allskipires allsky.sh[2452]: None of RPI or ZWO Cameras were found. Exiting.
Jul 28 17:01:12 allskipires sudo[2476]:     root : unable to resolve host allskipires
Jul 28 17:01:12 allskipires sudo[2476]:     root : problem with defaults entries ; TTY=unknown ; PWD=/ ; USER=root ;
Jul 28 17:01:12 allskipires allsky.sh[2452]: sudo: unable to resolve host allskipires
Jul 28 17:01:12 allskipires sudo[2476]:     root : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/bin/systemctl stop allsky
Jul 28 17:01:12 allskipires sudo[2476]: pam_unix(sudo:session): session opened for user root by (uid=0)
Jul 28 17:01:12 allskipires systemd[1]: Stopping All Sky Camera...
Jul 28 17:01:12 allskipires sudo[2476]: pam_unix(sudo:session): session closed for user root
Jul 28 17:01:12 allskipires systemd[1]: Stopped All Sky Camera.

maserowik, 2021-07-28:
Verify your host name is correct in raspi-confit

Michael A. Serowik

GandaGit, 2021-07-29:
> 
> 
> Yes, fix your camera problem first. It could be a USB cable issue, a power supply issue or just a raspberry pi configuration (need to enable the camera interface if using the RPiHQ camera).

did that. ribbon cable not fully inserted. sorted out.
now looks much better but still no gui:
pi@allsky:~/allsky $ sudo service allsky status
sudo: unable to resolve host allsky
● allsky.service - All Sky Camera
   Loaded: loaded (/lib/systemd/system/allsky.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2021-07-28 22:41:22 BST; 3h 3min ago
 Main PID: 465 (allsky.sh)
   CGroup: /system.slice/allsky.service
           ├─  465 /bin/bash /home/pi/allsky/allsky.sh
           ├─  669 /home/pi/allsky/capture_RPiHQ -angle -9 -autoexposure 0 -autogain 0 -awb 0 -background 0 -bin 1 -brightness 50 -daytimeDelay 30000 -delay
           ├─12948 sh -c nice raspistill --nopreview --thumb none --output image.jpg --burst -st --mode 3 --exposure off --shutter 60000000 --analoggain 4 -
           └─12950 raspistill --nopreview --thumb none --output image.jpg --burst -st --mode 3 --exposure off --shutter 60000000 --analoggain 4 --awb off --

Jul 29 01:39:42 allsky allsky.sh[465]: Capture command: nice raspistill --nopreview --thumb none --output image.jpg --burst -st --mode 3 --exposure off --sh
Jul 29 01:39:42 allsky allsky.sh[465]: Capturing & saving image done, now wait 0 seconds...
Jul 29 01:39:42 allsky allsky.sh[465]: Determine if it is day or night using variables: desired sun declination angle: -9 degrees, latitude: 52.57N, longitu
Jul 29 01:39:42 allsky allsky.sh[465]: Check for day or night: NIGHT (waiting for changing NIGHT into DAY)...
Jul 29 01:39:42 allsky allsky.sh[465]: Capturing & saving image...
Jul 29 01:39:42 allsky allsky.sh[465]: Capture command: nice raspistill --nopreview --thumb none --output image.jpg --burst -st --mode 3 --exposure off --sh
Jul 29 01:39:42 allsky allsky.sh[465]: Capturing & saving image done, now wait 0 seconds...
Jul 29 01:39:42 allsky allsky.sh[465]: Determine if it is day or night using variables: desired sun declination angle: -9 degrees, latitude: 52.57N, longitu
Jul 29 01:39:42 allsky allsky.sh[465]: Check for day or night: NIGHT (waiting for changing NIGHT into DAY)...
Jul 29 01:39:42 allsky allsky.sh[465]: Capturing & saving image...
lines 1-20/20 (END)

thomasjacquin, 2021-07-29:
Does it work with either 

- http://192.168.0.212
- http://localhost (from the RPi)

The `.local` doesn't always work out of the box, it depends what device and OS you're using to connect to it. It's part of your zeroconf configuration. It should be active by default if using a mac (the service is called Bonjour). It may be active in Linux. I'm not sure about Windows.

GandaGit, 2021-07-29:
HI. does not work on either.
- locally:
- http://localhost
-
- http://localhost.local
-
- http://allsky <http://localhost>
- http://allsk <http://localhost>y.local
- remotely (I've checked and the ip is 129.168.0.212 by dhcp from my router)
- http://allsky
-
- http://192.168.0.212
amd I've tried with Https too.
- I'm lost now...
-
-
- http://192.168.0.212
- http://localhost (from the R

EricClaeys, 2021-08-05:
GrandaGit,
What error message do you receive?
Is lighttpd installed (it's the web server allsky uses)?  If so, look in /var/log/lighttpd/error.log to see if it gives a clue.
Eric

EricClaeys, 2022-05-31:
Closing Issue - no response from user in 9 months, despite 2 attempts.

closed 2022-05-31 by EricClaeys