AllskyTeam/allsky #401: External text file help

Jonk2, 2021-07-29:
I've been playing with the external text file option and there are 3 things puzzling me.

1. How do I have multiple lines, do I need to use a special character for a new line? Currently I have 'Test text 1' new line then 'test text 2'. These do show up as 2 separate lines of text, but the first line has a ? on the end. 
2. Is there a simple way of using external weather data from a website as a text file and linking to that, or maybe saving it locally with an automatic formatting to suit? Most websites I've quickly looked at have xml or json files.
3. How do I position this external text and change the font, size etc. in isolation from the main displayed text?

I'm not a prgrammer, just a user who likes to modify things on a basic level.

[image]

EricClaeys, 2021-08-01:
Jonk2,
Try printing a debug line of the text, then look in the log file (usually /var/log/allsky.log) to see if the "?" is really a question mark or something else.  I suspect it's an unprintable character that's getting translated into a "?" by the function that adds text to the image.
In ~/allsky/capture.cpp, look for the line that starts with:
       cvText(pRgb, line, iTextX, 

It should be around line 996.  After that line, add these lines (make sure you use the exact upper/lower case):
    sprintf(textBuffer, "line='%s'\n", line);
    displayDebugText(textBuffer, 0);

then save the file, and run "make allsky" in the same directory, and restart the allsky service.
When every image is saved, you should see two lines in the log file that look like:
    line='Text test 1X'
    line='text test 2'

You're interested in what comes after the "1" (i.e., the "X" in my example).
Eric

Jonk2, 2021-08-01:
Hi Eric, thanks for the reply.
I've done exactly as suggested, so capture.cpp now looks like this:

```
cvText(pRgb, line, iTextX, iTextY + (iYOffset / bin), fontsize * 0.08, linewidth, linetype[linenumber], fontname[fontnumber], smallFontcolor, Image_type, outlinefont);
sprintf(textBuffer, "line='%s'\n", line);
displayDebugText(textBuffer, 0);
iYOffset += iTextLineHeight;
```

When running 'make allsky', it reports 'Nothing to be done for allsky'.

[image]

So I've restarted the service anyway and checked the log file and I see this:

`Aug  1 22:36:17 allsky allsky.sh[2553]: can't get debug descriptor: Resource temporarily unavailable`

I can follow your instructions, but I don't know much about programming so I can't go any further myself.

Any ideas?

EricClaeys, 2021-08-02:
Jonk2,
Sometimes the "make" program doesn't realize the file changed.  It's supposed to say, "is 'capture.cpp' newer than 'capture'?  If so, then recompile it.".
Rename the current "capture" program to something like "capture-original".  Then try the "make capture" again.
Don't worry about that message in the log file; it's harmless.

Eric

Jonk2, 2021-08-04:
What I did, was open the capture.cpp added the lines, then saved as a separate file - renamed the original capture.cpp and uploaded the revised capture.cpp.
Running make allsky resulted as above, so I don't know what else to do, I've tried it a couple of times.

EricClaeys, 2021-08-04:
Jon,
Did you do a "make allsky" or "make capture"?  You need to do "make capture".
Eric

Jonk2, 2021-08-05:
I haven't tried 'make capture' but I will now. Please bear with me, I'm not a Linux or software guy, just a very nosey user!

I'll get back to you.

Jonk2, 2021-08-05:
Ok, I get the following:
[image]

Jonk2, 2021-08-05:
Ignore that, I didn't copy the new 'capture.cpp'....see I told you I was a newbie. Now I get this, is this correct?

[image]

And the 'capture' file has a new timestamp:

[image]

EricClaeys, 2021-08-05:
Jon, that's correct.  It just created a new "capture" command.  Now you need to restart the allsky service however you normally do it.  Let me know what the log file shows.
Eric

Jonk2, 2021-08-05:
Ok, restarted and waiting for half an hour before sunset before any images are saved. I'll post the relevant log sections tonight (in about 4 hours).

Jonk2, 2021-08-05:
Here's an extract from the log:

> Aug  5 19:36:09 allsky allsky.sh[22257]: Auto Exposure value: 23 ms
> Aug  5 19:36:09 allsky allsky.sh[22257]: line='Text test 1#015'
> Aug  5 19:36:09 allsky allsky.sh[22257]: line='text test 2'
> Aug  5 19:36:09 allsky allsky.sh[22257]: Exposure value: 55963 Âµs
> Aug  5 19:36:09 allsky allsky.sh[22257]: 20210805 19:33:08
> Aug  5 19:36:09 allsky allsky.sh[22257]: Auto Exposure value: 55 ms
> Aug  5 19:36:09 allsky allsky.sh[22257]: line='Text test 1#015'
> Aug  5 19:36:09 allsky allsky.sh[22257]: line='text test 2'
> Aug  5 19:36:09 allsky allsky.sh[22257]: Exposure value: 29136 Âµs
> Aug  5 19:36:09 allsky allsky.sh[22257]: 20210805 19:34:08
> Aug  5 19:36:09 allsky allsky.sh[22257]: Auto Exposure value: 29 ms
> Aug  5 19:36:09 allsky allsky.sh[22257]: line='Text test 1#015'
> Aug  5 19:36:09 allsky allsky.sh[22257]: line='text test 2'
> Aug  5 19:36:09 allsky allsky.sh[22257]: Exposure value: 103039 Âµs
> Aug  5 19:36:09 allsky allsky.sh[22257]: 20210805 19:35:09
> Aug  5 19:36:09 allsky allsky.sh[22257]: Auto Exposure value: 103 ms
> Aug  5 19:36:09 allsky allsky.sh[22257]: line='Resizing
> Aug  5 19:36:10 allsky allsky.sh[22257]: Uploading
> Aug  5 19:37:09 allsky allsky.sh[22257]: Resizing
> Aug  5 19:37:10 allsky allsky.sh[22257]: Uploading
> Aug  5 19:38:09 allsky allsky.sh[22257]: Resizing
> Aug  5 19:38:10 allsky allsky.sh[22257]: Uploading
> Aug  5 19:39:09 allsky allsky.sh[22257]: Resizing
> Aug  5 19:39:10 allsky allsky.sh[22257]: Uploading
> Aug  5 19:40:10 allsky allsky.sh[22257]: Resizing
> Aug  5 19:40:10 allsky allsky.sh[22257]: Uploading
> Aug  5 19:41:10 allsky allsky.sh[22257]: Resizing
> Aug  5 19:41:11 allsky allsky.sh[22257]: Uploading
> Aug  5 19:42:10 allsky allsky.sh[22257]: Resizing
> Aug  5 19:42:11 allsky allsky.sh[22257]: Uploading

You can see a few issues. The ? seems to be caused by #015, but all I did was type the first line of text, press enter then type the second line. I even added a single '?' but 2 were shown. I'm sure this #015 means something to you.

You can also see exposure value when expressed in µs shows as Âµs in the log. Probably a bug. My suggestion, if exposure is <1ms, show it in µs, >1ms but <1s, show in ms, >1s, show in s, obviously if possible.

The way it dumps a lot of information into the log you already appear to know about, but you can plainly see it here.

I'll leave it running as is, but if you need any futher information from my particular setup, please let me know.

Jonk2, 2021-08-05:
The forum formatting has changed the text - where it says #15, it should be # 015 (without a space):

[image]

EricClaeys, 2021-08-06:
Jon,
This is what I was looking for.  15 is the ASCII Carrage Return character.  Linux uses Line Feed to signal the end of a line in a text file, but Windows uses Carrage Return and Line Feed.  I will fix this by looking for both characters. 

FYI, if you afe old enough to remember manual typewriters, a Line Feed simply went to the next line but didn't move the head.  And Carrage Return moved the head to the left, but didn't go to the next line.  On electric typewriters and computers, the Return key does both.

Eric

Jonk2, 2021-08-06:
I do remember early typewriters, just! Good to see if it can be fixed.

One of the reasons I chose this software over a couple of others (I've tested a lot) was the fact it runs on the rpi which is by far so much more reliable than a small windows pc.

I'm very keen to see this software develop and am willing to help where possible.

EricClaeys, 2021-08-09:
Once I figure out how to update GitHub, I'll submit the fix officially, but in the meantime, this version of capture.cpp should remove the question mark.  Copy it to ~/allsky then run "make capture" and restart allsky.

Eric
[capture.zip](https://github.com/thomasjacquin/allsky/files/6957794/capture.zip)

Jonk2, 2021-08-10:
Ok great, I'll add it during the day as it's saving night time images right now. I'll report back when I can.

Now I just need to find a way to display some meaningful information!

EricClaeys, 2021-08-20:
Jon,
A version of capture.cpp has been uploaded that fixes this problem.  Would you please install it and if the problem is resolved, close this issue?  Thanks.

Jonk2, 2021-08-23:
Eric, I tried the capture.cpp from both your attached zip file and the main website. The files differ in size quite a bit (zip version 112kb, main website version 817kb).

Uploading the website version onto the PI and running make capture does not work, there are literally hundreds of lines of errors.

So I tried the attached zip version and it seemed to work.

Running a fairly fresh installation of 0.8 on a 2nd PI (I'm not touching the allsky outside until I'm confident I can without upsetting it), the carriage return issue seems to have been resolved:

[image]

However, now the software restarts after 2 exposures, in a loop. I don't know if this is due to the capture.cpp that you attached, or something I've (probably) done. It won't expose below 1.8s and the log isn't very helpful:

As this thread is about a specific issue that looks to have been fixed, I'll close it.

Jonk2, 2021-08-23:
Here's an extract from the log: 

> Aug 23 22:59:59 allsky2 allsky.sh[2257]: === Starting nighttime capture ===
Aug 23 22:59:59 allsky2 allsky.sh[2257]: ==========
Aug 23 22:59:59 allsky2 allsky.sh[2257]: Saving auto exposed night images with delay of 0 ms (0 sec)
Aug 23 23:00:00 allsky2 allsky.sh[2257]: STARTING EXPOSURE at: 2021-08-23 23:00:00
Aug 23 23:00:06 allsky2 allsky.sh[2257]:   > Extra Text File Age Disabled So Displaying Anyway
Aug 23 23:00:06 allsky2 allsky.sh[2257]:   > Saving image 'image.jpg' that started at 2021-08-23 23:00:00  (10 µs)
Aug 23 23:00:06 allsky2 allsky.sh[2257]:   > Sleeping: 26937 ms
Aug 23 23:00:33 allsky2 allsky.sh[2257]: STARTING EXPOSURE at: 2021-08-23 23:00:33
Aug 23 23:00:37 allsky2 allsky.sh[2257]:   > Extra Text File Age Disabled So Displaying Anyway
Aug 23 23:00:37 allsky2 allsky.sh[2257]:   > Saving image 'image.jpg' that started at 2021-08-23 23:00:33  (9 µs)
Aug 23 23:00:37 allsky2 allsky.sh[2257]:   > Sleeping: 28124 ms
Aug 23 23:01:05 allsky2 allsky.sh[2257]: STARTING EXPOSURE at: 2021-08-23 23:01:05
Aug 23 23:01:07 allsky2 allsky.sh[2257]:   > Extra Text File Age Disabled So Displaying Anyway
Aug 23 23:01:07 allsky2 allsky.sh[2257]: malloc(): unsorted double linked list corrupted
Aug 23 23:01:07 allsky2 allsky.sh[2257]: /home/pi/allsky/allsky.sh: line 146:  2345 Aborted                 $ALLSKY_HOME/capture "${ARGUMENTS[@]}"
Aug 23 23:01:07 allsky2 allsky.sh[2257]: capture exited with retcode=134
Aug 23 23:01:12 allsky2 allsky.sh[2461]:      ***** Starting AllSky *****

I thought that the string of odd characters in the text file may have caused an issue, so I removed them.

At first I thought it had worked, as I got 3 exposures rather than 2, but then it restarted again.

Do I need to 'make allsky' as well as 'make capture'?

Jonk2, 2021-08-23:
I might add that the indoors setup is using a different zwo to the one outside (a 290MM in this case). I doubt that is anythign to do with it?

thomasjacquin, 2021-08-23:
@Jonk2 ,
I am seeing the same issue in v0.8. 
@EricClaeys is aware of the problem. I believe he knows how to fix it. 
```The code assumes that C++ only executes the part after the && if the statement before it is true.  If this is incorrect and the input line has one character (e.g., a newline) then the "if" will throw an error since it will have line[1-2], or line[-1].  Maybe it would be safer to split into 2 "if" statements.```

Eric, would you be able to submit a fix onto the `dev` branch ? After testing at home I will merge it into master.

In the meantime, I'm using the old code (but it will probably bring back your line feed error):

```
if (bAddExtra) {
    char *line = NULL;
    size_t len = 0;
    while (getline(&line, &len, fp) != -1) {
        if (line[strlen(line)-1] == 10) {
            line[strlen(line)-1] = '\0';
        };
        cvText(pRgb, line, iTextX, iTextY + (iYOffset / bin), fontsize * 0.08, linewidth, linetype[linenumber], fontname[fontnumber], smallFontcolor, Image_type, outlinefont);
        iYOffset += iTextLineHeight;
    }
}
```

EricClaeys, 2021-08-24:
Thomas, I submitted a pull request to fix the malloc() error.  The "free(line);" was causing the error.  I must be misreading the getline() man page where it mentions having to free() the line.
I also put in a check for blank lines (i.e., line with only a LF on it).

EricClaeys, 2021-08-24:
Jon,
I come from the school that says if a program crashes or aborts, it's the fault of the programmer, not the user, even if the user enters something they shouldn't, the program should print an error message rather than crash.
Hence, you didn't cause the problem; it was already there.
Eric

EricClaeys, 2021-09-29:
Problem resolved so closing the issue.  Thanks for bringing it to my attention.

closed 2021-09-29 by EricClaeys