AllskyTeam/allsky #402: High temperatures durning day

LinuxHeki, 2021-07-30:
Default settings* ASI224MC at night gets hot from 36 to 40 degrees at day is up to 60!

Default settings*:
Delay: 10
Daytime delay: 5000
DAYTIME="0"
CAPTURE_24HR=false

jcauthen78, 2021-07-30:
Heat is rough at times, especially when seeing numbers that warm - can be alarming. 
As of right now, my 178MC is sitting at 75.7c (167f). Over the coming years I'm sure it'll degrade, but for the length of time its been running (in full sunlight during the day), no image degradation yet. Yes, its' above the data-sheet's stated operating temperature, but it's just one of those things that's gonna happen in a dome with direct sunlight. 

My ASI120MC showed some damage after about 3 years, but that was in an arc pattern on the chip, so assuming it was the sun itself, and not the temperatures.

EricClaeys, 2021-08-01:
LinuxHeki,
Do you have a fan?  I have one and it lowers the temperature by about 15 C.
Eric

LinuxHeki, 2021-08-01:
No, I have ASI224MC without fan.

LinuxHeki, 2021-08-01:
I changed "Daytime delay" from 5000 to 10000 and it was good decision. Temperature is now from 5 to 10 degrees lower (durning day).

EricClaeys, 2021-08-01:
A few of us are working on enhancements to allsky, which include a new algorithm for taking exposures that lowers the sensor temperature some.  We're pretty close to submitting to Thomas for review.  When this comes out it should help you.
Eric

LinuxHeki, 2021-08-01:
Great!

closed 2021-08-01 by LinuxHeki