AllskyTeam/allsky #404: Random white and "corrupted" images

LinuxHeki, 2021-08-02:
In galery I found some "corrupted" images and some groups of 3 white images. Look: https://1drv.ms/u/s!AovgSk7aQnyUea7zzzg0-XB2yBw

I can't delete them because it's a lot of them:
https://1drv.ms/v/s!AovgSk7aQnyUemNziOJNTkLsmu8

What is this!?

LinuxHeki, 2021-08-02:
"Corrupted" images are just durning day.

LinuxHeki, 2021-08-02:
Some images have also white line at the bottom:
https://1drv.ms/u/s!AovgSk7aQnyUe0X_Ymh9Juw9d9E

Do just I have this images?

EricClaeys, 2021-08-02:
LinuxHeki,

The white blocks aren't images.  There's something in the HTML that causes gaps to occur.  You'll get different widths depending on how large your window is.

A lot of us get the corrupted images (with the diagonal lines).  It's probably related to USB transmission from the camera to the Pi.  You can try changing the USB speed from the default of 40 to something else.
I suspect it's caused by a bug in the ZWO library because it only happens during the day.  I haven't seen it in a while, possibly because I'm using our new capture.cpp that allows auto USB speed, which I have turned on.

Can you post a full-sized image with white lines at the bottom?  I want to make sure it's in the image and not the thumbnail.

Eric

LinuxHeki, 2021-08-02:
Oh that explain why are always 3 white images. And white lines are also HTML bug. What is something else for USB speed? From 10 to 100?

EricClaeys, 2021-08-02:
40 is the minimum and I believe 100 is the maximum.

closed 2021-08-02 by LinuxHeki