AllskyTeam/allsky #407: V0.8 - ASI Timeout

Jonk2, 2021-08-06:
So, I've removed the previous allsky folder, and re-cloned the new one (0.8), putting back in all my settings.

Initially, I had an image appear, no problem, but after changing a couple of settings (in the camera settings I think, I haven't touched anything I don't understand) I now have a black image with the following in the log:

```
Aug  6 14:03:10 allsky allsky.sh[1831]: ==========
Aug  6 14:03:10 allsky allsky.sh[1831]: === Starting daytime capture ===
Aug  6 14:03:10 allsky allsky.sh[1831]: ==========
Aug  6 14:03:10 allsky allsky.sh[1831]: Saving images with delay of 60,000 ms (60 sec)
Aug  6 14:03:11 allsky allsky.sh[1831]:   > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
Aug  6 14:03:13 allsky allsky.sh[1831]:   > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
```

I know the camera is fine, as after a reboot etc. I can open it in another program on the PI and it's fine.

I'm going to reset the settings to default, but I've mentioned it here in case anyone else sees the same and collectively, we can give the devs the info to find a possible issue.

Jonk2, 2021-08-06:
No, that hasn't worked - here's more of the log, showing the camera detected, initialised and ready to go:

```
Aug  6 14:07:49 allsky allsky.sh[3725]:      ***** Starting AllSky *****
Aug  6 14:07:49 allsky allsky.sh[3725]: Making sure allsky.sh is not already running...
Aug  6 14:07:49 allsky allsky.sh[3725]: Trying to automatically choose between ZWO and RPI camera
Aug  6 14:07:49 allsky allsky.sh[3725]: /home/pi/allsky/allsky.sh: line 89: /home/pi/allsky/scripts/copy_notification_image.sh: Permission denied
Aug  6 14:07:49 allsky allsky.sh[3725]: CAMERA: ZWO
Aug  6 14:07:49 allsky allsky.sh[3725]: CAMERA_SETTINGS: /etc/raspap/settings_ZWO.json
Aug  6 14:07:49 allsky allsky.sh[3725]: Starting allsky camera...
Aug  6 14:07:51 allsky allsky.sh[3725]: #033[32m ******************************************
Aug  6 14:07:51 allsky allsky.sh[3725]: #033[32m *** Allsky Camera Software v0.8 | 2021 ***
Aug  6 14:07:51 allsky allsky.sh[3725]: #033[32m ******************************************
Aug  6 14:07:51 allsky allsky.sh[3725]: #033[32mCapture images of the sky with a Raspberry Pi and an ASI Camera
Aug  6 14:07:51 allsky allsky.sh[3725]: #033[33mAdd -h or -help for available options
Aug  6 14:07:51 allsky allsky.sh[3725]: #033[0mAuthor: Thomas Jacquin - <jacquin.thomas@gmail.com>
Aug  6 14:07:51 allsky allsky.sh[3725]: #033[0mContributors:
Aug  6 14:07:51 allsky allsky.sh[3725]: -Knut Olav Klo
Aug  6 14:07:51 allsky allsky.sh[3725]: -Daniel Johnsen
Aug  6 14:07:51 allsky allsky.sh[3725]: -Yang and Sam from ZWO
Aug  6 14:07:51 allsky allsky.sh[3725]: -Robert Wagner
Aug  6 14:07:51 allsky allsky.sh[3725]: -Michael J. Kidd - <linuxkidd@gmail.com>
Aug  6 14:07:51 allsky allsky.sh[3725]: -Chris Kuethe
Aug  6 14:07:51 allsky allsky.sh[3725]: #033[0m
Aug  6 14:07:51 allsky allsky.sh[3725]: Listing Attached Cameras:
Aug  6 14:07:51 allsky allsky.sh[3725]:   - 0 ZWO ASI178MC
Aug  6 14:07:51 allsky allsky.sh[3725]: ZWO ASI178MC Information:
Aug  6 14:07:51 allsky allsky.sh[3725]:   - Resolution:3096x2080
Aug  6 14:07:51 allsky allsky.sh[3725]:   - Pixel Size: 2.4ÃŽÂ¼m
Aug  6 14:07:51 allsky allsky.sh[3725]:   - Supported Bin: 1 2 3 4
Aug  6 14:07:51 allsky allsky.sh[3725]:   - Color Camera: bayer pattern:RG
Aug  6 14:07:51 allsky allsky.sh[3725]:   - SDK version 1, 19, 0, 0
Aug  6 14:07:51 allsky allsky.sh[3725]:   - Initialise Camera OK
Aug  6 14:07:51 allsky allsky.sh[3725]: - Sensor temperature:37.60
Aug  6 14:07:51 allsky allsky.sh[3725]: #033[32m
Aug  6 14:07:51 allsky allsky.sh[3725]: Capture Settings:
Aug  6 14:07:51 allsky allsky.sh[3725]:  Image Type: ASI_IMG_RGB24
Aug  6 14:07:51 allsky allsky.sh[3725]:  Resolution (before any binning): 3096x2080
Aug  6 14:07:51 allsky allsky.sh[3725]:  Quality: 95
Aug  6 14:07:51 allsky allsky.sh[3725]:  Daytime capture: 1 (yes)
Aug  6 14:07:51 allsky allsky.sh[3725]:  Exposure (day): 0.500ms
Aug  6 14:07:51 allsky allsky.sh[3725]:  Auto Exposure (day): 1 (yes)
Aug  6 14:07:51 allsky allsky.sh[3725]:  Exposure (night): 10000ms
Aug  6 14:07:51 allsky allsky.sh[3725]:  Max Exposure (night): 20000ms
Aug  6 14:07:51 allsky allsky.sh[3725]:  Auto Exposure (night): 1 (yes)
Aug  6 14:07:51 allsky allsky.sh[3725]:  Delay (day): 5000ms
Aug  6 14:07:51 allsky allsky.sh[3725]:  Delay (night): 10ms
Aug  6 14:07:51 allsky allsky.sh[3725]:  Gain (night only): 50
Aug  6 14:07:51 allsky allsky.sh[3725]:  Auto Gain (night only): 0 (no)
Aug  6 14:07:51 allsky allsky.sh[3725]:  Max Gain (night only): 200
Aug  6 14:07:51 allsky allsky.sh[3725]:  Gain Transition Time: 1800 seconds
Aug  6 14:07:51 allsky allsky.sh[3725]:  Brightness (day): 50
Aug  6 14:07:51 allsky allsky.sh[3725]:  Brightness (night): 50
Aug  6 14:07:51 allsky allsky.sh[3725]:  Cooler Enabled: 0 (no)
Aug  6 14:07:51 allsky allsky.sh[3725]:  Target Temperature: 0C
Aug  6 14:07:51 allsky allsky.sh[3725]:  Gamma: 50
Aug  6 14:07:51 allsky allsky.sh[3725]:  WB Red: 53
Aug  6 14:07:51 allsky allsky.sh[3725]:  WB Blue: 90
Aug  6 14:07:51 allsky allsky.sh[3725]:  Auto WB: 0 (no)
Aug  6 14:07:51 allsky allsky.sh[3725]:  Binning (day): 1
Aug  6 14:07:51 allsky allsky.sh[3725]:  Binning (night): 1
Aug  6 14:07:51 allsky allsky.sh[3725]:  USB Speed: 40
Aug  6 14:07:51 allsky allsky.sh[3725]:  Auto USB Speed: 0 (no)
Aug  6 14:07:51 allsky allsky.sh[3725]:  Text Overlay: [none]
Aug  6 14:07:51 allsky allsky.sh[3725]:  Text Extra Filename: [none]
Aug  6 14:07:51 allsky allsky.sh[3725]:  Text Extra Filename Age: 0
Aug  6 14:07:51 allsky allsky.sh[3725]:  Text Line Height 30px
Aug  6 14:07:51 allsky allsky.sh[3725]:  Text Position: 15px left, 35px top
Aug  6 14:07:51 allsky allsky.sh[3725]:  Font Name:  0 (SIMPLEX)
Aug  6 14:07:51 allsky allsky.sh[3725]:  Font Color: 255 , 255, 255
[... 22 lines cut ...]
Aug  6 14:07:51 allsky allsky.sh[3725]:  Preview: 0 (no)
Aug  6 14:07:51 allsky allsky.sh[3725]:  Darkframe: 0 (no)
Aug  6 14:07:51 allsky allsky.sh[3725]:  Debug Level: 0
Aug  6 14:07:51 allsky allsky.sh[3725]:  TTY: 0 (no)
Aug  6 14:07:51 allsky allsky.sh[3725]: #033[0m
Aug  6 14:07:51 allsky allsky.sh[3725]: Will adjust gain at transitions
Aug  6 14:07:51 allsky allsky.sh[3725]: Stop the allsky service to end this process.
Aug  6 14:07:51 allsky allsky.sh[3725]: ==========
Aug  6 14:07:51 allsky allsky.sh[3725]: === Starting daytime capture ===
Aug  6 14:07:51 allsky allsky.sh[3725]: ==========
Aug  6 14:07:51 allsky allsky.sh[3725]: Saving images with delay of 5000 ms (5 sec)
[... cut ...]
Aug  6 14:07:41 allsky allsky.sh[3593]: sh: 1: scripts/copy_notification_image.sh: Permission denied
Aug  6 14:07:41 allsky allsky.sh[3593]: saveImageDay.sh: Uploading 'image.jpg' with timestamp: 14:07:41
Aug  6 14:07:44 allsky allsky.sh[3593]:      ***** Stopping AllSky *****
Aug  6 14:07:44 allsky allsky.sh[3593]: capture exited with retcode=2
Aug  6 14:07:44 allsky allsky.sh[3593]: /home/pi/allsky/allsky.sh: line 152: /home/pi/allsky/scripts/copy_notification_image.sh: Permission denied
```
Something to check?

Jonk2, 2021-08-06:
[image]

Jonk2, 2021-08-06:
I've deleted the allsky folder, and settings files from raspap, cloned a fresh copy, ran the install.sh and gui installation.

Rebooted the PI and cleared browser cahse to log back into the web interface.

I still have a black image with the same permission denied entries in the log, and the camera is fine in another capture package.

Soooooo....what's next?

Jonk2, 2021-08-06:
I've change the permissions on scripts/copy_notification_image.sh to match the others (755) and I now see 'Allsky is not running' in red in place of the black image, and this changes to 'Allsky is starting up' in green. They alternate every 5 seconds (probably tied to the image auto-refresh on the browser?). 
In the log, it is 'Starting AllSky' every 20 seconds, each time ending with the ASI_ERROR_TIMEOUT.

Jonk2, 2021-08-06:
I've moved the camera onto one of the USB3 ports, having previously moved it to the USB2 port and it's now working. That must have caused the issue.

Food for thought - why does it not work on the USB2 port when in other programs it's fine?

EricClaeys, 2021-08-08:
Jon,
I'll have to ask Thomas about the permissions on scripts/copy_notification_image.sh, but you did the right thing.
Some people have found it helpful to change the USB speed (either higher or lower) when the software can't get pictures.  I don't know why it didn't work on your USB2 port when it used to.
When you say you got "a black image", was there a black "image.jpg" file, or the web browser simply didn't show any image?

After two failed attempts to get an image, the software exits and displays the "Allsky is not running" message.  If you have the service start allsky, it restarts the software and one of the first things it does is display the "Allsky is starting up" message.  If Allsky gets 2 failed images it'll exit, get restarted.... for ever.

Eric

Jonk2, 2021-08-09:
Eric,

A new file has been added byt Thomas, with the correct permissions.

Is was a black background or placeholder, i.e. no image - I didn't dig any deeper, but let's say you took a complete black jpg and displayed it, that's what it looked like - there was no 'broken link' image or anything like that.

Regarding the ASI timeout, as mentioned on another thread, it may have been the bandwidth setting. May I suggest that you set it to auto by default for all new installs, and make it clear that the user can specify if they want to, rather than setting it to 40 default and finding it may not work?

i suppose it could easily be tested, reduce the bandwidth figure until a timeout occurs, then increase until the error is no longer valid. That would prove it.

I will close this now as this appears to be resolved, at least from an understanding point of view.

closed 2021-08-09 by Jonk2