AllskyTeam/allsky #409: GUI Interface

maserowik, 2021-08-06:
On the GUI Admin screen connection status there is a status for the wireless.  Is there a way to get the statistics of the ethernet aka hardwire?

EricClaeys, 2021-08-08:
Edit line 46 of /var/www/html/index.php: change the "false" to "true".  Then reload the GUI admin page.  You'll see a "LAN connection status" link on the left.

**Please let me know how it goes, especially if you click on the "Stop eth0" button.  That hasn't been thoroughly tested.**

Eric

maserowik, 2021-08-09:
Eric, 

my line 46 had this listed in it

      " 46 define('RASPI_HOSTAPD_CONFIG', '/etc/hostapd/hostapd.conf');"

this is my version number
     " 13  * @version    0.0.1"

Thanks 
Mike

EricClaeys, 2021-08-09:
Mike,
Do you have the new code that was uploaded a few days ago?
If not, you might want to download all three pieces: "allsky" (the bulk of AllSky), "allsky-website" (optional website), and "allsky-portal" (the GUI).

If you do, you'll need to change permissions of script/copy_notification_image.sh to 775, and may need to increase your "USB Bandwidth".

Eric

maserowik, 2021-08-09:
I uploaded to the newest version of allsky.  I do see the "LAN Dashboard"  the Stop eth0 button does not appear to do anything or at lease I am still seeing activity on the stats.  I click the stop etho button and click on the refresh button and the statistics updated.  is there a way to change the interface name it is currently set to "wlan0" 

A big thanks to  all the developers and testers who keep making updates enhancements to the software.

EricClaeys, 2021-08-09:
Ooops.  That was a copy/paste error.  Line 65 of includes/dashboard_eth0.php has "wlan0" - change to "eth0".

I'm pretty sure the problem is that the "pi" user doesn't have permission to turn eth0 on and off.  Add a couple lines to /etc/sudoers.d/allsky that look like the "wlan0" lines but say "eth0" instead.  You'll need to sudo in order to edit the file.
Then try the "Stop eth0" and "Start eth0" and let me know how it goes please.

Eric

maserowik, 2021-08-15:
I made the changed and now it show correctly.

I added the lines to the allsky file, rebooted the pi and still unable to stop the eth0.  not a show stopper.

EricClaeys, 2021-08-15:
Mike,
Thanks for trying.
I just ordered another Pi to play around with, so I'll see if I can get it to work. I'm sure the fix is trivial, once I determine what it is.
Eric

maserowik, 2021-08-15:
Eric,
    I still have to modify this line
"Line 65 of includes/dashboard_eth0.php has "wlan0" - change to "eth0"."

Mike

EricClaeys, 2021-08-15:
That line is FYI and doesn't actually do anything but changing it is a good idea.

maserowik, 2021-08-16:
closing this due to it working correctly.

closed 2021-08-16 by maserowik

EricClaeys, 2021-08-16:
Mike,
are you saying it now stops / starts eth0?  If so, that's great - what did you do to get it working?

maserowik, 2021-08-16:
nope that still does not work.  I am closing it because the original issue of the eth0 not showing up.  the start and stop was a after thought.

EricClaeys, 2021-08-16:
Mike,
My test Pi arrived today so I can test it myself.  Thanks for the initial test of the page.  I'm glad at least that's working.  I'm sure I'll be able to get the start/stop working, too.
Eric