AllskyTeam/allsky #413: Standalone operation of capture_RPiHQ.cpp

widobsono, 2021-08-09:
Hi I a still trying to understand behind the scenes operation of Allsky, I am curious is it possible to run `capture_RPiHQ.cpp` alone? I found some limitations on piHQ camera exposure time using `picamera`, I found Allsky but the other features are not suitable for my application (sampling the twilight sky near horizon with various exposure time) , so I am super curious is is possible to run `capture_RPiHQ.cpp` on it's own to replace `PiCamera` library.

[`Keogram`](https://github.com/thomasjacquin/allsky#keograms) and [`Startrails`](https://github.com/thomasjacquin/allsky#startrails) manual script is an amazing example. I am trying to integrate it with Python.

Looking forward to the guidance and comments. Thank you, cheers!

EricClaeys, 2021-08-09:
What do you mean by "run it alone"?
Normally the "allsky.sh" script calls cature_RPiHQ, passing the necessary arguments.
You could do the same thing, but it would be easier to simply run "./allsky.sh".

widobsono, 2021-08-10:
Hi Eric, thank you for answering, I want to be able to manually call `capture_RPiHQ` and pass the arguments using python, I saw the `capture_RPiHQ` is based on `raspistill` to capture image right? or a custom script? If it is `raspistill`, my plan manually using `capture_RPiHQ` is unnecessary

EricClaeys, 2021-08-10:
If you are using the code that just came out a few days ago, you can set the debug level to 3, then restart AllSky.  It will output to /var/log/allsky.log the raspistill commands it is executing.  Your Python script could do the same.

If this answers your question please close the issue.

Eric

widobsono, 2021-08-10:
Thank you so much Eric, very helpful :smile: Problem solved :+1:

closed 2021-08-10 by widobsono