AllskyTeam/allsky #414: RPiHQ captures only one image at daytime

MarcusL1979, 2021-08-09:
The latest version did not run correctly during the daytime. 
Only at the start, a picture will be captured.

MarcusL1979, 2021-08-09:
Some log sinppets. After several clean installs. no changes. 

Aug  9 20:51:54 allsky allsky.sh[1261]:  Camera: Raspberry Pi HQ camera
Aug  9 20:51:54 allsky allsky.sh[1261]: #033[0m
Aug  9 20:51:54 allsky allsky.sh[1261]: === Starting daytime capture ===
Aug  9 20:51:54 allsky allsky.sh[1261]: Saving 32000 ms exposed images with 1 seconds delays in between...
Aug  9 20:51:54 allsky allsky.sh[1261]: Stop the allsky service to end this process.
Aug  9 20:51:54 allsky allsky.sh[1261]: Capturing & saving image...
Aug  9 20:52:02 allsky allsky.sh[1261]: Capturing & saving image done, now wait 1 seconds...
Aug  9 20:56:12 allsky allsky.sh[1261]:      ***** Stopping AllSky *****
Aug  9 20:56:24 allsky allsky.sh[1524]:      ***** Starting AllSky *****
Aug  9 20:56:24 allsky allsky.sh[1524]: Making sure allsky.sh is not already running...
Aug  9 20:56:24 allsky allsky.sh[1524]: Trying to automatically choose between ZWO and RPI camera
Aug  9 20:56:24 allsky allsky.sh[1524]: jq: error: Could not open file /etc/raspap/settings_auto.json: No such file or directory
Aug  9 20:56:24 allsky allsky.sh[1524]: *** copy_notification_image.sh: ERROR: File 'notification_images/StartingUp.' does not exist or is empty!
Aug  9 20:56:24 allsky allsky.sh[1524]: CAMERA: RPiHQ
Aug  9 20:56:24 allsky allsky.sh[1524]: CAMERA_SETTINGS: /etc/raspap/settings_RPiHQ.json
Aug  9 20:56:26 allsky allsky.sh[1524]: Starting allsky camera...
Aug  9 20:56:28 allsky allsky.sh[1524]: #033[32m ******************************************
Aug  9 20:56:28 allsky allsky.sh[1524]: #033[32m *** Allsky Camera Software v0.8 | 2021 ***
Aug  9 20:56:28 allsky allsky.sh[1524]: #033[32m ******************************************
Aug  9 20:56:28 allsky allsky.sh[1524]: #033[32mCapture images of the sky with a Raspberry Pi and an ZWO ASI or RPi HQ camera
Aug  9 20:56:28 allsky allsky.sh[1524]: #033[33mAdd -h or -help for available options
Aug  9 20:56:28 allsky allsky.sh[1524]: #033[0mAuthor: Thomas Jacquin - <jacquin.thomas@gmail.com>

blueh0le, 2021-08-09:
Have exactly the same problems.
Ruediger

Jonk2, 2021-08-09:
This line here: 

> Aug 9 20:56:24 allsky allsky.sh[1524]: jq: error: Could not open file /etc/raspap/settings_auto.json: No such file or directory

Says to me you have auto set as the camera in config.sh.

Chage this to 'RPiHQ' and see if it works correctly then.

I 'think' that the settings_auto.json doesn't exist in the latest fileset, but not sure if it's auto generated?

MarcusL1979, 2021-08-09:
Hi Jonk2, 
I check this. No changes. This issue is annoying.

MarcusL1979, 2021-08-09:
Is there a chance to get instructions to revert to the latest 0.7 release?

EricClaeys, 2021-08-09:
The "jq" error and " copy_notification_image.sh" error should only impact the "Allsky is starting up" image that's displayed when you first start Allsky.  The attached version of allsky.sh will fix the messages.

The title of this issue implies only one picture is taken.  Are you saying the camera takes one picture than no more?  If so, please set the "Debug Level" to 3, then restart allsky, let it run a few minutes, then stop it and set the Debug level to what it was before, then send the complete log.

Thanks - Eric

[allsky.zip](https://github.com/thomasjacquin/allsky/files/6957624/allsky.zip)

MarcusL1979, 2021-08-10:
Hi Eric, 
Yes the v0.8 only captures one image and stop look at the time stamp. 
I need to check your advices later on 

Aug 9 20:51:54 allsky allsky.sh[1261]: Capturing & saving image...
Aug 9 20:52:02 allsky allsky.sh[1261]: Capturing & saving image done, now wait 1 seconds...
Aug 9 20:56:12 allsky allsky.sh[1261]: ***** Stopping AllSky *****
Aug 9 20:56:24 allsky allsky.sh[1524]: ***** Starting AllSky *****

EricClaeys, 2021-08-10:
At Debug Level 3 there should be a lot more log entries.  If you could include everything in the log that would be helpful.  It'll tell me what your options are and likely give a reason for only taking one image.
Eric

MarcusL1979, 2021-08-10:
Hi Eric,

I'v tried to change the log level bu it seems not have any effect.
attached the log and the config file 

[allsky.zip](https://github.com/thomasjacquin/allsky/files/6960180/allsky.zip)

MarcusL1979, 2021-08-10:
Hi Eric
I Think I found the issue. 
The time between two images is not longer in ms it is in seconds. Please check and confirm.

hoetzgit, 2021-08-10:
my temp workaround in capture_RPiHQ.cpp:

                        // Sleep for a moment
                        //usleep(currentDelay * US_IN_SEC);
                        usleep(10000);

EricClaeys, 2021-08-10:
Marcus,
You found a bug - good job.  Sorry about that.
The fix is to use US_IN_MS in all the usleep() calls.  Usleep() takes microseconds (US) but the user specifies delays in milliseconds (MS).  The current code is using a delay 1000 times longer than it should!

jmodrako, 2021-08-10:
I can confirm, when I've decrease delay it's started working :)

thomasjacquin, 2021-08-11:
@EricClaeys Is that a newly introduced bug? Do you have a fix for it? I can take a look but if you already have a fix, please submit a PR and I'll merge it into the `master` branch.

EricClaeys, 2021-08-12:
Thomas,
Have been driving all day home from my remote observatory and am pretty tired.  Not sure if I will get to it tonight, but the fix is easy.  Change the US_IN_SEC to US_IN_MS.  I believe there is only one place in that file that does a usleep().

Eric

EricClaeys, 2021-08-14:
MarcusL1979,
Thomas uploaded a fixed version of capture_RPiHQ.cpp.  If you haven't already gotten it, please do so and copy to /home/pi/allsky, then run "make capture_RPiHQ" and restart AllSky.  If that fixes the problem, please close this issue.
Thanks much - Eric

MarcusL1979, 2021-08-14:
Issue fixed

closed 2021-08-14 by MarcusL1979