AllskyTeam/allsky #415: FTP was solved with old version of AllSky

maserowik, 2021-08-10:
Greetings,  with the older version of Allsky I was able to follow this link to get the ftp working 

[https://github.com/thomasjacquin/allsky/issues/74](url)

I added the setting to the files but the software will not upload to my website.

any suggestions.
Mike

EricClaeys, 2021-08-13:
Mike, try adding "echo" before the lftp line in the .sh file that isn't uploading correctly.  Then try executing that command manually.   You will need to put quotes in the same places you see them in the .sh file.  This way you will see any error messages.

maserowik, 2021-08-15:
> Mike, try adding "echo" before the lftp line in the .sh file that isn't uploading correctly. Then try executing that command manually. You will need to put quotes in the same places you see them in the .sh file. This way you will see any error messages.

I not sure if it just lack of knowledge or I screwed up something good.  I placed the echo on the command line you stated and it echo the command but does nothing. i check my hosting site and there is nothing placed there.
the log file is stating it is upload the file but it is not on my website

Aug 14 20:10:29 Pi-AllSky allsky.sh[1355]: lftp ftp://**********:*********@n3bsq.com:/domains/n3bsq.com/public_html/allsky/ -e set net:max-retries 2; set net:timeout 20; put image.jpg -o si-3941; rm -f image.jpg; mv si-3941 image.jpg; bye

not sure if i did the command correctly or not.

EricClaeys, 2021-08-15:
Mike,
you did it correctly.  "echo" simply outputs everything on the line; it does not execute anything.  I wanted to see what that file was trying to execute.
Remove the "echo", then at a shell prompt, type the following from the /home/pi/allsky directory:

`lftp "ftp://*:@n3bsq.com:/domains/n3bsq.com/public_html/allsky/" -e "set net:max-retries 2; set net:timeout 20; put image.jpg -o si-3941; rm -f image.jpg; mv si-3941 image.jpg; bye"
`

Make sure you include the double quotes.
What's in the shell script is:
`lftp "$PROTOCOL://$USER:$PASSWORD@$HOST:$IMGDIR" -e "set net:max-retries 2; set net:timeout 20; put "$IMAGE_TO_USE" -o $TEMP_NAME; rm -f "$IMAGE_TO_USE"; mv $TEMP_NAME "$IMAGE_TO_USE"; bye" &`

The "echo" allows us to see what all the variables are (that start with $).
I notice some likely issues:
* Your $USER is "*".
* Your $PASSWORD is empty.

Make sure your settings in the scripts/ftp-settings.sh file are correct.  Here are the defaults:
```
PROTOCOL='ftp'
USER='username'
PASSWORD='password'
HOST='example.com'
IMGDIR='/allsky/'
MP4DIR='/allsky/videos/'
KEOGRAM_DIR='/allsky/keograms/'
STARTRAILS_DIR='/allsky/startrails/'
```

Once you can successfully login manually and have fixed your settings, allsky should be able to as well.

maserowik, 2021-08-15:
i have the setting set correctly in the ftp-settings.sh I just scrubbed the username and password from the lftp line

here is the output i get from the lftp command

pi@Pi-AllSky:~/allsky/scripts $  lftp ftp://*********:************@n3bsq.com:/domains/n3bsq.com/public_html/allsky/ -e set net:max-retries 2; set net:timeout 20; put image.jpg -o si-3941; rm -f image.jpg; mv si-3941 image.jpg; bye
cd ok, cwd=/domains/n3bsq.com/public_html/allsky
set cmd:prompt "lftp \\S\\? \\u\\@\\h:\\w> "
set cmd:term-status/*rxvt* "\\e[11;0]\\e]2;\\T\\007\\e[11]"
set cmd:term-status/*screen* \\e_\\T\\e\\
set cmd:term-status/*xterm* "\\e[11;0]\\e]2;\\T\\007\\e[11]"
set color:dir-colors "rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:"
set dns:order "inet6 inet"
set file:charset UTF-8
set ftp:auto-sync-mode "icrosoft FTP Service|MadGoat|MikroTik"
set ftp:sync-mode/ftp.idsoftware.com on
set ftp:sync-mode/ftp.microsoft.com on
set ftp:sync-mode/sunsolve.sun.com on
set ssl:ca-file /etc/ssl/certs/ca-certificates.crt
set ssl:verify-certificate no
set xfer:max-redirections 10
lftp u919658081@n3bsq.com:/domains/n3bsq.com/public_html/allsky>

maserowik, 2021-08-15:
i hit the enter key too quick.  i have to type bye and then get the following

lftp u919658081@n3bsq.com:/domains/n3bsq.com/public_html/allsky> bye
-bash: put: command not found
mv: cannot stat 'si-3941': No such file or directory

jcauthen78, 2021-08-15:
For the record - I haven't updated either of my external web-hosts with the new updates from Eric 9 days ago (https://bit.ly/3AFs1CP). But I did update my Allsky to v0.8. 

I had to change up the LFTP command a bit to the following: 
`lftp "$PROTOCOL://$USER:$PASSWORD@$HOST:$IMGDIR" -e "set net:max-retries 2; set net:timeout 20; put "$IMAGE_TO_USE" -o $TEMP_NAME; rm -f "$IMAGE_TO_USE"; mv $TEMP_NAME "image-resize.jpg"; rm -f $TEMP_NAME; bye" &`

It made it so the original "image-resize.jpg" would still be used. 

Just a thought that may help, if you're in the same boat.

EricClaeys, 2021-08-15:
Mike,
The "-bash" error implies the "put" part of the lftp was being executed by the shell, not by lftp.  Check your double quotes.  They should be like this:
`lftp "ftp://*:@n3bsq.com:/domains/n3bsq.com/public_html/allsky/" -e "set net:max-retries 2; set net:timeout 20; put image.jpg -o si-3941; rm -f image.jpg; mv si-3941 image.jpg; bye"`

lftp is taking 3 arguments - one with the protocol, user name, etc.  One with "-e" and the last with "set ... bye".

maserowik, 2021-08-15:
ok i just reimaged my sd card and did a new pull for the software.  I reset my ftp-settings.sh and now i see the image on my website.  hopefully it will work with the video/keograms/startrails tonight.

Mike

EricClaeys, 2021-08-15:
Mike, are you running version 0.8?  If so, do you have a ZWO camera?  What are your "USB Bandwidth" and "Auto USB" settings?

maserowik, 2021-08-15:
I have a zwo 224 cameras
I have the Auto USB Bandwidth set to no
My USB Bandwidth is set to 40
not sure how to find what version I have.  

my setup is a raspberry pi 3b+ 
pi@Pi-AllSky:~ $ cat /etc/debian_version
10.10
pi@Pi-AllSky:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@Pi-AllSky:~ $ uname -a
Linux Pi-AllSky 5.10.52-v7+ #1441 SMP Tue Aug 3 18:10:09 BST 2021 armv7l GNU/Linux

EricClaeys, 2021-08-15:
Mike,
The /var/log/allsky.log file will tell you if you're running AllSky 0.7 or 0.8.
I assume you aren't having problems capturing images?

maserowik, 2021-08-15:
I will look when I get home.  

So far it is capturing an image during daylight.  It is 6 pm where I am and will check to see what it looks like when it is dark

maserowik, 2021-08-16:
Eric,
here is what the allsky.log file says.

Aug 15 13:39:44 Pi-AllSky allsky.sh[31149]: #033[32m *** Allsky Camera Software v0.8 | 2021 ***

closed 2021-08-16 by maserowik