AllskyTeam/allsky #416: Admin page security (web access)

Jonk2, 2021-08-10:
I've set up my PI so I can at least see the public.php page with the latest image on it from the web.

Having played with website creation in the past, I noticed that there's no logout button on the admin page. This is a big security issue for anyone accessing their admin pages from the web.

I could add my own .htaccess file to control access and visibility etc. and a logout link using session_destroy(); but I feel this should be added by the devs.

linuxkidd, 2021-09-24:
Hi @Jonk2, would you mind creating the code changes necessary to add a logout button and create a subsequent PR against the [allsky-portal](https://github.com/thomasjacquin/allsky-portal) repo.  Thanks!

Jonk2, 2021-09-25:
I've had a look at the gui webpage code, and although adding an extra button for logging out, and having that page kill an open session should be very straighforwards, the authentication for this webpage uses csrf tokens which isn't something I used to use.
What this means is I cannot confidently add code to share with others as I haven't figured out how to kill a session that uses a csrf token. I will keep trying in case I can come up with something that works.
As I've said before however, my programming skills are limited, I'm more of a tester and 'try to break it' kind of guy!

EricClaeys, 2022-06-29:
Closing - this request was added to the Projects list - see the `Projects` link at the top of any Allsky GitHub page.
Project Title: **Add "logout" and "Remember me" buttons to WebUI #416**

closed 2022-06-29 by EricClaeys