AllskyTeam/allsky #417: ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)

bbillp, 2021-08-11:
Latest ALLSKY 8/10/2021 about 7PM PDT Los Angeles.

Pi 3b+
ZWO ASI178MC

Previous ALLSKY versions worked fine but except a recent version makes the ASI178 run hot

Attached as TEXT Files for the upload here,:  config.sh and settings_ZWO.json
[config.sh.txt](https://github.com/thomasjacquin/allsky/files/6965757/config.sh.txt)
[settings_ZWO.json.txt](https://github.com/thomasjacquin/allsky/files/6965758/settings_ZWO.json.txt)

Problem 
One successful DARK image then ASI_ERROR_TIMEOUT repeats for any image.

In subsequent attempts I tried increasing USB from 40 to 100; autoUSB, reduced photo quality,   No luck  What next ?

â— allsky.service - All Sky Camera
   Loaded: loaded (/lib/systemd/system/allsky.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2021-08-10 20:01:36 PDT; 11s ago
 Main PID: 10191 (allsky.sh)
   CGroup: /system.slice/allsky.service
           â”œâ”€10191 /bin/bash /home/pi/allsky/allsky.sh
           â””â”€10282 /home/pi/allsky/capture -alwaysshowadvanced 1 -angle -6 -autousb 0 -autowhitebalance 0 -coolerEnabled 0 -darkframe 1 -dayautoexposure 1 -daybin 1 -daybrightness 50 -daydelay 5000 -dayexposure .5 -debuglevel 1 -extratext -extratextage 600 -filename image.jpg -flip 0 -fontcolor 255 255 255 -fontline 1 -fontname 0 -fontsize 7 -fonttype 0 -gaintransitiontime 15 -gamma 50 -height 0 -histogrambox 500 500 50 50 -latitude 33.5N -locale en_US.UTF-8 -longitude 118.1W -nightautoexposure 1 -nightautogain 0 -nightbin 1 -nightbrightness 50 -nightdelay 10 -nightexposure 10000 -nightgain 195 -nightmaxexposure 20000 -nightmaxgain 200 -notificationimages 1 -outlinefont 0 -quality 95 -showBrightness 0 -showExposure 1 -showGain 1 -showHistogram 0 -showTemp 1 -showTime 1 -showhistogrambox 0 -smallfontcolor 0 0 255 -targetTemp 0 -temptype C -text -textlineheight 60 -textx 15 -texty 30 -timeformat %Y%m%d %H:%M:%S -type 99 -usb 90 -wbb 90 -wbr 53 -width 0 -daytime 0 -tty 0

Aug 10 20:01:40 Allsky3B allsky.sh[10191]:  Debug Level: 1
Aug 10 20:01:40 Allsky3B allsky.sh[10191]:  TTY: 0 (no)
Aug 10 20:01:40 Allsky3B allsky.sh[10191]: 
Aug 10 20:01:40 Allsky3B allsky.sh[10191]: Will NOT adjust gain at transitions
Aug 10 20:01:40 Allsky3B allsky.sh[10191]: Stop the allsky service to end this process.
Aug 10 20:01:40 Allsky3B allsky.sh[10191]: WARNING: Value of -1 less than min value allowed (32) for control 'Exposure' (#1).
Aug 10 20:01:40 Allsky3B allsky.sh[10191]: Taking dark frames...
Aug 10 20:01:41 Allsky3B allsky.sh[10191]:   > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
Aug 10 20:01:43 Allsky3B allsky.sh[10191]:   > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
Aug 10 20:01:46 Allsky3B allsky.sh[10191]:   > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)

bbillp, 2021-08-11:
I tried USBAUTO no and 90 for the bandwidth, no luck.  Also USBAUTO on , no luck.

pierrebilb, 2021-08-11:
Neither got luck, wathever value i put on bandwidth..
No idea..

palmito9, 2021-08-12:
Also a 178mc owner.

With previous version I had to lower quality (png) to 7 and had to plug into a USB2 port.
All this to lower amount of corrupted images and reduce "reset SuperSpeed Gen 1 USB device number X using xhci_hcd" errors.

With the new version of allsky, I had to use to USB3 plug, but 4/5 images fail (image already saving) after "reset SuperSpeed Gen 1 USB device number X using xhci_hcd" error.

pimmoon, 2021-08-13:
Got the same problem on my Pi Zero which was installed two days ago. I tired with both 120mc and 178mc.. didn't work..

EricClaeys, 2021-08-13:
This isn't good.  During testing we eliminated the problem by setting USB Bandwidth high.  One person has it at 100 with the camera the only thing on the USB 3 bus.  Another has it at 75.  I have 80 and auto-usb.  One of us had to move the camera to the USB port.
Only one of you resolved the problem by using "auto" no, and "USB Bandwidth" of 80.  Can other people try a different USB port?  And continue trying different combinations of "Auto USB" and "USB Bandwidth "?
I have a feeling there may be a Pi USB driver issue, since we see the reset errors as well.

**bbillp**, would you please set your Debug Level to 3 then restart allsky and upload the log?  The WARNING you got implies the exposure time was set to -1.   Thanks.

bbillp, 2021-08-13:
snip “bbillp, would you please set your Debug Level to 3 then restart allsky and upload the log? The WARNING you got implies the exposure time was set to -1. Thanks.”

Will do with V0.8 but NOTE:  Beginning with V0.7 the images from the first 5 minutes following power up begin with very short sub second exposures then with each following image the exposure slowly lengthens out to the correct exposure around 15 seconds at typically 5 minutes time.  (Night images only; ASI178MC)  Is there a setting that starts the camera at a longer exposure ?   Autoexpose = yes autogain = no gain = 195  max exposure is >20 seconds, forgot exact number but after 5 minutes the images look very good, no complaints (V0.7)   .  

ASICAP/ASISUITE, FIRECAP, PHD2, ASTROBERRY all work fine .

I have only 1 image from V0.8 until we figure out the Timeout problem with the 178.

Dentonknifeworks, 2021-08-13:
"would you please set your Debug Level to 3" 
Where is this done?
Thanks
Brad

EricClaeys, 2021-08-14:
bbillp,
If you have auto-exposure turned on, then the corresponding "exposure" setting is used as a staring point.
For example, if nighttime "Auto-Exposure" is turned on, then the nighttime "Manual Exposure" setting is used as a starting point if the program is started at night.  However, when the program has been running during the day and nighttime comes, it will use the last daytime exposure as a starting point for night, adjusting the gain if "Gain Transition Time" is greater than 0.  This is done to avoid abrupt changes in brightness between day and night.

In version 0.7, the nighttime exposure setting was called "Exposure".

I have my nighttime "Manual Exposure" set to 45 since when it's fully dark, the camera takes 60 second auto-exposures.
Does this help?

bbillp, 2021-08-14:
I’ll set debug to 3 Saturday night because today is Friday the 13th so the debug output will be corrupted.   My printer and microwave broke today.

bbillp, 2021-08-14:
V0.7 ASI178MC Pi3b+
autoexposure is 210
default ‘exposure’ = 10000
changed  ‘exposure’ to 15000
sudo service allsky stop, sudo service allsky start 
the starting images seem better …. I’ll adjust as needed following a fresh bootup.
Thanks ……

EricClaeys, 2021-08-14:
Brad,
If you are running the newest GUI, go to the "Camera Settings" page and at the bottom, click on the "Show Advanced Options" button, then change the "Debug Level" option.  We figured most people wouldn't use that option, so we hide it by default as an advanced option.
If you are NOT using the GUI (and I highly suggest you do), then edit the settings_ZWO.json file (I think that's what it's called), looking for the "debuglevel" setting.

Dentonknifeworks, 2021-08-14:
I am using the GUI but nothing shows up for the HQiPQ Advanced Options looks to be only for the ZWO. Advanced Options is there but nothing is added to the GUI when turned on.

EricClaeys, 2021-08-14:
Brad,
Debug level is in capture_RPiHQ.cpp, but I failed to put it in the settings file.  Ooops.
You can get it by adding the following lines to the camera_options_RPiHQ.json file (probably in /etc/raspap):

```
{
"name":"debuglevel",
"default":"0",
"description":"Debug level, 0 is lowest level.",
"label":"Debug Level",
"type":"select",
"options": [
	{"value": "0", "label": "0"},
	{"value": "1", "label": "1"},
	{"value": "2", "label": "2"},
	{"value": "3", "label": "3"}
],
"advanced":1
},
```

EricClaeys, 2021-08-14:
I just submitted an update for this to Thomas.  Thanks for pointing it out.

thomasjacquin, 2021-08-14:
@EricClaeys Did you send me a pull request for this?

EricClaeys, 2021-08-14:
@thomasjacquin Yes.  I submitted 15 pull requests last night.  Each was in its own branch.
5 pulls for allsky-portal and 10 for allsky.

thomasjacquin, 2021-08-14:
@EricClaeys Weird, I'm not seeing them on the GitHub page and I didn't get any notification.

EricClaeys, 2021-08-14:
Here's a screenshot showing some of them.  Could be my lack of GitHub knowledge and I did it incorrectly, but I created separate branches for each one, per your suggestion.

[image]

EricClaeys, 2021-08-14:
When I click on one of the pulls I get a page similar to this.  I did NOT do anything with the "Merge pull request" - should I ?

[image]

thomasjacquin, 2021-08-14:
@EricClaeys I think it's because you submitted a PR onto your own fork of my repository. Let me see if I can bring them over.

EricClaeys, 2021-08-14:
When I try to submit on yours, it tells me I don't have write permission, so I don't try.

thomasjacquin, 2021-08-14:
Can you select the base repository to push to? See item 4 on this page.
https://docs.github.com/en/github/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request-from-a-fork

EricClaeys, 2021-08-15:
Thomas,

You now have a bunch of PRs in allsky and allsky-portal.

Eric

 

https://docs.github.com/en/github/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request-from-a-fork

bbillp, 2021-08-15:
snip “bbillp, would you please set your Debug Level to 3 then restart allsky and upload the log? The WARNING you got implies the exposure time was set to -1. Thanks.”

V0.8 with debug =  3 , create dark images, one dark file created and the image folder 20210814 was created then saved sudo service allsky status to status.txt   Attached are the files for config.sh, settings_ZWO.json, status.txt, log.txt .  I was expecting to see a lot of info from debug level 3 in log.txt but there isn't .
.
[config.sh.txt](https://github.com/thomasjacquin/allsky/files/6987595/config.sh.txt)
[log.txt](https://github.com/thomasjacquin/allsky/files/6987596/log.txt)
[settings_ZWO.json.txt](https://github.com/thomasjacquin/allsky/files/6987597/settings_ZWO.json.txt)
[status.txt](https://github.com/thomasjacquin/allsky/files/6987598/status.txt)

bbillp, 2021-08-15:
Are Erics updates ready to download and update 0.8 on the Pi ?

EricClaeys, 2021-08-15:
Bill, my updates aren't there yet.  I'm having some problems with GitHub which is making it difficult to merge my changes into Thomas' master.

EricClaeys, 2021-08-15:
Bill,  sorry, I wasn't clear.  Post the **/var/log/allsky.log** file.  Maybe first stop the service, remove or truncate the file, then start the service and let it go a while, then upload allsky.log.  That way it will only contain the most recent data, which makes it easier to look at.
The log.txt file basically contains the arguments to allsky.sh and one entry for each picture taken.  It isn't very useful.
Eric

EricClaeys, 2021-08-15:
Bill, try the attached capture.cpp file - it should fix the -1 exposure for dark frames.

[capture.zip](https://github.com/thomasjacquin/allsky/files/6987740/capture.zip)

manfredwilhelm, 2021-08-15:
Same problem for me. I use a ZWO ASI120MC Cam. Worked with older version of allsky.
Raspberry PI 3 B+
Tried different USB-Settings or CAM-Modes (RAW...) No luck.

[allsky.log](https://github.com/thomasjacquin/allsky/files/6988493/allsky.log)

bbillp, 2021-08-15:
exhausting, i went thru all that time to generate and post debug 3.  I think I’ll wait and use 0.7 , the only issue with 0.7 is a warmer than normal camera.

EricClaeys, 2021-08-15:
Manfred,
If you're willing, in capture.cpp, please look for the line with ASI_HIGH_SPEED_MODE in it, then try two things:
1. Put "//" (without quotes) at the beginning of the line, save the file, run "make capture" from the /home/pi/allsky directory, and restart allsky.  Look at /var/log/allsky.log - you'll know within 30 seconds if it helped (i.e., look for the TIMEOUT errors).
2. If # 1 didn't help, remove the "//" from that line and change the "0" to "1", save the file, run "make capture", and restart allsky.  Again, look in the log file.
3. If neither of them helped, try increasing your "USB Bandwidth" to 90 or 100 via the GUI, and look in the log.
4. If still getting the TIMEOUT error, try setting the "Auto USB" to "Yes".

Let me know how it goes.  Since I changed my settings to USB speed 80 and Auto USB to "Yes", I haven't had any problems.

bbillp, 2021-08-16:
make capture.cpp has errors   in signed unsigned integer expressions.
if isuper(1{j

bbillp, 2021-08-16:
New Pull an hour ago now config.sh line 86 error !

EricClaeys, 2021-08-16:
That config.sh error is strange.  Somehow a few change control lines were added.
Anyway, I put in a request for Thomas to fix.
It should be:

```
# corrupt images which then generate funny colors in the summary images...
REMOVE_BAD_IMAGES=false

# Images whose mean brightness is less than THRESHOLD_LOW or over THRESHOLD_HIGH
```

I will fix the "isupper()" bug when I get my new Pi and can compile the code, bit I THINK the attached file should work.

Eric
[capture2.zip](https://github.com/thomasjacquin/allsky/files/6989496/capture2.zip)

bbillp, 2021-08-16:
Thanks for all your hard work coding but in my opinion TOO many changes are being applied with errors , I really think you need to slow down and post only a few changes and give those changes more time to test.  In the long past the updates were solid without errors but now Its getting very frustrating to experience one error after another with the latest updates.    Slow down ……..  Bill Prats

thomasjacquin, 2021-08-16:
Hi Bill,

Sorry about the errors that came up recently. Eric and a few others have made huge improvements to the source code and there have been a few conflicts when merging their code into the project. I will put in place a `dev` branch where new features will be merged and tested. Once in a while, the dev branch will be released into the `master` branch for everyone to enjoy.
I will keep merging hot fixes if there are outstanding errors with the new versions but I will keep new features on the side for now.
I believe that's the best way to manage the evolution of the source code going forward.

Thomas

bbillp, 2021-08-16:
I’ve had a lot of fun with Allsky since the earliest versions, thanks for all the hard work.  Yes creating a Development branch would be an excellent idea, something we did in Payroll and Inventory program development where errors could cause large and embarrassing problems in a company.      I’ll send some more coffee money next pay day, hopefully to include a donut ,  best regards    ….. :o))

thomasjacquin, 2021-08-16:
Thanks for understanding Bill.
Hang tight, I found an issue with saveImageDay.sh. I will have to test and push again to master. Hopefully that's one of the last hot fixes.

EricClaeys, 2021-08-16:
Thanks for understanding Bill.  Four of us were using the bulk of the big batch of changes for a couple months with no problems.  We all had the TIMEOUT issue that we resolved by using higher USB Bandwidth values, and in one case, by moving the camera to the USB 3 port.  So having users still have that problem after making those changes has stumped me.  I emailed ZWO support about it yesterday.

I will admit that the updates since the big batch haven't been tested as well, and I appologize for that.  I bought another Pi that should arrive today, so I can use that for testing.  It will have the RPiHQ camera.  I might buy another ZWO camera for testing since most people use them, and because the allsky camera in my remote observatory stopped working the day after I left and I won't be back for a few months.

Eric

bbillp, 2021-08-17:
Thanks for the update on the changes Eric.   What is the best way to install the new updates, one file at a time or just start from scratch and bulk reload using the allsky installation instructions which I have been doing.    Bill

EricClaeys, 2021-08-17:
Bill,

If you are all caught up on the updates, going forward I would suggest just downloading the changed files and replacing them on your Pi.  Probably helpful to keep a copy of ~/allsky (minus the /images folder) as a backup, in case something doesn’t work, or if you accidently delete a file.  I keep a copy on my NAS.  It takes up almost no space, and could easily be zipped.

 

The only bug I’m aware of is the dang TIMEOUT issue.  The other changes I want to make are small improvements and ways to help people debug problems (a lot of people seem to have problems with uploading, even with version 0.7).  With Thomas’ permission I want to create a Wiki with troubleshooting and other tips.

 

Eric

bbillp, 2021-08-17:
I am almost sure the timeout error is a ZWO firmware or ZWO SDK issue with timing because of the random nature of the beast.   The ZWO ASI120 has a firmware update that addresses packet size which could be a clue to the ASI178 timing issue.  To address the timing issue I set the JPG quality down from 95 to 10 which normally results in a very small image file that would transfer rapidly thru the USB stream but no the timeout still occurs so the question is does the time out error occur with Control Data and well as Image data .  If you could write a looping routine testing for timeout on control data with no image data the results would help determine where the root cause is.    Only ZWO can provide a fix for such a condition.

Whats all the talk about using a Swap file?   I can’t imagine why anyone would turn it off and Isn’t a swap file automatically setup by the Linux OS.   I have RPi running ham radio software defined receivers and telescope management software like Astroberry and have not heard issues with the swap file.  

I always keep a backup SD chip for 0.7 so nothing is lost and both 7  & 8 are backed up to the PC server.    I use FTP in manual mode to upload because here along the Pacific beach many nights are not worth saving.   

Anyway the Allsky will have to wait for a while as I work on other projects …. I’ll continue to read the posts ….. good luck …. Bill P.

EricClaeys, 2021-08-17:
Bill,

 

My guess is also that it’s a ZWO issue, just like I’m 99.9999% sure the daytime autoexposure issues is a ZWO bug.  Luckily it’s on their “to do” list.  I haven’t heard back regarding my email about the TIMEOUT issue.  Unfortunately I don’t really have any data to give them, other than “some people see it”.  I’ve asked several people who don’t appear to have the problem what settings they are using but never hear back.  I’d like to know which cameras have the problem, and which don’t, along with the settings used.  All I know for sure is the 4 of us who tested all have it working and I’m pretty sure 3 of us have the ASI178MC or MM.  I am going to buy a new camera tomorrow for testing, and I hope it has the problem so I can test the heck out of it.

The only thing in the 0.8 code I can think of that would cause the issue is that we start the video mode, take a picture, then stop the vide mode for ever picture.  It used to start video mode when the program started and left it on until the program ended.  The new way is what produces a cooler camera – it’s only doing something when taking a picture.

I can always go back to the 0.7 version and add 0.8 features one-by-one and see when it breaks, but if my new camera doesn’t have the issue I’d need to find someone who’s willing to help, and it could take a fair amount of their time.

 

I don’t have a Pi 3 so am not sure if it comes with a swap file.  Even if it does and it’s the same size as the Pi 4 swap file, since the Pi 4 has more RAM it needs the swap file less.  Pi Zero and any Pi less than 4 have less memory so will likely see the issue.  I have only heard of issues with timelapse which uses ffmpeg which I assume uses a lot of memory converting potentially thousands of images into a video.

Anyway, as I hear of issues I like to document the fixes so 1) I don’t forget, and 2) others can learn from them.

 

Do you know how I can write a looping routine testing for timeout on USB control data?

Have you ever noticed the /var/log/syslog (or one of those) has a gazillion USB reset messages?  I wonder if those are related…

 

Eric

pierrebilb, 2021-08-17:
How can we do to install v0.7?
Thanks
Pierre

bbillp, 2021-08-17:
I can’t help until next week due to my job schedule.   This Pi-3b easily creates videos from allsky and another video generator I don’t have in front of me right now.   

I have to run but here is a very small flow

Initialize
Main
send control stream
check for errors
if error output to standard out
Else Loop Main 

-or-

Do main while not error
output error to standard out
main
send usb commands
end main

pierrebilb, 2021-08-20:
Hi,
is there any way to get back to v0.7?
I still get the asi error timeout whatever parameter i put on v0.8 while it was working straightforward on v0.7...
Thanks 
Pierre

EricClaeys, 2021-08-20:
Let me see if I can find a copy of version 0.7 to send you.
What camera do you have, and it attached to a USB 2 or 3 port?  And the top or bottom port?

Eric

thomasjacquin, 2021-08-20:
You can select the version you want on github.
https://github.com/thomasjacquin/allsky/tree/v0.7

EricClaeys, 2021-08-21:
Thomas,

This gives the original 0.7 files, not the ones as they existed right before the 0.8 upgrades.  Is it possible to select the pre-0.8 files?

 

Eric

 

https://github.com/thomasjacquin/allsky/tree/v0.7

thomasjacquin, 2021-08-21:
Browse the commits and hit the `<>` button to get a snapshot of the code at a certain time.
https://github.com/thomasjacquin/allsky/commits/

I believe this is where the repo was before you committed all the new v0.8 features.
https://github.com/thomasjacquin/allsky/tree/cdd7c1e3b332bb3b73361be04e34d94ca502a737

You will have to apply the same logic for the allsky-portal. 
https://github.com/thomasjacquin/allsky-portal/tree/c7b732c4e2bd5c85bc4839c3c8388cd346213102

manfredwilhelm, 2021-08-21:
Hi Eric, sorry, i've been busy at work. I will try to do the changes this
weekend.

EricClaeys, 2021-08-21:
Manfred,
Thanks.  Attached is a newer capture.cpp to use, in capture0.8.zip

**pierrebilb and others,**
Also attached is the 0.7 version of capture.cpp that knows about the new 0.8 setting names.  If you use it, you will still get many of the other benefits of 0.8, but with the 0.7 capture, and don't need to back anything out.
To use it, rename your existing capture.cpp to capture0.8.cpp, then rename capture0.7.cpp to capture.cpp, and run "make capture".

[capture0.8.zip](https://github.com/thomasjacquin/allsky/files/7026134/capture0.8.zip)
[capture0.7.zip](https://github.com/thomasjacquin/allsky/files/7026135/capture0.7.zip)

mhammady911, 2021-08-22:
@EricClaeys you saved the day. Thank you. Now my 0.8 version is working without ASI_TIMEOUT_ERROR

Thank you

EricClaeys, 2021-08-23:
Glad it works, mhammady911.  I've been running it on my test Pi for a couple days with no issues, although as soon as I started running I noticed the ZWO daylight exposure bug - one shot was light, the next couple dark, then light, etc.  Version 0.8 fixed that.

I'd like to slowly include additional version 0.8 features into the modified 0.7 capture.cpp file until the ASI_TIMEOUT_ERROR returns, to help determine where the problem is.  Would you be willing to help?  You'd just need to save the current capture.cpp to a temporary name, then install the new one and recompile and let me know if you see the problem.  It tends to happen pretty quickly.

Unfortunately I don't see the problem on either of my two cameras so I have no way to test it myself.

Thanks - Eric

mhammady911, 2021-08-23:
Hi @EricClaeys, sure, I'd be happy to help. Keep sending me the updated capture.cpp and will test them

pierrebilb, 2021-08-25:
Hi,
also working for me! I will try the updated capture.cpp as they come. I have an orange pi+2e with a t7 color on a usb2 port.

EricClaeys, 2021-08-25:
Thanks in advance for helping, mhammady911 and pierrebilb.
I have a hunch the TIMEOUT error is related to the way version 0.8 handles exposures.
Version 0.7 would turn on video mode, then take a picture, sleep, take another, sleep, etc., going on until you stopped the program or rebooted the Pi.  That meant the camera was constantly taking pictures as fast as it can (which it does in video mode), which would heat up the camera.  Version 0.8 starts video mode, takes a picture, stops video mode, then sleeps until the next picture, restarting that sequence.  So the camera was only heating up when it was taking pictures, which was dependent on how long a delay you had.
For the next test version I think I'll add ALL the other version 0.8 enhancements EXCEPT the video mode one.  If that works, we'll know where the issue is, then I can try to find ways around it.
Eric

mhammady911, 2021-08-27:
Hi Eric, 

1. the "Version 0.8 Exposure" option present only in the json file and not the UI.
2. As soon as I save the camera settings, it override the new json with original structure
3. I didnt change the settings of .json to make sure i use the 0.8 exposure settings

the following is the log output

`Aug 27 19:39:13 allskycam01 allsky.sh[4490]: === Starting daytime capture ===
Aug 27 19:39:13 allskycam01 allsky.sh[4490]: ==========
Aug 27 19:39:13 allskycam01 allsky.sh[4490]: Saving images with delay of 5,000 ms (5 sec)
Aug 27 19:39:13 allskycam01 allsky.sh[4490]: WARNING: Value of 200 greater than max value allowed (100) for control 'AutoExpMaxGain' (#10).
Aug 27 19:39:17 allskycam01 allsky.sh[4490]: STARTING EXPOSURE at: 2021-08-27 19:39:17
Aug 27 19:39:17 allskycam01 allsky.sh[4490]:   > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
Aug 27 19:39:19 allskycam01 allsky.sh[4490]: STARTING EXPOSURE at: 2021-08-27 19:39:19
Aug 27 19:39:20 allskycam01 allsky.sh[4490]:   > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
Aug 27 19:39:22 allskycam01 allsky.sh[4490]:   > Saving DAY image 'image.jpg' that started at 2021-08-27 19:39:19
Aug 27 19:39:22 allskycam01 allsky.sh[4490]: jq: error (at /etc/raspap/settings_ZWO.json:592): Cannot index array with string "filename"
Aug 27 19:39:22 allskycam01 allsky.sh[4490]: jq: error (at /etc/raspap/settings_ZWO.json:592): Cannot index array with string "darkframe"
Aug 27 19:39:22 allskycam01 allsky.sh[4490]: jq: error (at /etc/raspap/settings_ZWO.json:592): Cannot index array with string "filename"
Aug 27 19:39:22 allskycam01 allsky.sh[4490]: *** copy_notification_image.sh: ERROR: File 'notification_images/NotRunning.' does not exist or is empty!
Aug 27 19:39:22 allskycam01 allsky.sh[4490]: jq: error (at /etc/raspap/settings_ZWO.json:592): Cannot index array with string "notificationimages"
`

[image]
[image]

EricClaeys, 2021-08-28:
mhammady911,
You copied the camera_options_ZWO.json file to /etc/raspap, correct?  And that file has the "newexposure" setting as shown in your screenshot?  What does config.sh have for the settings directory (it'snear the bottom)? /etc/raspap?  If so, then the server isn't getting the updated camera_settings_ZWO file.
After you install that file you need to refresh your browser window before pressing the "Save changes" button.

Which jason file are referring to:
`I didnt change the settings of .json to make sure i use the 0.8 exposure settings`

The jq errors imply something wrong with the settings_ZWO.json file.  Let's not worry about those until you can see the Version 0.8 option in the GUI.

A minor thing but your camera only supports a maximum gain of 100, not 200.
Eric

PrimeAmp2016, 2021-08-30:
May not fix the Issue described but for information: I loaded exactly the same 0.8 install on a Pi Zero W and a PI 4 8GB.  It worked first time on PI 4 using a ZWO ASI120MC-S but not on PI Zero. After a trying a few variations of the advanced USB settings in the Camera Config to no avail I connected the ZWO via a ANKER powered USB Hub and the ASI_ERROR_TIMEOUT has gone and camera working fine.

[107 more replies]

closed 2021-10-14 by EricClaeys