AllskyTeam/allsky #420: Timelapse MP4 not completed at end of night

ShianMull, 2021-08-13:
Hi,

Would anyone be able to point me in the right direction to solve the Timelapse generated being only 4kB and not containing any image sequence?  It seems to have been generated at the right time looking at the file timestamp, but just nothing in the file to view.

Keograms, startrails and general images are saved and viewable, so I'm just a little stuck on the reason the timelapse isn't working.

All the best

Mark

EricClaeys, 2021-08-13:
Mark,
Try running it manually to see what happens.  If you have a model 3 you may need to use swap space.

To run:  scripts/timelapse.sh date        where "date" is the day you want to process, e.g., 20210811

bbillp, 2021-08-13:
Windows ‘preview’ each image to find and delete a bad one that prevents mp4 creation.

EricClaeys, 2021-08-14:
If you do find a bad one that prevents mp4 creation, please upload it before deleting it.  I want to see if "REMOVE_BAD_IMAGES" would catch it.

ShianMull, 2021-08-14:
Thanks Eric and Bill,

All images look fine and appear to be 'good' jpegs.

I ran the script manually and still no joy, but looking at the output window of the movie process, I see the following output (just final processing section included) with a possible issue (highlighted in the output text) that I can't relate to the camera or script settings - does this look normal?  But also, there's a lot of 0 bytes in the frame selection, which is puzzling as well?

Many thanks

Mark

Input #0, image2, from 'images/20210813/sequence/%04d.jpg':
  Duration: 00:00:51.00, start: 0.000000, bitrate: N/A
    Stream #0:0: Video: mjpeg, yuvj420p(pc, bt470bg/unknown/unknown), 3096x2080 [SAR 1:1 DAR 387:260], 25 fps, 25 tbr, 25 tbn, 25 tbc
Stream mapping:
  Stream #0:0 -> #0:0 (mjpeg (native) -> h264 (libx264))
Press [q] to stop, [?] for help
**[swscaler @ 0x1c8f3b0] deprecated pixel format used, make sure you did set range correctly**
[libx264 @ 0x1c82f80] using SAR=1/1
[libx264 @ 0x1c82f80] using cpu capabilities: ARMv6 NEON
[libx264 @ 0x1c82f80] profile High, level 5.1
[libx264 @ 0x1c82f80] 264 - core 155 r2917 0a84d98 - H.264/MPEG-4 AVC codec - Copyleft 2003-2018 - http://www.videolan.org/x264.html - options: cabac=1 ref=3 deblock=1:0:0 analyse=0x3:0x113 me=hex subme=7 psy=1 psy_rd=1.00:0.00 mixed_ref=1 me_range=16 chroma_me=1 trellis=1 8x8dct=1 cqm=0 deadzone=21,11 fast_pskip=1 chroma_qp_offset=-2 threads=6 lookahead_threads=1 sliced_threads=0 nr=0 decimate=1 interlaced=0 bluray_compat=0 constrained_intra=0 bframes=3 b_pyramid=2 b_adapt=1 b_bias=0 direct=1 weightb=1 open_gop=0 weightp=2 keyint=250 keyint_min=25 scenecut=40 intra_refresh=0 rc_lookahead=40 rc=abr mbtree=1 bitrate=2000 ratetol=1.0 qcomp=0.60 qpmin=0 qpmax=69 qpstep=4 ip_ratio=1.40 aq=1:1.00
Output #0, mp4, to 'images/20210813/allsky-20210813.mp4':
  Metadata:
    encoder         : Lavf58.20.100
    Stream #0:0: Video: h264 (libx264) (avc1 / 0x31637661), yuv420p, 3096x2080 [SAR 1:1 DAR 387:260], q=-1--1, 2000 kb/s, 25 fps, 12800 tbn, 25 tbc
    Metadata:
      encoder         : Lavc58.35.100 libx264
    Side data:
      cpb: bitrate max/min/avg: 0/0/2000000 buffer size: 0 vbv_delay: -1
frame=    2 fps=0.0 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=    3 fps=2.1 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=    4 fps=2.1 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=    5 fps=2.1 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=    6 fps=2.1 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=    8 fps=2.0 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=    9 fps=2.0 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   10 fps=2.0 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   12 fps=2.0 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   13 fps=2.0 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   15 fps=2.0 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   16 fps=2.0 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   17 fps=2.0 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   18 fps=2.0 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   19 fps=2.0 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   20 fps=1.9 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   21 fps=1.9 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   22 fps=1.9 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   23 fps=1.9 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   24 fps=1.9 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   25 fps=1.9 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   26 fps=1.9 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   28 fps=1.9 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   29 fps=1.9 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   30 fps=1.9 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   31 fps=1.9 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   32 fps=1.9 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   33 fps=1.9 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   34 fps=1.9 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   35 fps=1.9 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   36 fps=1.9 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   37 fps=1.9 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=   38 fps=1.9 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   ./timelapse.sh: line 45: 24594 Killed                  ffmpeg -y -f image2 -r $FPS -i images/$1/sequence/%04d.$EXTENSION -vcodec libx264 -b:v 2000k -pix_fmt yuv420p -movflags +faststart $SCALE images/$1/allsky-$1.mp4
* Deleting sequence
* Timelapse was created

EricClaeys, 2021-08-14:
The "deprecated pixel format" message can be ignored.  The 0's are troublesome.

It looks like you are running the older verdion 0.7 release?

Create a directory in allsky/images called "test" and copy three images to it.  Then run "scripts/timelapse.sh test" and see what you get.
In the output should be messages for the "ln -sv" commands for each input file.  Are you seeing those?

ShianMull, 2021-08-14:
Thanks Eric,

It does appear I'm running 0.7 - this was just the install from the Github instructions that I followed, so I may need an update?  Forgive me on this as I'm not really very familiar with Raspberry Pi applications.

I ran 'test' with just 3 images and an MP4 was created.  It seems all three images were in the timelapse video, and the output was as follows:

Input #0, image2, from 'images/test/sequence/%04d.jpg':
  Duration: 00:00:00.12, start: 0.000000, bitrate: N/A
    Stream #0:0: Video: mjpeg, yuvj420p(pc, bt470bg/unknown/unknown), 3096x2080 [SAR 1:1 DAR 387:260], 25 fps, 25 tbr, 25 tbn, 25 tbc
Stream mapping:
  Stream #0:0 -> #0:0 (mjpeg (native) -> h264 (libx264))
Press [q] to stop, [?] for help
[swscaler @ 0x152f6b0] deprecated pixel format used, make sure you did set range correctly
[libx264 @ 0x15277f0] using SAR=1/1
[libx264 @ 0x15277f0] using cpu capabilities: ARMv6 NEON
[libx264 @ 0x15277f0] profile High, level 5.1
[libx264 @ 0x15277f0] 264 - core 155 r2917 0a84d98 - H.264/MPEG-4 AVC codec - Copyleft 2003-2018 - http://www.videolan.org/x264.html - options: cabac=1 ref=3 deblock=1:0:0 analyse=0x3:0x113 me=hex subme=7 psy=1 psy_rd=1.00:0.00 mixed_ref=1 me_range=16 chroma_me=1 trellis=1 8x8dct=1 cqm=0 deadzone=21,11 fast_pskip=1 chroma_qp_offset=-2 threads=6 lookahead_threads=1 sliced_threads=0 nr=0 decimate=1 interlaced=0 bluray_compat=0 constrained_intra=0 bframes=3 b_pyramid=2 b_adapt=1 b_bias=0 direct=1 weightb=1 open_gop=0 weightp=2 keyint=250 keyint_min=25 scenecut=40 intra_refresh=0 rc_lookahead=40 rc=abr mbtree=1 bitrate=2000 ratetol=1.0 qcomp=0.60 qpmin=0 qpmax=69 qpstep=4 ip_ratio=1.40 aq=1:1.00
Output #0, mp4, to 'images/test/allsky-test.mp4':
  Metadata:
    encoder         : Lavf58.20.100
    Stream #0:0: Video: h264 (libx264) (avc1 / 0x31637661), yuv420p, 3096x2080 [SAR 1:1 DAR 387:260], q=-1--1, 2000 kb/s, 25 fps, 12800 tbn, 25 tbc
    Metadata:
      encoder         : Lavc58.35.100 libx264
    Side data:
      cpb: bitrate max/min/avg: 0/0/2000000 buffer size: 0 vbv_delay: -1
frame=    2 fps=0.0 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   frame=    3 fps=2.5 q=0.0 size=       0kB time=00:00:00.00 bitrate=N/A speed=   [mp4 @ 0x151ffd0] Starting second pass: moving the moov atom to the beginning of the file
frame=    3 fps=0.5 q=-1.0 Lsize=      65kB time=00:00:00.00 bitrate=6856000.0kbits/s speed=1.4e-05x    
video:64kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 1.347848%
[libx264 @ 0x15277f0] frame I:1     Avg QP:28.85  size: 48688
[libx264 @ 0x15277f0] frame P:1     Avg QP:29.85  size: 13051
[libx264 @ 0x15277f0] frame B:1     Avg QP:32.42  size:  3513
[libx264 @ 0x15277f0] consecutive B-frames: 33.3% 66.7%  0.0%  0.0%
[libx264 @ 0x15277f0] mb I  I16..4: 38.9% 60.2%  0.9%
[libx264 @ 0x15277f0] mb P  I16..4:  7.7% 10.6%  0.0%  P16..4: 10.3%  0.8%  3.0%  0.0%  0.0%    skip:67.6%
[libx264 @ 0x15277f0] mb B  I16..4:  0.1%  0.0%  0.0%  B16..8: 13.9%  0.1%  0.0%  direct: 0.1%  skip:85.8%  L0:14.7% L1:84.8% BI: 0.5%
[libx264 @ 0x15277f0] final ratefactor: 36.50
[libx264 @ 0x15277f0] 8x8 transform intra:59.8% inter:99.7%
[libx264 @ 0x15277f0] coded y,uvDC,uvAC intra: 10.9% 28.9% 3.4% inter: 1.0% 4.2% 0.0%
[libx264 @ 0x15277f0] i16 v,h,dc,p: 41% 30%  7% 22%
[libx264 @ 0x15277f0] i8 v,h,dc,ddl,ddr,vr,hd,vl,hu: 36% 15% 32%  5%  2%  1%  1%  2%  6%
[libx264 @ 0x15277f0] i4 v,h,dc,ddl,ddr,vr,hd,vl,hu: 17% 24% 19%  9%  5%  3%  4%  3% 14%
[libx264 @ 0x15277f0] i8c dc,h,v,p: 73% 16%  9%  2%
[libx264 @ 0x15277f0] Weighted P-Frames: Y:100.0% UV:100.0%
[libx264 @ 0x15277f0] kb/s:4350.13
* Deleting sequence
* Timelapse was created

I then tried adding a further 30 images to test.  Running the script again produced the same result as my original problem.

So, I then tried 5 and 10 frames, which worked!  Making the total frames 20-off broke it again.

I still see 0 bytes for even the apparently successful lower number of frames in the process output, so there is something a little odd here.

Many thanks for your help.

Cheers

Mark

EricClaeys, 2021-08-14:
Mark,
I suspect the ffmpeg command is running out of memory and you'll need to add swap space, but just to make sure it's not a bad image causing the problem, take the 10 images that worked and copy them a couple times so you have 30 images that you know work.  Then try running it.  If it fails, it's not the images themselves.  Google how to add swap space.  Some people on the AllSky forum have mentioned it but I didn't pay attention.

bbillp, 2021-08-15:
I didn't read all the messages for this problem but like all Linux/Unix,
everything is case sensitive.  Make sure your image output is in lower
case, jpg and not JPG, The FFMPEG command is looking for lower case jpg
files only.     This was a problem I experienced long ago.  Bill

EricClaeys, 2021-08-15:
In the next update to capture.cpp I'll convert uppercase extensions to lowercase, e.g., "image.JPG" to "image.jpg".  That will eliminate the issue with ffmpeg.

ShianMull, 2021-08-15:
Thanks Eric,

Increasing the swap size worked perfectly, so it now generates whole night timelapse.  It does take over an hour to generate though!!

As far as I can see, the capture code is correctly saving .jpgs <all lower case>, but it is a good idea to trap all logical text case possibilities.

Many thanks for your help on this - for anyone else reading this with a Pi3 Model B, setting the swap file to 2048 Mb works well, but I wouldn't set it higher than this due to speed and memory card limitations.

All the best

Mark

closed 2021-08-15 by ShianMull

EricClaeys, 2021-08-15:
Mark,
Glad the change fixed your problem.

Do you have a link to directions on how to set the swap space?  I would like to include them in the README.md file.

ShianMull, 2021-08-15:
Hi Eric,

Yes, I used this procedure and it worked well:

https://pimylifeup.com/raspberry-pi-swap-file/

All the best

Mark