AllskyTeam/allsky #421: Cannot remove allsky directory

Lorenzi70, 2021-08-14:
Hi all, I am trying to remove the directory allsky to install a new version but I keep getting an error message:
pi@allsky:~ $ rm –rf allsky
rm: cannot remove '–rf': No such file or directory
rm: cannot remove 'allsky': Is a directory
What am I doing wrong?
Thanks for any reply provided
Marco

thomasjacquin, 2021-08-14:
Hi Marco, my guess is that the dash sign you’re using is not the same as the minus sign. Try using the minus character in front of the rf option.

Lorenzi70, 2021-08-15:
you are right, mind me! Thanks, it works now

EricClaeys, 2021-08-16:
Marco, would you please close the issue if it's resolved?

Just curious - how did you input that character?  I assume you are using a non-English keyboard that has that caracter?

Eric

Lorenzi70, 2021-08-17:
Closed now Eric. I am using a small keyboard instead of the usual one, no idea how did I used a wrong char, probably not the right settings.. whatever, it works now :)

closed 2021-08-17 by Lorenzi70