AllskyTeam/allsky #422: Problem to make DARK Frames

Amenophis54, 2021-08-14:
Hello, 

I am embarking on this adventure with an ASI120MC-S USB3.0 camera.
(Raspberry pi 4 serie B 8Go / SD 32Go with allsky portal).

For the moment, I have managed to install everything and make it work, but I have a problem with the DARK frames.

Indeed, after disconnecting the camera, modifying the parameters by activating the capture of DARK, then by reactivating the camera, I have the message "Currently capturing dark frames. You can turn this off on the camera settings page." in the "live view" tab.

After more than an hour of waiting, nothing happens. I still have the same message. CAMERA IS TAKING DARK FRAMES.

Thank you in advance for your help.

EricClaeys, 2021-08-14:
Is your camera taking pictures?
Did you refresh the web browser for the GUI?
You don't need to disconnect the camera unless it's to take it off and cover it.  After you were done taking darks, did you reset the variable in config.sh (DARKFRAME ?) To false, then restart allsky?

EricClaeys, 2021-08-14:
Sorry, the configuration setting is called " Dark Frame" and it's on the GUI Camera Settings page.

bbillp, 2021-08-15:
To restart allsky to read the configuration file changes you only have to
do the following.
sudo service allsky stop
sudo service allsky start
OR
sudo service allsky restart

Amenophis54, 2021-08-15:
I succeeded by going to modify the config.sh file in the editor menu.

It was necessary to put "true" at the level

# Set to true if you want to enable dark frame subtraction on your night time images
DARK_FRAME_SUBTRACTION=true

EricClaeys, 2021-08-15:
There are two settings related to dark frames.  To TAKE dark frames you set the "Dark Frame" option on the GUI to Yes.
Once done taking them, set it to No.  Then, to USE the dark frames, set DARK_FRAME_SUBTRACTION=true in config.sh.

If you have DARK_FRAME_SUBTRACTION=true but don't actually have any dark frames, you'll get a message in /var/log/allsky.log.

linuxkidd, 2021-10-04:
Sounds like this issue is resolved and things are working normally.  

We do aim to have options easier to edit and better explanations in the future via a unified config / all editable via the WebUI ( vs using the editor ).  Keep an eye on [this project](https://github.com/thomasjacquin/allsky/projects/3) for more updates.

closed 2021-10-04 by linuxkidd