AllskyTeam/allsky #435: Error in allsky log file

maserowik, 2021-08-16:
I did a new git pull of the allsky program and seeing a error in the log file.  

Aug 15 20:59:50 Pi-AllSky allsky.sh[887]: /home/pi/allsky/config.sh: line 86: syntax error near unexpected token `<<<'
Aug 15 20:59:50 Pi-AllSky allsky.sh[887]: /home/pi/allsky/config.sh: line 86: `<<<<<<< HEAD'

looking at the config.sh file i see in the config.sh file

 83 # Scan for, and remove corrupt images before generating keograms and startrails.
 84 # This can take several (tens of) minutes to run and isn't necessary unless your system produces
 85 # corrupt images which then generate funny colors in the summary images...
 86 <<<<<<< HEAD
 87 REMOVE_BAD_IMAGES=false
 88 =======
 89 REMOVE_BAD_IMAGES=true
 90 >>>>>>> origin/master

i see the REMOVE_BAD_IMAGES=false and then two down had the same variable set to true.  

do the <<<<<<<<, =========, >>>>>>> need to have a # before the command?
also which one is correct? false or true?

Mike

EricClaeys, 2021-08-16:
Mike,
I'm not sure how that happened, but some change control lines were added.  I put in a request for Thomas to fix, but in the meantime, you can change those lines to:

```
# corrupt images which then generate funny colors in the summary images...
REMOVE_BAD_IMAGES=false

# Images whose mean brightness is less than THRESHOLD_LOW or over THRESHOLD_HIGH
```

I changed the default to NOT remove bad images, since it takes tens of minutes and few people will need it.  That was also the default in the prior version.

closed 2021-08-16 by maserowik

EricClaeys, 2021-08-16:
The updated version of confid.sh.repo has been posted.