AllskyTeam/allsky #437: Hosting website on Pi3 but video and images not saved to var/www/html/allsky folders

ShianMull, 2021-08-16:
Hi All,

I'm puzzling the last issue before I package the allsky camera and put it outside,  I have installed Thomas's website on the Pi3 I'm using and made appropriate changes to ftp script, but the compiled images and timelapse etc are not copied to the html folders.  The live view is also not updated and I'm scratching my head here and the scripts associated with this function seem to be working fine.

Has this issue been seen before and resolved - I couldn't see anything on the closed issues.

Many thanks for your help.

Mark

EricClaeys, 2021-08-16:
Mark,
Are you using allsky version 0.8 and allsky-portal version 0.7?  If so, please post your /home/pi/allsky/config, /home/pi/allsky/scripts/ftp_settings.sh, and /etc/raspap/*ZWO.json files.  The problem is probably in your settings.

When you say "Thomas's website", what is the URL?

Eric

ShianMull, 2021-08-16:
Hi Eric,

**/home/pi/allsky/config:**

#!/bin/bash

# Choose between ZWO or RPiHQ or auto (prefers ZWO if present)
CAMERA="auto"

# Set to true to upload current image to your website
UPLOAD_IMG=true

# Set to true to upload timelapse video to your website at the end of each night
UPLOAD_VIDEO=true

# Set to true to send data to your server at the end of each night
POST_END_OF_NIGHT_DATA=true

# Set to true to generate a timelapse at the end of the night
TIMELAPSE=true

# Set the width and height parameter for timelapse generation (sizes must be divisible by 2)
TIMELAPSEWIDTH=0
TIMELAPSEHEIGHT=0

# Timelapse frame rate (number of frames per second)
FPS=25

# Set to true to generate a keogram at the end of the night (image summary of the night)
KEOGRAM=true

# Set to true to upload the keogram after it has been generated
UPLOAD_KEOGRAM=true

# Set to true to generate a startrails image of the night. Will skip brighter images to avoid over-exposure
STARTRAILS=true

# Images with a brightness higher than the threshold will be skipped for startrails image generation
BRIGHTNESS_THRESHOLD=0.1

# Set to true to upload the startrails after it has been generated
UPLOAD_STARTRAILS=true

# Set to true to enable automatic deletion of archived data (images + videos)
AUTO_DELETE=true

# Set this value to the number of archived nights you want to keep. Needs AUTO_DELETE=true to work
NIGHTS_TO_KEEP=7

# Set to true if you want to enable dark frame subtraction on your night time images
DARK_FRAME_SUBTRACTION=false

# Set to 0 to disable Daytime Capture
DAYTIME="1"

# Set 24Hr capture to true to save both night and day images to disk. By default, only night images are saved.
CAPTURE_24HR=false

# Resize images before cropping and saving. Adjust width and height according to your own sensor ratio
IMG_RESIZE=false
IMG_HEIGHT=1520
IMG_WIDTH=2028

# Crop the captured image, used to improve the images when using a fisheye lens
CROP_IMAGE=false
CROP_WIDTH=640
CROP_HEIGHT=480
CROP_OFFSET_X=0
CROP_OFFSET_Y=0

[... 346 lines cut ...]
"label":"Time",
"type":"checkbox"
},
{
"name":"darkframe",
"default":"0",
"description":"Activate to capture a dark frame. This mode hides text and time overlays",
"label":"Dark Frame",
"type":"checkbox"
},
{
"name":"showDetails",
"default":"1",
"description":"Activate to display the image metadata on the overlay",
"label":"Show Details",
"type":"checkbox"
}

]

I meant Thomas’s default website that is copied onto the local RasPi as part of the install.sh - I think it just ports the website over.

All the best

Mark

ShianMull, 2021-08-18:
Hi Eric,

I realised I had made an error in the FTP_settings.sh and included “spaces” in between the “=“ and the directory names - rectified and now Startrails and Keograms upload fine.

The live view is also uploading to var/www/html/allsky OK, but the file name is “image-resize.jpg”, which isn’t being displayed on the default webpage instant view.

There are also no time lapse videos being saved to the default MP4 directly for the website, so I suspect there is a file name/directory mismatch for the video and the live image.

Would you be able to point me in the right direction to check what name the website uses for the live image, and any thoughts on why the time lapse isn’t uploading?

Many thanks for your help.

Mark

EricClaeys, 2021-08-18:
Mark,
There are two websites:
1. The GUI site where you can display rhe "Liveview", change camera settings, and do other things.  This is the "allsky-portal"Git package.  Dou you see the Live image?  If not, right click on the little icon and open in a new window and look at the URL to see what it is looking for.
2. The website where you can view a constellation overview.  This is Git "allsky-website".  Do you see this image?  If not, do the same as above.  Also modify the filename in /var/www/html/allsky/config.js.

ShianMull, 2021-08-18:
Thanks Eric,

GUI website is working fine, no problem with live view.

I’ve already tried editing config.js to change “image.jpeg” to “image-resize.jpeg” (the posted file that seems to update as expected as a file), but this hasn’t resulted in the live image being displayed on the Pi hosted website (with the constellation view)…

The other issue is that the Timelapse video (<datename>.mp4) is not being copied to the var/www/html/allsky/videos folder.  Keogram and Startrial pictures are being copied fine, so I suspect it is also the video file name causing issues on this.

Many thanks for your help.

Mark

ShianMull, 2021-08-18:
Update before the bedtime!  I have edited both website script and timelapse.sh to use a consistent name.  Not sure why I have scripts that use differing file names to each other, but it must have been down to installing website separately from GUI and original install script.  Will update if successful, but I think your pointers, Eric, have helped me identify the salient issues - many thanks.

Mark

ShianMull, 2021-08-20:
Everything except the copy of time lapse works now, but my recommendation is that the website is not hosted from the Pi (Pi4 might be more capable), as it probably puts more workload on the flash drive than necessary.

The issue I have found is tracking what files are saved where under which config file when using GUI vs command prompt.  I guess one needs to keep up to date on the various updates, so my suggestion to anyone making this excellent project is that it’s worth making your own note of what is installed where and how it’s edited to keep on top of things 😊

closed 2021-08-20 by ShianMull