AllskyTeam/allsky #439: Possible to add web interface in addition to the portal interface?

Amenophis54, 2021-08-17:
Hello, 
Thank you very much for all the hard work and the help found here :)
Is it possible to add, in particular the option of displaying constellations on the portal interface? If yes, how ?

I would like to add my allsky on my website (managed by WIX) but it seems to me to have read that WIX does not take into account the FTP protocol.

EricClaeys, 2021-08-17:
Hi,
If you are wanting to run the "allsky-website" application on a different computer than your Pi, you'll obviously need to be able to get the files there somehow.  It would be unusual for a web hosting service not to support FTP or SFTP.  How do they suggest you put files on the web host?

Using "allsky-website" it's very easy to display constellations - just click the Cassiopeia-like icon on the left.  I suggest installing it on your Pi first so you can become familiar with it, then port it to your WIX server.

Amenophis54, 2021-08-18:
Thank you Eric. 
Actually, I just wanted to add the constellation option on the "allsky-portal" GUI.

But I believe that is not possible. You have to completely reinstall the Thomas "allsky-web" interface.

Can we format the Raspberry SD card remotely to erase its contents and install the "allsky-web" GUI interface?

EricClaeys, 2021-08-18:
In order to get the constellation overlay on the GUI, you would have to modify the code to add in the virtual sky javascript. If you are an experienced coder it shouldn't be too hard. But I'm not sure it's worth it since adding allsky-website is easy and takes very little disk space.

What do you mean by "remotely"?  If it's via VNC or something similar, once you format the disk you will lose your connection and can't get it back without physically being at the Pi.

FYI, the GUI is the "allsky-portal" application.  The "allsky-website" is the one with the constellation overlay.

EricClaeys, 2022-02-10:
Closing - never heard back from user.

closed 2022-02-10 by EricClaeys