AllskyTeam/allsky #440: Failed to upload the timelapse file to the web

bribon2, 2021-08-18:
Hello.
A couple of days ago I updated to version 0.8 but now for some reason the timelapse file is not uploaded to the web. Everything else works correctly. The keogram, startrail and images are uploaded without problem.
In the .log I see the following message:
Aug 18 06:34:06 allsky allsky.sh [909]: timelapse.sh: # 033 [0; 32mDeleting sequence # 033 [0m
Aug 18 06:34:06 allsky allsky.sh [909]: timelapse.sh: # 033 [0; 32mTimelapse was created # 033 [0m
Aug 18 06:34:07 allsky allsky.sh [909]: cd: Access failed: 550 No such directory. (/ allsky / videos)
Aug 18 06:34:34 allsky allsky.sh [909]: STARTING EXPOSURE at: 2021-08-18 06:34:34
Aug 18 06:34:37 ​​allsky allsky.sh [909]: mv: Access failed: 550 No such file. (tl-8444)
Aug 18 06:34:38 ​​allsky allsky.sh [909]: \ 033 [0; 31m *** timelapse.sh: ERROR: lftp failed with RET = 1 on '// home / pi / allsky / images / 20210817 / allsky-20210817.mp4 '\ 033 [0m

It seems that this last parameter is the one that fails, right?
I can't find a possible solution ...
Also say that via GUI the configuration settings of the ZWO are not applied, I have to go to the .json file to apply changes.
Thanks and best regards.

EricClaeys, 2021-08-18:
Please post your ftp_settings.sh file (don't copy/paste the contents, but attach the file).  I suspect your settings are wrong - they look like there are spaces in them, and "// home" should only have one "/".

bribon2, 2021-08-19:
Problem solved. As they said, it was a bad configuration of the ftp file ... But it was strange since all the files were uploaded except the timelapse video.
Yesterday after finding the bug, today everything has been uploaded correctly.
Thanks!

closed 2021-08-19 by bribon2