AllskyTeam/allsky #441: Missing video thumbnail on hosted allsky website

wixtron89, 2021-08-18:
How to generate thumbnail for video on hosted allsky-website?
https://www.wijaya.com/allsky/videos/

It is hosted on Dreamhost. 

[image]

thomasjacquin, 2021-08-18:
It seems that you're missing this file. https://github.com/thomasjacquin/allsky-website/blob/master/NoThumbnail.png
Make sure it has been uploaded properly to your website.

wixtron89, 2021-08-18:
Noted, The image file was corrupted, and now it has been replaced,
But still no thumbnails from the videos.

[image]

EricClaeys, 2021-08-18:
Thomas,
do you know how to create thumbnails from video files?  Will ffmpeg do it?
Eric

thomasjacquin, 2021-08-19:
Yes, @EricClaeys. See this thread https://stackoverflow.com/questions/27145238/create-thumbnail-from-video-using-ffmpeg
I meant to do it a while ago but lacked the time. I believe extracting a frame in the middle of the video should be fine as it would avoid dusk and dawn over-exposure.

SamCrask, 2021-09-15:
Hi Thomas & Eric -
I also ran into this 'No Thumbnail' problem after moving your website code to my NAS.  Turns out to be a known issue with some NAS's (and some site hosting) not supporting ffmpeg h264 encoding/decoding on the host - something about commercial usage rights for codecs from VLC.  
The " --disable-decoder=h264" shows up on the host's ffmpeg command line list, and "Decoder (codec h264) not found for input stream #0:0" is the result.  No luck searching for a way to enable it other than installing another version of ffmpeg on the host and hard coding to that.
Workaround:  I got it working by adding the ffmpeg video thumbnail command line from functions.php into timelapse.sh after the video generation, and having the PI do the generation and lftp upload to the thumbnail folder just after the lftp of the video.  Otherwise the website's PHP code fails to generate the video thumbnail.  I have more details on the fix I configured if that would be helpful. Seems to be working well and not much extra load on the Pi to create/upload a thumbnail each morning.

Sam

int48, 2021-09-17:
Hi Sam, i have the same problem aith allsky on Azure App Service, but i am not an expert with php.
Please can you explain at which point do you made the changes in functions.php and timelapse.sh.
Sascha

SamCrask, 2021-09-17:
@int48 (for ref @thomasjacquin  & @EricClaeys )

After a few PMs requesting more info, here's the revised section of timelapse.sh that creates and uploads a thumbnail from the PI, rather than relying on the PHP server-side code.  No changes are needed to the PHP code on the web server, as the web page will now find an existing thumbnail instead of trying to create one.

Rather than upload a revised timelapse.sh to GIT and get sucked into GIT file mgmt for this one change, I'm simply showing the changes I made below, and assuming anyone using this is familiar with reading & editing shell scripts.  Make a copy of any files being modified before making changes!  

Note that all added code is a copy and reuse of existing functionality from Thomas and others work, I just duplicated sections rather than spending time to rewrite it, so it works but isn't as efficient as it could be.  

Also, this uses a hard coded THUMBNAIL_SIZE_X value of 100 rather than the value from config.js, and creates the thumbnail jpg file in the $DIR folder on the PI, so it will get cleaned up by the existing auto-delete function.

### (existing section in timelapse.sh that generates the video file:)
```
ffmpeg -y -f image2 \
	-loglevel warning \
	-r $FPS \
	-i $DIR/sequence/%04d.$EXTENSION \
	-vcodec libx264 \
	-b:v ${TIMELAPSE_BITRATE:-2000k} \
	-pix_fmt yuv420p \
	-movflags +faststart \
	$SCALE \
	$DIR/allsky-$DAY.mp4
RET=$?
```

### (add this revised code below the section above:)
```
if [ $RET -ne 0 ]; then  # if any error generating video then quit, otherwise create a thumbnail
	echo -e "\n${RED}*** $ME: ERROR: ffmpeg failed with RET=$RET"
	echo "Links in '$DIR/sequence' left for debugging."
	echo -e "Remove them when the problem is fixed.${NC}\n"
	exit 1
else # Create thumbnail - per code from functions.php
	ffmpeg -ss 00:00:05 -i $DIR/allsky-$DAY.mp4 \
		-frames:v 1 \
		-filter:v scale='100:-1' \
		-frames:v 1 $DIR/allsky-$DAY.jpg
	RET=$?
fi
```

### (finally, add the lftp of the thumbnail to the server after copying the video file - add the second section to the existing code as shown:)

```
		TEMP_NAME="tl-$RANDOM"
		(
			FILE="allsky-$DAY.mp4"
			FULL_FILE="/$DIR/$FILE"
			lftp "$PROTOCOL://$USER:$PASSWORD@$HOST:$MP4DIR" -e "set net:max-retries 1; put "$FULL_FILE" -o $TEMP_NAME; rm -f "$FILE"; mv $TEMP_NAME "$FILE"; bye"
			RET=$?
			if [ $RET -ne 0 ]; then
				echo "${RED}*** $ME: ERROR: lftp failed with RET=$RET on '$FULL_FILE'${NC}"
		       fi
			FILE="allsky-$DAY.jpg"
			FULL_FILE="/$DIR/$FILE"
			lftp "$PROTOCOL://$USER:$PASSWORD@$HOST:$MP4DIR/thumbnails" -e "set net:max-retries 1; put "$FULL_FILE" -o $TEMP_NAME; rm -f "$FILE"; mv $TEMP_NAME "$FILE"; bye"
			RET=$?
			if [ $RET -ne 0 ]; then
				echo "${RED}*** $ME: ERROR: lftp failed with RET=$RET on '$FULL_FILE'${NC}"
		       fi
		) &
```

Feel free to reply on this thread if any questions or comments...
Sam

kujbida, 2021-09-22:
Hi,

Can you include the entire timelapse.sh file?

Thanks
Baltazar

EricClaeys, 2022-05-31:
Closing Issue - fixed in PR #1255 which will be in the next version of Allsky.

closed 2022-05-31 by EricClaeys