AllskyTeam/allsky #442: Readme.md incorrect refers to ‘settings.json’

MrACP1911, 2021-08-21:
File does not exist

EricClaeys, 2021-08-23:
Thomas, I asked MrACP1911 to submit this after reading his Facebook post about it.
The README.md file refers to "settings.json" but the actual name of the file is "settings_RPiHQ.json" or "settings_ZWO.json".  I will fix and commit to the dev branch.

MrACP1911, 2021-08-23:
My apologies did I screw it up?

EricClaeys, 2021-08-23:
No, you did exactly what I asked.  Thanks you.
I was just giving Thomas some background.

thomasjacquin, 2021-08-23:
Thanks Eric, I'll take a look.

MrACP1911, 2021-08-23:
Perfect, hey this little scrip took some time to put together and for free as well, is there a way through GitHub to send this guy some money. 

Thank you

EricClaeys, 2021-08-23:
You can send Thomas money using the link at the very bottom the page with README.md on it.

swatty007, 2021-08-24:
Good, I see this post now that explains my confusion at least a bit 😁

Looks like I hit a good timing to do my initial deployment then as I stumbled a bit across this one 🤣

The documentation referred to a file I couldn’t find. I did notice the settings_RPiHQ.json within the project's root folder and tried adjusting it without any effect. Then I found one under “/etc/raspap/settings_RPiHQ.json” which does affect my camera settings, e.g. time display, etc., and then there's also the GUI which displays a camera config.

This is the point I got lost. What is now the correct path/file to adjust things?
There's maybe some documentation that explains what I missed. What the correct/intended way of adjusting things is?

What I understand is there are camera-dependent configs for the ZWO/RPiHQ camera. 
And I assume the "*RPiHQ/ZWO.json.repo" files within the project are essentially the source/template file that gets published during deployment? But why within different locations :/

Would appreciate it if someone could light me up on this one :)

EricClaeys, 2021-08-24:
swatty07,
I was also very confused about this.  It is mentioned in the documentation but is still confusing.
When you install "allsky" the first time, those configuration files are in ~/allsky.  If you then install the admin GUI ("allsky-portal"), those files are copied to /etc/raspap and they are used instead.
I will ask Thomas why they are not removed from ~/allsky, or at least renamed.  I will also ask why they are moved at all, since config.sh isn't moved.
In summary,  since you have the GUI installed, use the file in /etc/raspap.

You are correct that the .repo files are the templates with default settings.  They are never updated.
Eric

EricClaeys, 2021-08-24:
... never updated unless we change the defaults.

thomasjacquin, 2021-08-24:
Hi Martin,

I'm glad you asked. By default, the location of the settings file is at the root of the project. (/home/pi/allsky).
However, if you install and use the GUI, the `Camera Settings` page is located at `/var/www/html/includes/camera_settings.php`
I couldn't figure out how to give the web server (lighttpd) write access to `/home/pi/allsky` from `/var/www/html`. I'm sure this is some kind of permission or lighttpd configuration. 
To solve this, I decided to copy the settings file to `/etc/raspap/` where I have write access.
Ideally, there should be only 1 location for the settings file, I agree.

In summary, 

- if you're not using the GUI, modify settings in `/home/pi/allsky`.
- if you're using the GUI, modify settings in `/etc/raspap`. But the GUI should take care of this so you don't have to manually edit it.

Thomas

EricClaeys, 2021-08-24:
Thomas,
To avoid confusion, simplify things, and make them slightly more uniform, could the installation of "allsky" put the files, plus the config.sh file, in /etc/raspap?  There would then only be one location.  Moving config.sh would require updating lots of scripts, but the change should be trivial.

I find the admin GUI to be very useful and easier to explain to people who ask about settings.  Have you ever considered merging it with "allsky"?
Eric

swatty007, 2021-08-24:
Ah yeah, I see makes sense. Thanks for the quick reply both 😁

Just now I found the note about this within the config.sh properties table (last line of the config.sh table). Maybe it's an idea to separate this note from the table and put it into a pull quote below to avoid people overlook this? 
I probably scrolled over this a “few” times 😖 If you like I can commit a pull request for it.

Yeah, haven’t touched lighttpd either for quite a while, more used to its bigger brothers. But I would imagine you could get it done via a symlink. Maybe:
`sudo ln -s /home/pi/allsky/config.sh /var/www/html/config.sh`
during the installation would do the job (haven’t it tested yet though).

This brings me to another thing I just noticed, but I will make a separate ticket for it.

swatty007, 2021-08-24:
> I find the admin GUI to be very useful and easier to explain to people

Actually, it could already help to add the path within the GUI as well. e.g. just display the absolute path below the editor & camera config.

Should be a trivial adjustment as well and which makes it more obvious for people what files to look for and more dynamic than a documentation you have to keep in sync.

EricClaeys, 2021-08-25:
Martin,
One of the advantages of the GUI is that people don't need to know where the file is, or even that there IS a file.  It hides that complexity.

closed 2021-08-25 by thomasjacquin

EricClaeys, 2021-09-05:
Martin,
Now that I have both a ZWO and RPiHQ camera I wanted a way to easily distinguish them in the GUI, so I added the pathname to the camera options file.  I'm still not sure I like the idea of putting the file name there, rather than just "ZWO Camera" or "RPiHQ Camera" (maybe with the path as a rollover).

commit f53abbcc (2021-08-23): Update README.md

Fixes #442 
and adds some additional clarifications.
commit f08264ad (2021-08-25): Multiple bug fixes and improvements

* Disable alwaysshowadvanced for default

It was accidentally left as "1" from my Pi.

* Create videos/thumbnails

This changes goes with the changes to "allsky-website" that automatically creates video thumbnails.
Once people grab the "allsky-website" changes they'll need to create the vidoes/thumnails directory.  Should we move the "Create" and "Fixing" code block to a different shell script that users can invoke, and have install.sh also invoke it?  That should make it easier for non-technical users to create the directory.  The "mkdir" will need to chan [... cut]