AllskyTeam/allsky #443: Transition from day to night: manual night exposure time does not apply but last day exposure time

vnawrath12159, 2021-08-22:
Dear developers,

I recently updated (installed) Version 0.8 of this great software package. My setting for capture is as follows:

- Day Capture: auto-exposure = yes
- Night Capture: auto-exposure = no / manual setting = 6 seconds
- Night Capture: auto-gain = yes

I noticed, that when changing from day to night capture, the software doesn't take the nightly auto-exposure manuell setting (6 seconds). Instead it repeats to apply the last exposure time from the day capture. Auto-gain for night is used correctly.

In my opinion, there is a typo in capture.ccp in line 1866:

Instead of
`            if (numExposures == 0 || **asiDayAutoExposure** == ASI_FALSE)`
it should be
`            if (numExposures == 0 || **asiNightAutoExposure** == ASI_FALSE)`

This evening, I will check, whether my manual code change leads to the expected correct change in the exposure time.

Code section for switching to nighttime with the according IF instruction at the end:
`        {
            sprintf(textBuffer, "==========\n=== Starting nighttime capture ===\n==========\n");
            displayDebugText(textBuffer, 0);

            // Setup the night time capture parameters
            if (asiNightAutoExposure == 1)
            {
                currentAutoExposure = ASI_TRUE;
                setControl(CamNum, ASI_AUTO_MAX_EXP, asiNightMaxExposure, ASI_FALSE);
                printf("Saving auto exposed night images with delay of %'d ms (%d sec)\n\n", nightDelay, nightDelay / MS_IN_SEC);
            }
            else
            {
                currentAutoExposure = ASI_FALSE;
                printf("Saving %ds manual exposure night images with delay of %'d ms (%d sec)\n\n", (int)round(currentExposure / US_IN_SEC), nightDelay, nightDelay / MS_IN_SEC);
            }

            currentBrightness = asiNightBrightness;
            currentDelay = nightDelay;
            currentBin = nightBin;
            if (numExposures == 0 || asiDayAutoExposure == ASI_FALSE)
            {
             	currentExposure = asiNightExposure;
            }
`

EricClaeys, 2021-08-24:
I believe you are correct.  I probably did a copy/paste from the night-to-day code but didn't change it correctly.
Were you able to change it and see if it worked better?
Eric

vnawrath12159, 2021-08-24:
Yes, after two evenings, the code change worked correctly. The manual night exposure value is now taken when switching from day to night.
Volker

commit 87a5bea4 (2021-08-25): Fix to day-to-night transition

Fixes #443
commit f08264ad (2021-08-25): Multiple bug fixes and improvements

* Disable alwaysshowadvanced for default

It was accidentally left as "1" from my Pi.

* Create videos/thumbnails

This changes goes with the changes to "allsky-website" that automatically creates video thumbnails.
Once people grab the "allsky-website" changes they'll need to create the vidoes/thumnails directory.  Should we move the "Create" and "Fixing" code block to a different shell script that users can invoke, and have install.sh also invoke it?  That should make it easier for non-technical users to create the directory.  The "mkdir" will need to chan [... cut]
closed 2021-08-25 by thomasjacquin