AllskyTeam/allsky #445: Installation Path & Portal Images

swatty007, 2021-08-24:
Not quite sure if this belongs in this repo, or the portals repo. 

However, I will keep it here for now as I found the references to solve it within this config. Also, it's no real issue since it's already resolved. Rather a question/information post if someone else comes across the same.

I have installed everything out of habit under **/opt/allsky**. The installation had no problems, and the services all work flawlessly. However, the entire image tab within the portal was blank and when I clicked any of the icons e.g., for timelapse I could just see a warning telling me **“There are no image directories”**.
After a closer look at the **config.sh** I noticed you can define either **“current”** or **“allsky”** as **IMG_DIR** variable which relates to some fixed paths e.g. **“/home/pi/allsky”**

So a simple symlink will fix the emptiness:
`sudo ln -s /opt/allsky /home/pi/allsky`

However, this brings up the question, are all these paths hardcoded, so you are supposed to install the system within those directories? Didn’t found that statement anywhere within the documentation so I assumed the installation path shouldn’t matter. After reading this I also thought this wasn’t the case anymore https://github.com/thomasjacquin/allsky/pull/214

Maybe some leftover string in the portal?
If it's not a requirement anymore to place the project within any of those files I would recommend updating the comment block within the **config.sh** and then I should probably create a ticket for this in the portal repo if desired :)

EricClaeys, 2021-09-01:
swatty007,
You bring up a couple good points.
1. Installing the AllSky software somewhere other than /home/pi/allsky.  If I understand correctly, your symbolic link fixed things and everything is now working?  Did you try the GUI (allsky-portal) and website (allsky-website)?  If so, I will add this to the "troubleshooting and tips" WiKi I'm creating.
2. The scripts themselves have lots of hard coded paths to files and directories within the "allsky" directory, like "images", "keograms", "scripts", etc.  Not sure how likely someone will want to install, for example, the "scripts" directory somewhere outside the "allsky" directory, so this probably isn't work changing.

If you consider this issue resolved, please close it.

EricClaeys, 2021-09-15:
No response after 2 weeks so closing.  Plus, we are looking at ways to make it easier to install Allsky in places other than /home/pi/allsky.

closed 2021-09-15 by EricClaeys

swatty007, 2021-09-15:
Sorry that I didn't reply yet, time flies. It's fine to be closed from my side 👌

But to follow up on your questions:
1. yep the symbolic links fixed it and everything is running fine since then. But I only tried the portal, not the website.
2. relative paths should indeed not matter at all. But I guess there are some absolute ones included as well. Otherwise, the symlink would have not been necessary. But like mentioned simple fix once you are aware :)

I am pleased to hear that you are working on further improvements. Maybe this would be something for the Hacktober as well if you look for support on that: https://hacktoberfest.digitalocean.com/

EricClaeys, 2021-09-16:
@swatty007, thanks for the link to Hacktoberfest.  I never heard of it before, but it sounds interesting.  I'll leave it up to @thomasjacquin to decide if he wants to "opt-in" Allsky.