AllskyTeam/allsky #454: v0.8 day to night, auto to manual exposure time not transitioning

jcauthen78, 2021-09-03:
PI 4 8gb
ASI178MC
Allsky v0.8 (installed 8/9/2021)
- no updates since install

I swapped back to the manual night exposures to avoid the fluctuation of brightness the smoke/weather was causing with the script. But I'm noticing that the transition from the Daytime auto exposure to Nighttime Manual exposure isn't making the full transition and getting stuck in the sub-10second exposure length. Attached screenshot of the 'details', and snip of the allsky.log. I've rebooted (soft and power pull), but the transition just isn't fully happening for some reason.

I'm not seeing any errors in the logs that would be causing what it's doing. If I do a 'Save Settings' (without changing any settings) from the allsky admin portal, the programmed/desired 30sec image starts capturing. 

[image]

[image]

[allsky_log.txt](https://github.com/thomasjacquin/allsky/files/7103145/allsky_log.txt)
[settings_ZWO_json.txt](https://github.com/thomasjacquin/allsky/files/7103195/settings_ZWO_json.txt)

EricClaeys, 2021-09-03:
Hi Jim,
Can you include the part of the log file where it transitions from day to night?  Include a few exposures before and after the transition.
Thanks - Eric

EricClaeys, 2021-09-03:
@jcauthen78, if you could also include the lines that are in the log file when the software starts, that show all the settings and other items prior to the first exposure, that would help.
I looked at the code and can't find anything obvious.  There's a line that sets the night exposure to what you requested if nighttime auto exposure is off, so that should cause it to use 30 seconds in your case.

jcauthen78, 2021-09-04:
Here are the items you asked for. 
Looking at the images at transition, the exposure time seems to stick at what the last daytime auto exposure was. 

I'm not against reinstalling from Github, just thought it was a bit odd and was wondering if it was noticed by anyone.  If you see anything glaringly wrong, let me know and I'll see what I can do to dig & tweak. 

I'll also see if it's repeatable on the CampingPI tomorrow evening (its currently on auto day, auto night). 

[allsky_log_restart.txt](https://github.com/thomasjacquin/allsky/files/7109233/allsky_log_restart.txt)
[allsky_log_transition.txt](https://github.com/thomasjacquin/allsky/files/7109234/allsky_log_transition.txt)
[transition.zip](https://github.com/thomasjacquin/allsky/files/7109235/transition.zip)

EricClaeys, 2021-09-05:
@jcauthen78, would you please try this capture.cpp?  It will output a log entry saying if it's using the night exposure or the day exposure when it transitions from day to night.  It likely won't fix anything but will give me a better idea where the problem is.

You are right that the exposure at night sticks with the one from daytime.  I'm just not sure why yet.
Thanks Jim.
[capture.zip](https://github.com/thomasjacquin/allsky/files/7110845/capture.zip)

jcauthen78, 2021-09-05:
is it an entry in allsky.log? or a separate? 
I wasn't able to get it incorporated before the turn this evening, so I'll check the logs for you during tomorrow's transition.

EricClaeys, 2021-09-05:
The log entry will either be:
  `==== Using night exposure of X`
or
  `==== Using day exposure: numExposures=X, asiNightAutoExposure=[0 or 1]`

We want the first one (where X would be 30,000 for you), but if we get that and it's still not working, I'll be really perplexed.
Eric

jcauthen78, 2021-09-08:
[allskyLog-7sep-transition.txt](https://github.com/thomasjacquin/allsky/files/7125683/allskyLog-7sep-transition.txt)

Well... I'm not sure what was changed between my capture and the test capture you sent is, but the transition is happening smoothly now. Here's the log during the transition. 

If you need me to test anything else, let me know. if not, let me know and I'll close this out.

EricClaeys, 2021-09-08:
@jcauthen78, you don't  need to test anything else.  I can test on my end.  I made a small change in the capture.cpp I sent you that I didn't think would make a difference, but did.   Go ahead and close the issue.  Glad it's working.

jcauthen78, 2021-09-08:
Eric Rocks!

closed 2021-09-08 by jcauthen78