AllskyTeam/allsky #456: V0.7 ASI exposure will not increase beyond 60 seconds

ShianMull, 2021-09-04:
Hi Thomas, Eric,

Is there any known issue with V0.7 not allowing auto exposure beyond 60 seconds?  I currently have set 120 seconds as I’m interesting in seeing if I can get Milky Way images, but the auto exposure is not moving past 60 seconds.  I’ve checked all setting files manually and they’re all set to 120000 milliseconds.

Many thanks

Mark

EricClaeys, 2021-09-04:
Mark,  it's a limit with the camera.  They will only do 60 second autoexposure, but will do up to 2000 seconds manual exposure.  On the cameras I've seen, here are the limits (1 millisecond (MS) = 1 second):
```
- AutoExpMaxExpMS:
   - MinValue = 1
   - MaxValue = 60,000
   - DefaultValue = 100
```
I also wish it went higher than 60 seconds, but there's nothing we can do about it in software.  Your options are to use manual exposure, and/or contact ZWO and "encourage" them to increase the value.

FYI, in version 0.8 if you set the Debug Level to 3, you'll get a list of the cameras capabilities along with their min, max, and default values, as above.

If this answered your question, would you please close the issue?
Eric

EricClaeys, 2021-09-04:
@ShianMull, "AutoExpMaxExpMS" is the internal name ZWO uses for "Auto Exposure".  Each of the capabilities has an internal name; most are easier to understand, like "Gain".

ShianMull, 2021-09-05:
Thanks Eric - that explains why I can set five minutes on the camera in MaximDL and only get 60 secs in auto.

closed 2021-09-05 by ShianMull