AllskyTeam/allsky #457: Question about the LFTP call

maserowik, 2021-09-05:
So I had to redo my sd card in my all sky raspberry pi.  I redid the config.sh and the FTP-Settings file to match what I had before.  I was getting a live image but was unable to upload any images to my website.  Looking at the allsky.log I noticed a Certificate  Validation 

Sep  4 19:33:12 Pi-AllSky allsky.sh[624]: cd: Fatal error: Certificate verification: certificate common name doesn't match requested host name ‘n3bsq.com’ (**:**:**:**:**:**:**:**:**:**:**:**:**:*:**:**:**:**:**:**)
Sep  4 19:33:13 Pi-AllSky allsky.sh[624]: put: image.jpg: Fatal error: Certificate verification: certificate common name doesn't match requested host name ‘n3bsq.com’ (**:**:**:**:**:**:**:**:**:**:**:**:**:*:**:**:**:**:**:**)
Sep  4 19:33:14 Pi-AllSky allsky.sh[624]: mv: Fatal error: Certificate verification: certificate common name doesn't match requested host name ‘n3bsq.com’ (**:**:**:**:**:**:**:**:**:**:**:**:**:*:**:**:**:**:**:**)

I then remembered on the old version I had to add the following command to the lftp.conf 

set ssl:verify-certificate no

as soon as I added that the error in the log file went away

so my question is what is the best way to do this via the lftp.conf or somewhere else.

Mike

EricClaeys, 2021-09-05:
Mike,  I don't remember the error I was getting, but here are my notes:
```
To allow lftp to work with unknown certificates, add:
		set ssl:check-hostname false
	to /etc/lftp.conf
```
I believe it's also possible to add that to a file in the $HOME directory.  Either way, I'm trying to get away from telling people to edit the scripts - that's too difficult for most people.
Another possibility that might be easier for people would be for Thomas or I to put a variable in config.sh that the lftp commands would include.  People like you and I would set that variable to "set ssl:verify-certificate no".  This might also make it easier to move to a new device as long as someone kept their config.sh file.

Your command and mine are different; I'm not sure if that means we had different error messages or there are two ways to resolve the same issue.

Anyway, I'll add this to the "troubleshooting and tips" WiKi I'm creating.
If this answers your question, please close this issue.   Thanks.

maserowik, 2021-09-05:
Eric,   
   I added you command also to the lftp.conf and commented my command.  It appears both commands work.  Not sure which one is the best one to use for this issues.  I will close it out for now unless anyone has any suggestion as to the best one to use.

closed 2021-09-05 by maserowik

EricClaeys, 2021-09-05:
Thanks Mike.  Interesting that both our commands work.  I'll have to undo my fix and see what error message I get.  Either way, I'll add both to the WiKi.