AllskyTeam/allsky #459: No issue just a question about a lens to use?

robmorin, 2021-09-06:
Hello all, not sure where to post this question, if not here,  please let me know where it should be posted.

I ordered my Raspberry PI HQ Camera and will use with Raspberry 3 B +, or might buy a pi 4
But I am confused about what lens to buy?
If fish eye,  180 degree, 160 degree, 150 degree?
Any recommendations?
Any help help or links appreciated.

Thanks & have a great day!

Rob, Montreal Canada

bbillp, 2021-09-06:
I believe our ZWO cameras that are furnished with a fish eye lens are 170 degrees which fits my needs.   You can check other sources such as cloudynights.com or raspberrypi.org for additional info.  I think adafruit.com has a couple of lenses available too.   BP

ckuethe, 2021-09-11:
Use whatever lens gives you a good signal to noise ratio, eg. as much interesting sky as possible with as little boring scenery as  possible, given your location.

Does your camera have an unobstructed view from horizon to horizon? Maybe a 180 degree lens is right for you. Are you in an urban area with buildings and lights below 20 degrees? Maybe a 150 degree lens, with an additional light baffle is right for you. I think my ZWO came with a 155 degree lens and after a couple of weeks I switched to a 180 degree lens but then had to add some extra shielding on one side to block some street lights.

EricClaeys, 2021-09-15:
@bbillp, if @ckuethe's response was sufficient, please close this issue.  Thanks

robmorin, 2021-09-20:
@ckuethe 

I am to put this camera at the cottage, my field of view might not be 180 degrees, but i need a lens to test at least, that's why i was asking about the lens you were using. Its difficult to find a lens other than the raspberry lenses that come with the camera for a 1/2.3 sensor  So if i just get the 6mm lens(60 degree wide angle) from raspberry is that good enough for some testing then i can move on from there?
I do not know much about photography so i figure i can learn a bit by doing this project, but i need a jump off point! :)
Thanks..

ckuethe, 2021-09-20:
A completely free place to start: google for bintel calculator. That'll let you plug in some parameters like focal length, focal ratio, pixel size, sensor dimension, and you'll get a summary of your FoV.

A second completely free tool: stellarium. Download and install it, and spend some time with the ocular/sensor tools. You can simulate an arbitrary location, at an arbitrary time, with an arbitrary field of view... to estimate the amount of sky you'd see. You can also run time in high speed mode, effectively allowing you to simulate the end-of-night timelapse too.

My advice: get a 160 degree lens. A 60 degree lens will absolutely put photons on your sensor, and you'll be very frustrated at how little sky you'll see... Try it with stellarium. :)

robmorin, 2021-09-20:
Thanks for all the info ckuethe!

I will for sure check out that calculator and download the software.  

From what I understand so far is I need a lens that is compatible with the 1/2.3 sensor,  which is hard to find in a wide angle of 160 or so. Thanks again!!

Have a great day

ckuethe, 2021-09-20:
No, what you need is a lens that is compatible with the pi camera CS mount. Don't worry too much about how much image will be projected on your sensor just yet.

I think of CS as "C short" - it has 5mm less focus depth, so you can put a C-lens+CS adapter (aka a 5mm thick spacer) on a CS-mount camera, but you can't use a CS-lens on a C-mount camera. The Pi HQcam has a CS mount so it can use a CS lens directly, or a C lens with a CS adapter.

So once you find a reasonably priced lens from your preferred supplier, plug in the specs into bintel or stellarium and get a good approximation of what your view will be. Repeat until you find something you're happy with

ckuethe, 2021-10-05:
Closing this issue.

closed 2021-10-05 by ckuethe