AllskyTeam/allsky #460: ASI_ERROR_TIMEOUT always

KrisAstro, 2021-09-07:
Hi,

i tested the last release and I always an ASI_ERROR_TIMEOUT. I use an old asi 120MM that's work fine.

I tested directly with that :
.`/capture -autousb 0 -usb 40 -latitude 43.8N -longitude 5.93E -newexposure 0 -debuglevel 4 -dayautoexposure 1 -daytime 1`

  > Exposure set to 30000 Âµs (30.00 ms), timeout: 560 ms
  > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
  > Exposure set to 30000 Âµs (30.00 ms), timeout: 560 ms
  > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
  > Exposure set to 30000 Âµs (30.00 ms), timeout: 560 ms
  > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
I played with the usb value and newexposure flag but i always there messages.

Thank.
Christian

mpv1953, 2021-09-08:
I'm having the same issue.
This is a brand new install of RPiOS, headless with no GUI installed.
I'm using an ASI120MC, and have tried using a Pi3 and a Pi4 with no joy either time.  I loaded the ZWO SDK and compiled their demos.  The demos work, so I know the camera is working and reachable by the Pi.
Thanks,
Mike

EricClaeys, 2021-09-09:
@KrisAstro, are you using the latest capture.cpp, updated 5 days ago?
If so, do you have a powered USB hub you can try?
Did the prior version 0.7 version work ok?

KrisAstro, 2021-09-09:
Hi Eric,

yes, is the last release of capture.cpp (I upgraded 2 days ago) and she's plugged directly to the pi and yes I used the prior version all summer. Like Mike, I tested the camera with an other program (AstroDmxCapture for me) and she is work fine.

I tried with and without autousb, usb speed from 40 to 80, with and without newexposure without success. I directly used the capture program like that : 

./capture -autousb 0 - usb 40  -latitude 43.8N -longitude 5.93E -newexposure 0 -debuglevel 1 -dayautoexposure 1 -daytime 1

calberts, 2021-09-10:
i have the same, here

Saving images with delay of 5000 ms (5 sec)
Sep 10 17:47:17 allsky allsky.sh[2466]: WARNING: Value of 200 greater than max value allowed (100) for control 'AutoExpMaxGain' (#10).
Sep 10 17:47:17 allsky allsky.sh[2466]: WARNING: Value of 50 greater than max value allowed (20) for control 'Offset' (#5).
Sep 10 17:47:18 allsky allsky.sh[2466]:   > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
Sep 10 17:47:20 allsky allsky.sh[2466]:   > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
Sep 10 17:47:23 allsky allsky.sh[2466]:   > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
 
i have install it to day from the GIT, the previous version worked.  Besides this should be solved is there a way to get version 0.7 or before that from some where

KrisAstro, 2021-09-10:
Hi,

I think i have found my problem.... the day exposure value is by default at .5 (0,0005 second) . I think this value is too small (especially for my asi 120). I tried to set to 500 (1/2 second) and it work now. 

I hope this tips must be help.

calberts, 2021-09-10:
No success, At this site

still the following errors
WARNING: Value of 200 greater than max value allowed (100) for control 'AutoExpMaxGain' (#10).
Sep 10 20:34:05 allsky allsky.sh[1146]: WARNING: Value of 50 greater than max value allowed (20) for control 'Offset' (#5).
Sep 10 20:34:05 allsky allsky.sh[1146]: ===Taking 3 images to clear buffer...
Sep 10 20:34:05 allsky allsky.sh[1146]:   > Exposure set to 30000 Âµs (30.00 ms), timeout: 560 ms
Sep 10 20:34:06 allsky allsky.sh[1146]:   > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
Sep 10 20:34:08 allsky allsky.sh[1146]:   > Exposure set to 30000 Âµs (30.00 ms), timeout: 560 ms
Sep 10 20:34:09 allsky allsky.sh[1146]:   > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
Sep 10 20:34:11 allsky allsky.sh[1146]:   > Exposure set to 30000 Âµs (30.00 ms), timeout: 560 ms
Sep 10 20:34:11 allsky allsky.sh[1146]:   > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
Sep 10 20:34:16 allsky allsky.sh[1146]:      ***** Stopping AllSky *****
Sep 10 20:34:16 allsky allsky.sh[1146]: capture exited with retcode=2

I played with the USB Setings no success

Chris

EricClaeys, 2021-09-12:
@KrisAstro, @mpv1953,
Please try the attached file.  It's the version 0.7 capture.cpp that knows about the new option names (e.g., "daybrightness" instead of "brightness").
Unpack the file, rename the .cpp file to "capture.cpp", put in ~/allsky, then "cd allsky ; make capture" and restart AllSky.  It won't give any errors in the log file but you'll know it's working if you get images.

Let me know how it goes.
Eric
[capture0.7with0.8arguments.zip](https://github.com/thomasjacquin/allsky/files/7148959/capture0.7with0.8arguments.zip)

EricClaeys, 2021-09-12:
@calberts, please the file above and report back if it worked or not.
Eric

dbainbridge, 2021-09-13:
Just pulled latest code from repo few minutes ago and now mine is also getting this same error :| Will try to investigate more.

dbainbridge, 2021-09-13:
I turned debugLevel to 5 and saw this:
`===Taking 3 images to clear buffer...
  > Exposure set to 30000 Âµs (30.00 ms), timeout: 560 ms
  > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
  > Exposure set to 30000 Âµs (30.00 ms), timeout: 560 ms
  > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
  > Exposure set to 30000 Âµs (30.00 ms), timeout: 560 ms
  > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
     ***** Stopping AllSky *****
`
I then updated SHORT_EXPOSURE in capture.cpp and added another 0 so it was 10x longer and that got rid of that error.
I have ASI 178MC.

Now seeing this:
`STARTING EXPOSURE at: 2021-09-13 00:07:07
  > Exposure set to 5000000 Âµs (5000.00 ms), timeout: 10500 ms
  > Got image @ exposure: 7475000 Âµs (7475.00 ms)
  > Saving image 'image.jpg' that started at 2021-09-13 00:07:07  (13 Âµs)
  > Sleeping: 12525 ms
STARTING EXPOSURE at: 2021-09-13 00:07:26
  > Exposure set to 7475000 Âµs (7475.00 ms), timeout: 15450 ms
  > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
STARTING EXPOSURE at: 2021-09-13 00:07:45
  > Exposure set to 7475000 Âµs (7475.00 ms), timeout: 15450 ms
  > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
     ***** Stopping AllSky *****
`
So, checking what may be wrong with that..

dbainbridge, 2021-09-13:
This was also fixed by increasing the timeout like @ckuethe did in his PR: https://github.com/thomasjacquin/allsky/pull/475/files
I had mine set to 5000, but probably 1500 would work.  The current time being used is way too short.  Looks like this was changed last month and why it started breaking at least for cameras with large number of pixels

dbainbridge, 2021-09-13:
./capture itself is now working without errors but allsky.sh still gets same errors

==========
=== Starting nighttime capture ===
==========
Saving auto exposed night images with delay of 2000 ms (2 sec)

  > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
  > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
  > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
^C     ***** Stopping AllSky *****

ckuethe, 2021-09-13:
I was only getting timeouts with heavy usb contention, regular imaging worked just fine. 1500 was a fairly arbitrary choice that happened to be a relatively small change that would probably address my likely issue: lots of usb I/O doing end of night processing. 

It might be useful to investigate the relationship between image size, bit depth, bandwidth, and other usb traffic... Under real world conditions.

ckuethe, 2021-09-13:
Do you get timeouts with both the new and old exposure algorithm?

dbainbridge, 2021-09-13:
Setting the newexposure to 1 in settings does get rid of the timeout issue when running ./allsky.sh.  No errors displayed but my site not updating..  checking what that is..

dbainbridge, 2021-09-13:
Crap.. the install overwrote my original ftp settings file :(

calberts, 2021-09-13:
For all this time out errors, when you see them, please check https://github.com/thomasjacquin/allsky/issues/417 the solution provide by Eric worked for me. I have now lucky working version again

Clear night

Chris

ckuethe, 2021-09-13:
USB2. This was intentionally pushing the bus.

EricClaeys, 2021-09-14:
@ckuethe, can you do me a favor and look at the "use_new_exposure_algorithm" code in the latest capture.cpp and see if you can tell why it doesn't seem to be working for everyone when they set "newexposure" to 0?  I have a newer capture.cpp with that code and other changes that a few people have successfully used, so I can't figure out why the current capture.cpp doesn't seem to work.

ckuethe, 2021-09-14:
Sure. While we're digging into things, can you put up a PR or a fork or something with the same changes as in your zip file? It's easier to see what changes that way...

ckuethe, 2021-09-14:
OK, I figured it out.

There's exactly one place now that `ASIStartVideoCapture()` is called, and that's inside `takeOneExposure()`, and only when `use_new_exposure_algorithm` is true. ([capture.cpp#L462](https://github.com/thomasjacquin/allsky/blob/27c09347bac50f65ed17d9c7c1eb8bb6e60bb8db/capture.cpp#L462)) 

Anyone using the old mode would be trying to call `ASIGetVideoData()` on a camera that isn't in video streaming mode, and of course it wouldn't have any frames ready in time.

This would also explain flooding my dmesg with bus resets - `ASIStopVideoCapture()` is being called at the end of the exposure ([capture.cpp#L490](https://github.com/thomasjacquin/allsky/blob/27c09347bac50f65ed17d9c7c1eb8bb6e60bb8db/capture.cpp#L490))

I'm not sure of the benefit of starting/stopping video mode for every frame. It would be nice if ZWO could explain why that's supposed to be better, and it would be great if they could fix it.

EricClaeys, 2021-09-14:
Thanks Chris.  I just submitted a pull request to fix it.  [](https://github.com/thomasjacquin/allsky/pull/480).
I had it in my newer capture.cpp, but it was left out of the fix that was applied a couple weeks ago.

The reason we implemented the start/stop video mode for every picture in version 0.8 was that it decreased the sensor temp by 10-15 degrees C since the camera is idle during delays.

EricClaeys, 2021-09-14:
@KrisAstro,
A good default daytime exposure is hard to come up with since different cameras have different sensitivities, and people's Angle is different, which means some people start daytime when it's a little dark and some start when it's fully bright, which all affect the required exposure.
The daylight auto-exposure algorithm in version 0.8 handles that within an image or two. so it's not much of an issue.

KrisAstro, 2021-09-14:
Thank Eric for you time and you investigations. Since, I've completely reinstall my system and it work fine now. I'll do the update soon.
I take the opportunity for a question : why do you used ASIStartVideoCapture instead ASIStartExposure ? When i investigate into the code, i seen that and I ask myself why.

Christian

ckuethe, 2021-09-14:
IIUC, single captures don't do autoexposure, while video mode allows the camera to autoadjust. I'm not a fan of the nighttime autoexposure behavior of the camera and would love to just switch to snapshot capture, with exposure settings computed by allsky, rather than the camera.

EricClaeys, 2021-09-15:
@ckuethe is correct, ZWO's "snap" mode, which uses ASIStartExposure() doesn't support auto exposure or (I'm pretty sure) auto gain.  For daytime shots we use our own auto-exposure algorithm since there's a bug in ZWO's; we could probably implement it for nighttime as well but it would take some work.  Incorporating auto-gain would take a LOT of work, and I don't think it's worth it.

KrisAstro, 2021-09-15:
Ah, OK, is more clear now ! 

Thx for your response

PKarhu, 2021-09-16:
Hey!
 Could there possibly be somekind of folder assignation problem?! 0.7 had in config.sh -file something like this:
------
# Path to the camera settings (exposure, gain, delay, overlay, etc)
CAMERA_SETTINGS_DIR="$ALLSKY_HOME"
CAMERA_SETTINGS="$CAMERA_SETTINGS_DIR/settings_$CAMERA.json"

but, the newer 0.8 has only one line considering the path to the correct file name:
-----
if [[ $CAMERA -eq "auto" ]]; then
  # restore currently saved autodiscovered camera mode if any
  source "$ALLSKY_HOME/autocam.sh"
fi
**### _CAMERA_SETTINGS="$CAMERA_SETTINGS_DIR/settings_$CAMERA.json"_**

--
somehow seems to be mixed and confused together, when attepting to update to the newest version (0.8)
and therefore affecting how and where to obtain proper settings ...

i'm confused, because we have used this allsky-software with ASI-120MC S  for a year now, (of course noticed all the previous problems with this type of camera), but I've also tried to solve these problems ,,, and hope that this update-problem will be fixed somehow, so our project will continue to exist.

PKarhu, 2021-09-16:
Outs...somehow the fonts,boldening etc. seems to be out of my understanding (i'm sorry about that)  ... I wrote that with raspberry 😬

but, when adding the new update, all went to this "ASI_timeout"-hell...
I tried to restore that old backup of 0.7 folder ->> renamed it back to 'allsky' , instead of allsky_backup... then ran "sudo ./install.sh"

of course, this action might have also been the reason why everything is so mixed up. 
but could there possibly be this folder-assignation problem, which I tried to explain in the previous (messed up) message??

EricClaeys, 2021-09-16:
GIT web pages interpret certain characters as meaning bold, italics, etc.  Unfortunately, "#" is interpreted a a header which is why when you copy/paste from a *.sh file (with "#" as comments) those lines appear large in bold on the GIT web page.  I hate that...

We recently uploaded what we hope is a final fix for the timeout issue for users that turn off the version 0.8 exposure mode.  Could you try the newest capture.cpp?  If it works, I suggest doing a clean install of the newest code.

EricClaeys, 2021-09-16:
forgot to mention the issue with config.sh.  The last few lines should look like this if you have the GUI installed:
```
CAMERA_SETTINGS_DIR="/etc/raspap"

if [[ $CAMERA -eq "auto" ]]; then
  # restore currently saved autodiscovered camera mode if any
  source "$ALLSKY_HOME/autocam.sh"
fi
CAMERA_SETTINGS="$CAMERA_SETTINGS_DIR/settings_$CAMERA.json"
```

If that's not what you have, edit the file and let me know how it goes.  Thanks

ckuethe, 2021-09-17:
As documented on the main page, there is no easy, one-click upgrade... Especially since the addition of recent new features and controls.

I recommend you make a copy of all your old allsky configuration and reinstall using the latest code from git

PKarhu, 2021-09-17:
Hey!
 Don't know what went wrong at the first time, when I tried to follow the instructions, but I've got it working after all!
All the same procedures (including direct download from the github) like the first time I tried.

Am I ever so grateful to you all, for this amazing view from 'Karhukamera' ... I managed to update
the software, brought back the raspberry to its current location and this is what happened:
incredible Auroras filling almost the whole scene!

[image]

---

Perfect timing, for the perfect update!
 I must salute you all!
Sensor temperature fix,adding automatic white balance, different day&night settings, etc... ...
perhaps I'm still confused about the Auroras, but what an update you've just created!

Keep on working with this project <3 !!

EricClaeys, 2021-09-18:
@PKarhu, nice aurora shot.
You're welcome for all the new features in version 0.8, and I'm glad you like them.  We enjoyed developing them.
We have a lot more improvements on our "to do" list.  One of my goals is to make Allsky easier to use, update, and troubleshoot for non-technical users so they don't have as many issues.

EricClaeys, 2021-10-14:
Closing this Issue - the ASI_ERROR_TIMEOUT problem has been resolved as far as I can tell.