AllskyTeam/allsky #461: Loop of All Sky Starting then failing

DamoclesSword, 2021-09-08:
It seems that me, and Patriot Astro, have ran into a new issue where everything works, but in the GUI it will say "starting up" then give a "not running" error and do that cycle a few times.

EricClaeys, 2021-09-08:
@DamoclesSword, please check in the log file /var/log/allsky.log.  AllSky is starting, getting an error of some type, exiting, then restarting, getting the error, etc.
If you see ASI_ERROR_TIMEOUT messages in the log file, please make sure you have the latest capture.cpp and camera_options_ZWO.json files, and in the GUI, go to the bottom, click on "Show advanced options", and make sure "Version 0.8 Exposure" is UNchecked.

ckuethe, 2021-09-12:
Try applying the patch from #469. It won't fix your problem, but maybe it'll tell you what's failing.

DamoclesSword, 2021-09-12:
> @DamoclesSword, please check in the log file /var/log/allsky.log. AllSky is starting, getting an error of some type, exiting, then restarting, getting the error, etc.
> If you see ASI_ERROR_TIMEOUT [...]

I finally got time to check. Here is the log: 
Sep  7 20:08:26 allsky allsky.sh[2377]:      ***** Starting AllSky *****
Sep  7 20:08:26 allsky allsky.sh[2377]: Making sure allsky.sh is not already running...
Sep  7 20:08:26 allsky allsky.sh[2377]: Trying to automatically choose between ZWO and RPI camera
Sep  7 20:08:26 allsky allsky.sh[2377]: CAMERA: ZWO
Sep  7 20:08:26 allsky allsky.sh[2377]: CAMERA_SETTINGS: /home/pi/allsky/settings_ZWO.json
Sep  7 20:08:27 allsky allsky.sh[2377]: Starting allsky camera...
Sep  7 20:08:43 allsky allsky.sh[498]:      ***** Starting AllSky *****
Sep  7 20:08:43 allsky allsky.sh[498]: Making sure allsky.sh is not already running...
Sep  7 20:08:44 allsky allsky.sh[498]: Trying to automatically choose between ZWO and RPI camera
Sep  7 20:08:44 allsky allsky.sh[498]: None of RPI or ZWO Cameras were found. Exiting.

I dont see any clear error. 

As for Ckuethe, I am not sure how to do what you suggest. I am new to all of this.

ckuethe, 2021-09-12:
"None of RPI or ZWO Cameras were found. Exiting." You haven't said what kind of camera you're using.

If you are using a ZWO camera, does it show up when you run `lsusb`, eg. `ID 03c3:120d ZWO ASI120MM-S`? If you're using the rpi camera, does `vcgencmd get_camera` say `supported=1 detected=1`

DamoclesSword, 2021-09-12:
Ive tried it with a 120MC-S and 178MM

DamoclesSword, 2021-09-12:
Sorry didnt see the other half. As for that "run lsusb, eg. ID 03c3:120d ZWO ASI120MM-S? If you're using the rpi camera, does vcgencmd get_camera say supported=1 detected=1" I have no idea how to do that

ckuethe, 2021-09-12:
In the same way that you installed the allsky software, you'll need to open up a shell to run those commands. Just type `lsusb` or `vcgencmd get_camera` and see what happens.

DamoclesSword, 2021-09-12:
When running lsusb, I get: 
Bus 001 Device 002: ID 03c3:178c
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

 and with the get_camera it says zero for both. Granted, this is using my 178MM and not my 120MC-S

DamoclesSword, 2021-09-12:
The new error log shows: 

Sep 12 16:50:19 allsky allsky.sh[2265]: Stop the allsky service to end this process.
Sep 12 16:50:19 allsky allsky.sh[2265]: ==========
Sep 12 16:50:19 allsky allsky.sh[2265]: === Starting daytime capture ===
Sep 12 16:50:19 allsky allsky.sh[2265]: ==========
Sep 12 16:50:19 allsky allsky.sh[2265]: Saving images with delay of 5,000 ms (5 sec)
Sep 12 16:50:19 allsky allsky.sh[2265]:   > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
Sep 12 16:50:22 allsky allsky.sh[2265]:   > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
Sep 12 16:50:24 allsky allsky.sh[2265]:   > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
Sep 12 16:50:30 allsky allsky.sh[2265]:      ***** Stopping AllSky *****
Sep 12 16:50:30 allsky allsky.sh[2265]: capture exited with retcode=2

ckuethe, 2021-09-12:
Progress, then. Your camera is detected. Now try Eric's suggestion and see if changing the 0.8 Exposure makes any difference.

DamoclesSword, 2021-09-12:
It was already turned off. So no change besides the new error log

EricClaeys, 2021-09-13:
@DamoclesSword,
Please try the attached file. It's the version 0.7 capture.cpp that knows about the new option names (e.g., "daybrightness" instead of "brightness") that were added in version 0.8. It does not use the "Version 0.8 exposure" mode.

Unpack the file, rename the .cpp file to "capture.cpp", put in ~/allsky, then "cd allsky ; make capture" and restart AllSky. It won't give any errors in the log file but you'll know it's working if you get images.

Let me know how it goes.
Eric
[capture0.7with0.8arguments.zip](https://github.com/thomasjacquin/allsky/files/7150749/capture0.7with0.8arguments.zip)

EricClaeys, 2021-09-13:
@DamoclesSword,
what did you do to have the Pi recognize your camera?  Your first attempt didn't even find the camera but the second attempt did.
Eric

EricClaeys, 2021-10-18:
@DamoclesSword,
any update?  If you have things working with the newest code, please close this issue.

EricClaeys, 2022-02-10:
Closing issue, never heard back from user.

closed 2022-02-10 by EricClaeys