AllskyTeam/allsky #462: Captured images go dark at night time recording

maserowik, 2021-09-09:
Not sure what changed but the issue I am having currently is this.  I have the version 8 up and running.  recently my images at night only have went from a nice capture to a back screen and stays that way until the next morning.  I have review the logs and I do not see anything that would cause this.  I will attach two images to this message on at night.
[image]
[image]

the first file is sized at 154 KB 
the second file is sized at 88 KB

Sep  8 20:09:15 Pi-AllSky allsky.sh[564]: STARTING EXPOSURE at: 2021-09-08 20:09:15
Sep  8 20:12:05 Pi-AllSky allsky.sh[564]:   > Saving image 'image.jpg' that started at 2021-09-08 20:09:15  (13 Âµs)
Sep  8 20:12:05 Pi-AllSky allsky.sh[564]:   > Sleeping from histogram exposure: 150000 ms
Sep  8 20:12:05 Pi-AllSky allsky.sh[564]: saveImageDay.sh: Uploading 'image.jpg' with timestamp: 20:12:05
Sep  8 20:14:35 Pi-AllSky allsky.sh[564]: ==========
Sep  8 20:14:35 Pi-AllSky allsky.sh[564]: === Starting nighttime capture ===
Sep  8 20:14:35 Pi-AllSky allsky.sh[564]: ==========
Sep  8 20:14:35 Pi-AllSky allsky.sh[564]: Saving auto exposed night images with delay of 10 ms (0 sec)
Sep  8 20:14:35 Pi-AllSky allsky.sh[564]: STARTING EXPOSURE at: 2021-09-08 20:14:35
Sep  8 20:14:45 Pi-AllSky allsky.sh[564]:   > Saving image 'image.jpg' that started at 2021-09-08 20:14:35  (11 Âµs)
Sep  8 20:14:45 Pi-AllSky allsky.sh[564]:   > Sleeping from auto exposure: 10 ms (0 sec)

I do have a observation.  this is from my allsky.log  

On the third line is see that the Sleeping from histogram exposure: 15000 ms 
In the night time capture I see is went to the sleeping from auto exposure:10 ms 

not sure on it that something or not but as soon a the night time starts i get black screens on capture.
but that is usually when I start to get black images

my set up is as follows
RASPI 3b+
ASI 224
POE to RASPI

website
http://n3bsq.com/allsky/

currently the allsky cam in inside the house.

Thanks
Mike

EricClaeys, 2021-09-11:
Mike,
The "Sleeping from ..." messages are based on the daytime delay and nighttime delay you specified.

For night you probably need to increase the exposure and max exposure times and maybe the gain.  Do the dark frames start with the first nighttime exposure ("nighttime" is based on the Angle you specified)?
I can't tell if the two pictures you attached were both taken during nighttime or not; the timestamps in the log don't match with the timestamps on the pictures.
An easy way to determine when nightime pictures begin is to change the night brightness to 51, and when you see "Brightness: 50" change to "Brightness: 51" you know it switched.
Also, if you set the Debug Level to 3 you'll get more info in the log file that may help debug the problem.

EricClaeys, 2021-09-14:
@maserowik, any update on this?

maserowik, 2021-09-14:
I reloaded the software and the issue went away

closed 2021-09-14 by maserowik