AllskyTeam/allsky #464: Meteors and Keograms.

drsaguirre, 2021-09-09:
Hi Everyone,

I have a question for everyone. :-)

Is it possible to see the trace of a meteor in the Keograms? And if it is feasible, does anyone have images to share?

Thanks for the answers. :-)

Salvador Aguirrre
Mex.

EricClaeys, 2021-09-11:
Salvador,
You're more likely to get an answer if you post it on the Facebook AllSky page.  GitHub's Issues list is used primarily by people reporting issues, so the audience isn't very big.
Given that Keograms take a small slice of the sky (I think just at the meridian), it's unlikely you'll see a trace.  You might see a bright dot.
If you post this question to Facebook, please close this issue.
Thanks - Eric

drsaguirre, 2021-09-11:
Hi Eric.

Sorry, dont have facebook¡

regards

Salvador

drsaguirre, 2021-09-11:
Eric.

Sorry, I dont have Facebook¡

Best regards

Salvador

closed 2021-09-11 by drsaguirre