AllskyTeam/allsky #466: capture exited with retcode=134

nickalter, 2021-09-10:
i get this errorOpenCV Error: Bad flag (parameter or structure field) (Unrecognized or unsupported array type) in cvGetMat, file /build/opencv-L65chJ/opencv-3.2.0+dfsg/modules/core/src/array.cpp, line 2501
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/opencv-L65chJ/opencv-3.2.0+dfsg/modules/core/src/array.cpp:2501: error: (-206) Unrecognized or unsupported array type in function cvGetMat

./allsky.sh: regel 146:  3411 Afgebroken              $ALLSKY_HOME/capture "${ARGUMENTS[@]}"
capture exited with retcode=134
*** Waiting for you to fix this.  Restart when done fixing. ***

and allsky is not runnuing

EricClaeys, 2021-09-11:
@nickalter, do you have "preview" turned on?  That only works on the desktop, not on VNC or other remote desktops.
If you don't have it turned on, please send your settings_ZWO.json file.  Look in /etc/raspap; if it's not there, look in ~/allsky.
Eric

nickalter, 2021-09-11:
Hello Eric

I have reinstall the software on new sd cart
When I do ./allsky.sh
I gets this error :
Saving images with delay of 5000 ms (5 sec)

  > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
  > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
  > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
     ***** Stopping AllSky *****
capture exited with retcode=2

And on the allsky admin panel. In live view "All sky software is starting up and”  after 2sec   "allsky software is not running”

When I use sudo service allsky status after restart allsky

I get this:
 
Sep 11 16:05:19 allsky allsky.sh[3793]:  Debug Level: 1
Sep 11 16:05:19 allsky allsky.sh[3793]:  TTY: 0 (no)
Sep 11 16:05:19 allsky allsky.sh[3793]: 
Sep 11 16:05:19 allsky allsky.sh[3793]: Will adjust gain at transitions
Sep 11 16:05:19 allsky allsky.sh[3793]: Stop the allsky service to end this process.
Sep 11 16:05:19 allsky allsky.sh[3793]: ==========
Sep 11 16:05:19 allsky allsky.sh[3793]: === Starting daytime capture ===
Sep 11 16:05:19 allsky allsky.sh[3793]: ==========
Sep 11 16:05:19 allsky allsky.sh[3793]: Saving images with delay of 5000 ms (5 sec)
Sep 11 16:05:20 allsky allsky.sh[3793]:   > ERROR: Failed getting image, status = 11 (ASI_ERROR_TIMEOUT)
lines 1-19/19 (END)

I use a raspberry pi4 2gb and zwo asi  camera on usb 3

This my settings_zwo.json files

Greetings from Nico Altena and thank you

EricClaeys, 2021-09-12:
@nickalter, the image of your settings only goes down to "targetTemp".  Would you please copy/paste your settings_ZWO.json file to your reply?
Also, did you have "preview" turned on when you got the errorOpenCV message?
Eric

nickalter, 2021-09-12:
Hello Eric
I send you the file
Nico

EricClaeys, 2021-09-13:
@nicalter, The file wasn't attached to your reply.
Eric

nickalter, 2021-09-13:
Hello Eric
I send the file

 {
   "dayautoexposure":"1",
   "dayexposure":".5",
   "daybrightness":"50",
   "daydelay":"5000",
   "daybin":"1",
   "nightautoexposure":"1",
   "nightmaxexposure":"20000",
   "nightexposure":"10000",
   "nightbrightness":"50",
   "nightdelay":"10",
   "nightautogain":"0",
   "nightmaxgain":"200",
   "nightgain":"50",
   "gaintransitiontime":"15",
   "nightbin":"1",
   "width":"0",
   "height":"0",
   "type":"99",
   "gamma":"50",
   "autowhitebalance":"0",
   "wbr":"53",
   "wbb":"90",
   "quality":"95",
   "autousb":"1",
   "usb":"80",
   "filename":"image.jpg",
   "flip":"0",
   "showTime":"1",
   "timeformat":"%Y%m%d %H:%M:%S",
   "showTemp":"1",
   "temptype":"C",
   "showExposure":"1",
   "showGain":"1",
   "showBrightness":"0",
   "showHistogram":"0",
   "text":"",
   "extratext":"",
   "extratextage":"600",
   "textlineheight":"60",
   "textx":"15",
   "texty":"30",
   "fontname":"0",
   "fontcolor":"255 255 255",
   "smallfontcolor":"0 0 255",
   "fonttype":"0",
   "fontsize":"7",
   "fontline":"1",
   "outlinefont":"0",
   "notificationimages":"1",
   "coolerEnabled":"0",
   "targetTemp":"0",
   "latitude":"60.7N",
   "longitude":"135.05W",
   "angle":"-6",
   "histogrambox":"500 500 50 50",
   "showhistogrambox":"0",
   "darkframe":"0",
   "locale":"en_US.UTF-8",
   "debuglevel":"1",
   "newexposure":"0",
   "alwaysshowadvanced":"0"
}

Thanks. 
Nico

EricClaeys, 2021-09-13:
Nico,
please try the attached file.  Unzip it, rename the *.cpp file to "capture.cpp" and move it to ~/allsky, then run "cd allsky ; make capture" and restart the service.
Let me know how it goes.
Eric
[capture0.7with0.8arguments.zip](https://github.com/thomasjacquin/allsky/files/7157330/capture0.7with0.8arguments.zip)

nickalter, 2021-09-14:
Hi Eric

I have used the file and it works only the photos are mono.
Then I use the zip file Capture 9-8-021  and run sudo make capture and sudo
reboot and that works perfect
for daytime   so tomorrow I can see if night time also works.

Thanks you Eric

EricClaeys, 2021-09-14:
@nickalter, stick with the 9-8-2021 file.  It is more recent than the other file.

If your problem is resolved, please close this issue.  Thanks.

closed 2021-09-14 by nickalter