AllskyTeam/allsky #471: New capture mode completely broken on rpi3b+ ASI120MM-S

ckuethe, 2021-09-12:
I encourage you to read this in a tone of maximum irritation, but since cussin' isn't gonna fix this any faster... just imagine some well placed profanity.

I finally got around to updating my camera since bird pooped right on the dome and I was getting reminders from dietpi about updating to Debian Bullseye...

Only to find that capture is completely broken starting with f67fa1286c04f387d6ec9047eee7dda430f1e526. My camera is detected alright:

```
Listing Attached Cameras:
  - 0 ZWO ASI120MM-S

ZWO ASI120MM-S Information:
  - Resolution:1280x960
  - Pixel Size: 3.8Î¼m
  - Supported Bin: 1 2 
  - Mono camera
  - SDK version 1, 19, 0, 0
  - Initialise Camera OK
- Sensor temperature:23.00
```

Of course it fails most unhelpfully (added instrumentation from #469 and fixed the busted defaults in #468):

```
==========
=== Starting nighttime capture ===
==========
==========
Saving auto exposed night images with delay of 10,000 ms (10 sec)

WARNING: Value of 150 greater than max value allowed (100) for control 'Gain' (#0).
WARNING: Value of 200 greater than max value allowed (100) for control 'AutoExpMaxGain' (#10).
WARNING: Value of 50 greater than max value allowed (20) for control 'Offset' (#5).
===Taking 3 images to clear buffer...
buffer clearing exposure 1 failed: -1 (Non-ASI ERROR)
buffer clearing exposure 2 failed: -1 (Non-ASI ERROR)
buffer clearing exposure 3 failed: -1 (Non-ASI ERROR)
----- SaveImgThd(): pRgb.data is null
     ***** Stopping AllSky *****
```

This happens with `-newexposure 0` or `-newexposure 1` . No surprise since  `use_new_exposure_algorithm` doesn't gate the buffer clearing operation...

To rule out hardware/kernel failure, I tried the SDK demo apps with all versions of the SDK from 1.15.0430 to 1.19.1 and the demos work just fine.

Doing a little bit of git history work, I find that the last working version was 6a493291432234e67d5cc436c8a6f786dd1b6b6a - right before @EricClaeys  big update to capture.cpp. All following versions also don't work.

Unfortunately strace doesn't show  anything useful; no failing ioctls or other system calls between those buffer clearing exposures.

ckuethe, 2021-09-12:
Repros on both x64 and armv7 with asi120mm-s and asi120mm-mini. At least on x64 it's faster to investigate...

EricClaeys, 2021-09-13:
@ckeuthe, I understand your frustration and disappointment.
Not everyone has problems with the version 0.8 exposure method.  I gave several people who were having a problem a newer capture.cpp with the "Version 0.8 Exposure" method and other changes, and it worked for everyone that tried it.  When we updated the original version 0.8 capture.cpp to include only the "Version 0.8 Exposure" method fix, several people said it doesn't work.
ZWO is looking into the problem since it doesn't happen with everyone and it happens across camera models.  Until ZWO and I determine what the issue is, I've made available the version 0.7 capture.cpp that knows about the newer version 0.8 command-line argument names (e.g., "exposure" became "nightexposure).  If you'd like to use it, rename the .cpp file in the attached zip file to "capture.cpp" and move it to ~/allsky, then "make capture".
It won't give any error messages nor will it produce helpful log messages since version 0.7 didn't do those.

Assuming that works, if you're willing to help test, I have a newer capture.cpp and a few other scripts that need testing.

Eric
[capture0.7with0.8arguments.zip](https://github.com/thomasjacquin/allsky/files/7152073/capture0.7with0.8arguments.zip)

EricClaeys, 2021-09-13:
@ckeuthe,
Neither of my ZWO cameras has the ASI_ERROR_TIMEOUT issues, and the other people that help with version 0.8 were able to get around the issue by adjusting their USB Bandwidth to 75 or 100.  I don't think Thomas' camera/Pi has the issue either, so there's no good way for us to test.
I'd love to find someone with the problem that's willing to lend me their Pi and camera, or sell them cheap so I can test.

PR #474 (2021-09-14): unbreak ROI selection
New exposure mode did not correctly set the capture ROI if the image size was manually specified. It was calling SetROI with a 0x0 window which would cause subsequent captures to fail.

At least partially fixes #471 

cc @EricClaeys
closed 2021-09-14 by thomasjacquin