AllskyTeam/allsky #472: Question about image

maserowik, 2021-09-12:
I am seeing two bright images on my views.  I have circled the bight images.  They do not move during night exposure.  Could it be bad pixel on the camera?

setup is 
raspberry pi 3b+
asi 224 camera
running a poe to the raspberry pi.

[image]

thanks

maserowik, 2021-09-12:
So I re did my dark image and the two dots do not show up at all.  could it be a reflection from something?

[image]

jcauthen78, 2021-09-12:
I'd think it's probably a reflection - internal or external is what you'll need to find out. I put a piece of cardboard over my Pi to cut out any LED lights, and it helped on one of my systems.  my 2nd system, a neighbors flood light was just high enough that the light was entering the bottom part of the flange and transferring up onto the dome in similar ways as yours (i painted the flange black along the edge and it stopped the reflection).

ckuethe, 2021-09-12:
It's probably reflections between your lens and the dome. I had the same thing on mine until I put a little bit of black PVC pipe around my dome to shield it. You might still see this on very brightly moonlit nights but shielding will generally be your friend.

maserowik, 2021-09-12:
I hope this helps I followed to following steps to turn off the leds

The line below is used to turn off the power LED
sudo sh -c 'echo 0 > /sys/class/leds/led1/brightness'

The line below is used to turn off the action LED
sudo sh -c 'echo 0 > /sys/class/leds/led0/brightness'

lets see what happens tonight
Mike

closed 2021-09-13 by maserowik