AllskyTeam/allsky #473: VirtualSky

maserowik, 2021-09-12:
Not sure what git hub to post this in so I will post it in here.  on my website when the Virtual Sky is activated the sun in not in the correct location.  I have attached a screen scrape of it.  I remember seeing somewhere that a variable need set in the virtualsky.json to flip the sun.  Anyone point me to the correct location or how to fix?

[image]

jcauthen78, 2021-09-12:
take a look at the 'config.js' file. It gives you all the options you'll want to change for your specific system, & lat/lon. It lives in the root directory of your allsky folder on your website. It should be pretty self explanatory once you open it up. It took me a few rounds of tweaking & re-uploading and refreshing to get the changes to where I needed them to be

bbillp, 2021-09-12:
I would like to see your settings for the TEXT which looks very good compared to mine.     Thanks in advance.

maserowik, 2021-09-12:
%A, %m/%d/%Y %I:%M:%S %p

EricClaeys, 2021-09-13:
@bbillp, you can set the format of the time in version 0.8 using the GUI.  It's called "Time Format".  See "man 3 strfman" for options.  If you aren't using the GUI the variable is called "timeformat" in settings_ZWO.json.

@maserowik, if this answered your question, please closet the issue.
FYI, the "allsky-website" application in GitHub is the proper place to enter issues related to the website.  "allsky-portal" is for the GUI, and "allsky" is for the software that takes and saves the pictures.

Eric

closed 2021-09-13 by maserowik