AllskyTeam/allsky #476: FTP Upload….

Shand359, 2021-09-13:
I have the time lapse, keograms and startrails working on the rpi however no matter what I do I can’t get ANYTHING to upload to my new website….

I have changed all the ftp settings to the required details but something still must be wrong…. 

Any help is appreciated please…..

Shand359, 2021-09-13:
I also just noticed the Aurora part along the top doesn’t show either

EricClaeys, 2021-09-14:
@Shand359, try the attached command.  Move it to the "allsky/scripts" folder and make it executable, then run it with no arguments and it will display a usage message.  Then run with the arguments.  Try it with a dummy file.  It will produce a log file you can look at to see what's happening.

A couple common causes for ftp problems:
1. the FTP server doesn't have the necessary directories.  If you want the keogram to go into "/keogram" on the FTP server, that directory has to already exist.
2. SSL Certificates are missing or have the wrong server name or something else.  To allow lftp to work with unknown certificates, add:   `set ssl:check-hostname false`  to `/etc/lftp.conf`

Let me know how it goes.
[runLftp.zip](https://github.com/thomasjacquin/allsky/files/7160007/runLftp.zip)

EricClaeys, 2021-10-04:
@Shand359 you can ignore the runLftp.zip file.  Instead, download the latest files.  It includes the "upload.sh" command that you can try executing.  If it fails it will display some lines that should help debug the problem.