AllskyTeam/allsky #481: Allsky stops capturing images with no specific reason

manumarchal, 2021-09-14:
I run allsky on a raspberry 4 with RPiHQ

For whatever reason, it stops capturing images within hours of being restarted. 

Looking at the  /var/log/allsky.log, there is nothing explicit there. it just stops.

Sep 12 22:58:47 allsky allsky.sh[639]: Capture command: nice raspistill --nopreview --thumb none --output image.jpg --burst -st --mode 3 --exposure off --shutter 60000000 --analoggain 10 --awb off --awb
gains 2.8,2 --vflip --saturation 50 --quality 95 -a 1104 -a 1036 -ae 32,0xff,0x808000
Sep 12 22:58:47 allsky allsky.sh[639]: Capturing & saving image done, now wait 0 seconds...
Sep 12 22:58:47 allsky allsky.sh[639]: Determine if it is day or night using variables: desired sun declination angle: -6 degrees, latitude: 51.28N, longitude: 0.18E
Sep 12 22:58:47 allsky allsky.sh[639]: Check for day or night: NIGHT (waiting for changing NIGHT into DAY)...
Sep 12 22:58:47 allsky allsky.sh[639]: Capturing & saving image...
Sep 12 22:58:47 allsky allsky.sh[639]: Capture command: nice raspistill --nopreview --thumb none --output image.jpg --burst -st --mode 3 --exposure off --shutter 60000000 --analoggain 10 --awb off --awb
gains 2.8,2 --vflip --saturation 50 --quality 95 -a 1104 -a 1036 -ae 32,0xff,0x808000
Sep 12 22:58:47 allsky allsky.sh[639]: Capturing & saving image done, now wait 0 seconds...
Sep 12 22:58:47 allsky allsky.sh[639]: Determine if it is day or night using variables: desired sun declination angle: -6 degrees, latitude: 51.28N, longitude: 0.18E
Sep 12 22:58:47 allsky allsky.sh[639]: Check for day or night: NIGHT (waiting for changing NIGHT into DAY)...
Sep 12 22:58:47 allsky allsky.sh[639]: Capturing & saving image...
Sep 12 22:58:47 allsky allsky.sh[639]: Capture command: nice raspistill --nopreview --thumb none --output image.jpg --burst -st --mode 3 --exposure off --shutter 60000000 --analoggain 10 --awb off --awb
gains 2.8,2 --vflip --saturation 50 --quality 95 -a 1104 -a 1036 -ae 32,0xff,0x808000
Sep 12 22:58:47 allsky allsky.sh[639]: Capturing & saving image done, now wait 0 seconds...
Sep 12 22:58:47 allsky allsky.sh[639]: Determine if it is day or night using variables: desired sun declination angle: -6 degrees, latitude: 51.28N, longitude: 0.18E
Sep 12 22:58:47 allsky allsky.sh[639]: Check for day or night: NIGHT (waiting for changing NIGHT into DAY)...
Sep 12 22:58:47 allsky allsky.sh[639]: Capturing & saving image...
Sep 12 22:58:47 allsky allsky.sh[639]: Capture command: nice raspistill --nopreview --thumb none --output image.jpg --burst -st --mode 3 --exposure off --shutter 60000000 --analoggain 10 --awb off --awb
gains 2.8,2 --vflip --saturation 50 --quality 95 -a 1104 -a 1036 -ae 32,0xff,0x808000

The Pi has plenty of memory. Any clue what that could be?

Pi RevisionModel 4B Revision 1.2 (2 GB)
SD Card28.95 GB (24.02 GB free)

EricClaeys, 2021-09-15:
I'm not positive, but I believe there might be a firmware update that fixes this.  Try asking in the Facebook Allsky group.
I searched Google for `rpihq camera "firmware"` and there are a few links that _may_ help.
I think the bug is that raspistill hangs.  Next time is happens, do a "ps -ef | grep raspistill" to see how long it's been running.

If this answers your question would you please close the issue?  Thanks.

manumarchal, 2021-09-16:
Thank you Eric,

I've done a a full firmware update. So far camera works fine. will report on progress tomorrow and then close issue if all good

EricClaeys, 2021-09-16:
@manumarchal, thanks for the update and I am glad it seems to be working. When you close the issue would you please describe whow you did the upgrade or provide a URL to instructions you found?  I would like to include them in the Wiki I am going to create for troubleshooting tips.
Thanks much.  Eric

manumarchal, 2021-09-17:
Hi, things are continuing to work. 

I ended up doing the following after researching rpihq camera firmware update. Did a full raspi update as per instructions here: https://www.seeedstudio.com/blog/2019/11/29/raspberry-pi-4-firmware-update-pi-4-now-runs-cooler-than-ever/

sudo apt update
sudo apt full-upgrade
sudo shutdown - r now

and then erased Allsky directory, reinstalled it (I think I was still on version 0.7), reinstalled the GUI
So far so good.

I realised my approach was a bit brutal... but working so far, and using allsky 0.8 which has some nice additions

closed 2021-09-17 by manumarchal