AllskyTeam/allsky #483: Numerical instability in histogram exposure, maybe

ckuethe, 2021-09-16:
So I just noticed my camera hunting for good exposures. Not a big problem for me during the day, but it might be worth looking into. After restarting capture, the flapping has gone away.

exp 0.1ms, gain 0, brt 0, hist 255
[image]

exp 4.28ms, gain 0, brt 0, hist 255
[image]

exp 0.1ms, gain 0, brt 0, hist 136
[image]

exp 3.90ms, gain 0, brt 0, hist 255
[image]

exp 0.1ms, gain 0, brt 0, hist 255
[image]

exp 9.75ms, gain 0, brt 0, hist 117
[image]

exp 0.33ms, gain 0, brt 0, hist 109
[image]

Running under systemd, logs going to the journal - here's a sample of the debug output...
```
Sep 15 18:10:50 skycam allsky.sh[9863]: STARTING EXPOSURE at: 2021-09-15 18:10:50
Sep 15 18:10:50 skycam allsky.sh[9863]:   > Exposure set to 100 us (0.10 ms), timeout: 5,000 ms
Sep 15 18:10:50 skycam allsky.sh[9863]:   > Got image @ exposure: 100 us (0.10 ms)
Sep 15 18:10:50 skycam allsky.sh[9863]:   > Attempt 1,  current exposure 100 us,  mean 129,  temp min exposure 100 us,  tempMaxExposure 5,000,000 us,  new exposure 100 us
Sep 15 18:10:50 skycam allsky.sh[9863]:   > Current exposure of 100 us with mean 129 was ok - no additional attempts needed.
Sep 15 18:10:50 skycam allsky.sh[9863]:   > Saving image 'image.jpg' that started at 2021-09-15 18:10:50  (18 us)
Sep 15 18:10:50 skycam allsky.sh[9863]:   > Sleeping from histogram exposure: 5,000 ms
Sep 15 18:10:55 skycam allsky.sh[9863]: STARTING EXPOSURE at: 2021-09-15 18:10:55
Sep 15 18:10:55 skycam allsky.sh[9863]:   > Exposure set to 100 us (0.10 ms), timeout: 5,000 ms
Sep 15 18:10:55 skycam allsky.sh[9863]:   > Got image @ exposure: 100 us (0.10 ms)
Sep 15 18:10:55 skycam allsky.sh[9863]:   > Attempt 1,  current exposure 100 us,  mean 33,  temp min exposure 100 us,  tempMaxExposure 5,000,000 us,  new exposure 250 us
Sep 15 18:10:55 skycam allsky.sh[9863]:   > !!! Retrying @ 250 us because 'mean < (minAcceptableHistogram * 0.6) (63)'
Sep 15 18:10:55 skycam allsky.sh[9863]:   > Exposure set to 250 us (0.25 ms), timeout: 5,000 ms
Sep 15 18:10:55 skycam allsky.sh[9863]:   > Got image @ exposure: 250 us (0.25 ms)
Sep 15 18:10:55 skycam allsky.sh[9863]:   > Attempt 2,  current exposure 250 us,  mean 33,  temp min exposure 105 us,  tempMaxExposure 5,000,000 us,  new exposure 625 us
Sep 15 18:10:55 skycam allsky.sh[9863]:   > !!! Retrying @ 625 us because 'mean < (minAcceptableHistogram * 0.6) (63)'
Sep 15 18:10:55 skycam allsky.sh[9863]:   > Exposure set to 625 us (0.62 ms), timeout: 5,001 ms
Sep 15 18:10:55 skycam allsky.sh[9863]:   > Got image @ exposure: 625 us (0.62 ms)
Sep 15 18:10:55 skycam allsky.sh[9863]:   > Attempt 3,  current exposure 625 us,  mean 33,  temp min exposure 255 us,  tempMaxExposure 5,000,000 us,  new exposure 1,560 us
Sep 15 18:10:55 skycam allsky.sh[9863]:   > !!! Retrying @ 1,560 us because 'mean < (minAcceptableHistogram * 0.6) (63)'
Sep 15 18:10:55 skycam allsky.sh[9863]:   > Exposure set to 1,560 us (1.56 ms), timeout: 5,003 ms
Sep 15 18:10:55 skycam allsky.sh[9863]:   > Got image @ exposure: 1,560 us (1.56 ms)
Sep 15 18:10:55 skycam allsky.sh[9863]:   > Attempt 4,  current exposure 1,560 us,  mean 33,  temp min exposure 630 us,  tempMaxExposure 5,000,000 us,  new exposure 3,900 us
Sep 15 18:10:55 skycam allsky.sh[9863]:   > !!! Retrying @ 3,900 us because 'mean < (minAcceptableHistogram * 0.6) (63)'
Sep 15 18:10:55 skycam allsky.sh[9863]:   > Exposure set to 3,900 us (3.90 ms), timeout: 5,007 ms
Sep 15 18:10:55 skycam allsky.sh[9863]:   > Got image @ exposure: 3,900 us (3.90 ms)
Sep 15 18:10:55 skycam allsky.sh[9863]:   > Attempt 5,  current exposure 3,900 us,  mean 33,  temp min exposure 1565 us,  tempMaxExposure 5,000,000 us,  new exposure 9,750 us
Sep 15 18:10:55 skycam allsky.sh[9863]:   > !!! Retrying @ 9,750 us because 'mean < (minAcceptableHistogram * 0.6) (63)'
Sep 15 18:10:55 skycam allsky.sh[9863]:   > Exposure set to 9,750 us (9.75 ms), timeout: 5,019 ms
Sep 15 18:10:55 skycam allsky.sh[9863]:   > Got image @ exposure: 9,750 us (9.75 ms)
Sep 15 18:10:55 skycam allsky.sh[9863]:   > Using exposure of 9,750 us with mean 110
Sep 15 18:10:55 skycam allsky.sh[9863]:   > Saving image 'image.jpg' that started at 2021-09-15 18:10:55  (25 us)
Sep 15 18:10:55 skycam allsky.sh[9863]:   > Sleeping from histogram exposure: 5,000 ms
Sep 15 18:11:00 skycam allsky.sh[9863]: STARTING EXPOSURE at: 2021-09-15 18:11:00
Sep 15 18:11:00 skycam allsky.sh[9863]:   > Exposure set to 9,750 us (9.75 ms), timeout: 5,019 ms
Sep 15 18:11:00 skycam allsky.sh[9863]:   > Got image @ exposure: 9,750 us (9.75 ms)
Sep 15 18:11:00 skycam allsky.sh[9863]:   > Attempt 1,  current exposure 9,750 us,  mean 255,  temp min exposure 100 us,  tempMaxExposure 5,000,000 us,  new exposure 3,900 us
Sep 15 18:11:00 skycam allsky.sh[9863]:   > !!! Retrying @ 3,900 us because 'mean >= max (254)'
Sep 15 18:11:00 skycam allsky.sh[9863]:   > Exposure set to 3,900 us (3.90 ms), timeout: 5,007 ms
Sep 15 18:11:00 skycam allsky.sh[9863]:   > Got image @ exposure: 3,900 us (3.90 ms)
Sep 15 18:11:00 skycam allsky.sh[9863]:   > Attempt 2,  current exposure 3,900 us,  mean 255,  temp min exposure 100 us,  tempMaxExposure 9,745 us,  new exposure 1,560 us
Sep 15 18:11:00 skycam allsky.sh[9863]:   > !!! Retrying @ 1,560 us because 'mean >= max (254)'
Sep 15 18:11:00 skycam allsky.sh[9863]:   > Exposure set to 1,560 us (1.56 ms), timeout: 5,003 ms
Sep 15 18:11:00 skycam allsky.sh[9863]:   > Got image @ exposure: 1,560 us (1.56 ms)
Sep 15 18:11:00 skycam allsky.sh[9863]:   > Attempt 3,  current exposure 1,560 us,  mean 255,  temp min exposure 100 us,  tempMaxExposure 3,895 us,  new exposure 625 us
[... 231 lines cut ...]
Sep 15 18:12:13 skycam allsky.sh[9863]:   > Attempt 1,  current exposure 100 us,  mean 32,  temp min exposure 100 us,  tempMaxExposure 5,000,000 us,  new exposure 250 us
Sep 15 18:12:13 skycam allsky.sh[9863]:   > !!! Retrying @ 250 us because 'mean < (minAcceptableHistogram * 0.6) (63)'
Sep 15 18:12:13 skycam allsky.sh[9863]:   > Exposure set to 250 us (0.25 ms), timeout: 5,000 ms
Sep 15 18:12:13 skycam allsky.sh[9863]:   > Got image @ exposure: 250 us (0.25 ms)
Sep 15 18:12:13 skycam allsky.sh[9863]:   > Attempt 2,  current exposure 250 us,  mean 32,  temp min exposure 105 us,  tempMaxExposure 5,000,000 us,  new exposure 625 us
Sep 15 18:12:13 skycam allsky.sh[9863]:   > !!! Retrying @ 625 us because 'mean < (minAcceptableHistogram * 0.6) (63)'
Sep 15 18:12:13 skycam allsky.sh[9863]:   > Exposure set to 625 us (0.62 ms), timeout: 5,001 ms
Sep 15 18:12:13 skycam allsky.sh[9863]:   > Got image @ exposure: 625 us (0.62 ms)
Sep 15 18:12:13 skycam allsky.sh[9863]:   > Attempt 3,  current exposure 625 us,  mean 32,  temp min exposure 255 us,  tempMaxExposure 5,000,000 us,  new exposure 1,560 us
Sep 15 18:12:13 skycam allsky.sh[9863]:   > !!! Retrying @ 1,560 us because 'mean < (minAcceptableHistogram * 0.6) (63)'
Sep 15 18:12:13 skycam allsky.sh[9863]:   > Exposure set to 1,560 us (1.56 ms), timeout: 5,003 ms
Sep 15 18:12:13 skycam allsky.sh[9863]:   > Got image @ exposure: 1,560 us (1.56 ms)
Sep 15 18:12:13 skycam allsky.sh[9863]:   > Attempt 4,  current exposure 1,560 us,  mean 32,  temp min exposure 630 us,  tempMaxExposure 5,000,000 us,  new exposure 3,900 us
Sep 15 18:12:13 skycam allsky.sh[9863]:   > !!! Retrying @ 3,900 us because 'mean < (minAcceptableHistogram * 0.6) (63)'
Sep 15 18:12:13 skycam allsky.sh[9863]:   > Exposure set to 3,900 us (3.90 ms), timeout: 5,007 ms
Sep 15 18:12:13 skycam allsky.sh[9863]:   > Got image @ exposure: 3,900 us (3.90 ms)
Sep 15 18:12:13 skycam allsky.sh[9863]:   > Attempt 5,  current exposure 3,900 us,  mean 32,  temp min exposure 1565 us,  tempMaxExposure 5,000,000 us,  new exposure 9,750 us
Sep 15 18:12:13 skycam allsky.sh[9863]:   > !!! Retrying @ 9,750 us because 'mean < (minAcceptableHistogram * 0.6) (63)'
Sep 15 18:12:13 skycam allsky.sh[9863]:   > Exposure set to 9,750 us (9.75 ms), timeout: 5,019 ms
Sep 15 18:12:13 skycam allsky.sh[9863]:   > Got image @ exposure: 9,750 us (9.75 ms)
Sep 15 18:12:13 skycam allsky.sh[9863]:   > Using exposure of 9,750 us with mean 108
Sep 15 18:12:13 skycam allsky.sh[9863]:   > Saving image 'image.jpg' that started at 2021-09-15 18:12:13  (21 us)
Sep 15 18:12:13 skycam allsky.sh[9863]:   > Sleeping from histogram exposure: 5,000 ms

```

EricClaeys, 2021-09-16:
@ckuethe, I assume this is with your ASI120 camera with sensor size of 1280x960?  If so, try decreasing the histogram box size (the "500 500").  When Linda developed the code, we all had larger sensors so 500 px was about 15% the size and worked well.  Once I got my ASI290 (1936x1096) I had to decrease the box size to 400px.  You might want to try 300.
I emailed Jim and Nancy asking if we should change the "500 500" setting to a percent, like 15% or 20%.  This would be yet another way to better handle different sensors.

Let me know how it goes.

ckuethe, 2021-09-16:
Indeed I am using the 120.

I like the idea of percentages. If otherwise unspecified, default to 20% of the short side, centered in the image?

Is there a reason for doing the histogram by hand rather than using OpenCV? (docs https://docs.opencv.org/master/d6/dc7/group__imgproc__hist.html)

EricClaeys, 2021-09-22:
I assume Linda didn’t know about OpenCV or didn’t know how to use it when creating the histogram.

 

Eric

EricClaeys, 2021-10-04:
@ckuethe Chris, did you try shrinking the histogram box size?
Looking at the logs I'm not sure if shrinking the box will solve this.  The camera doesn't appear to be reacting to changes in exposure time consistently (or the sky brightness was drastically changing).  For example, this sequence:
```
Sep 15 18:11:32 skycam allsky.sh[9863]:   > Attempt 1,  **current exposure 100 us,  mean 246**,  temp min exposure 100 us,  tempMaxExposure 5,000,000 us,  new exposure 100 us
Sep 15 18:11:32 skycam allsky.sh[9863]:   > Current exposure of 100 us with mean 246 was ok - no additional attempts needed.
Sep 15 18:11:32 skycam allsky.sh[9863]:   > Saving image 'image.jpg' that started at 2021-09-15 18:11:31  (18 us)
Sep 15 18:11:32 skycam allsky.sh[9863]:   > Sleeping from histogram exposure: 5,000 ms
Sep 15 18:11:37 skycam allsky.sh[9863]: STARTING EXPOSURE at: 2021-09-15 18:11:37
Sep 15 18:11:37 skycam allsky.sh[9863]:   > Exposure set to 100 us (0.10 ms), timeout: 5,000 ms
Sep 15 18:11:37 skycam allsky.sh[9863]:   > Got image @ exposure: 100 us (0.10 ms)
Sep 15 18:11:37 skycam allsky.sh[9863]:   > Attempt 1,  **current exposure 100 us,  mean 33,**  temp min exposure 100 us,  tempMaxExposure 5,000,000 us,  new exposure 250 us
```

Why does it go from a mean of 246 @ 100 us exposure to mean of 33 @ 100 us exposure 5 seconds later?
I've seen this before and really think it's related to the ZWO exposure bug.
The camera also isn't very linear - it got a mean of 33 from 100 us to 3,900 us, then jumped to mean 110 at 9,750  us.  This makes doing auto exposure next to impossible.

ckuethe, 2021-10-05:
Yeah, I shrunk the histogram box size. It's not a big issue for me because I'm not saving images during the day, but it seemed odd.

I think we may want to:
* investigate whether or not OpenCV's histogram features (https://docs.opencv.org/master/d6/dc7/group__imgproc__hist.html) might be useful
* whether OpenCV can do some sort of autoexposure control (https://stackoverflow.com/questions/38101288/calculate-auto-exposure-in-opencv)
* maybe use a PID controller to servo the exposure settings
* replace ZWO auto-exposure with our own auto-exposure
* add an additional exposure compensation (-2 .. +2 exposure units) so you can intentionally over/under expose if you dislike what autoexposure does

EricClaeys, 2021-10-05:
@ckuethe,  your last two bullets:

- The daytime histogram code replaces ZWO's auto-exposure with our own, if the user has daytime auto exposure on.  We actually turn it off, the use the histogram code.  If ZWO ever fixes the bug, we'll have to see if it's better than our histogram method, and if so, we can delete the histogram code.
- Setting the "brightness" (which ZWO internally calls "ASI_AUTO_TARGET_BRIGHTNESS" already does exposure compensation if using the histogram method during the day.  It only works with auto exposure.  At night the camera handles it since the histogram code is only for daytime.

ckuethe, 2021-10-05:
Let's just say that I disagree with ZWO's implementation of autoexposure, and would like to have some additional controls and strategies for evaluating what a properly exposed image is.

EricClaeys, 2022-06-30:
Closing - this request was added to the Projects list - see the `Projects` link at the top of any Allsky GitHub page.
Project Title: **Port RPiHQ modeMean to ZWO**

Our plan is to port the Allsky auto-exposure algorithm from the RPi camera to ZWO.  Hopefully this will eliminate the nighttime flicker and ASI_ERROR_TIMEOUT's some users see.
That algorithm will also support user-generated masks that will determine what part of the picture is used to determine the mean image brightness.  The upcoming "overlay editor" will allow users to easily create the auto-exposure mask.

closed 2022-06-30 by EricClaeys